me-no-dev/ESPAsyncWebServer #3: AsyncServer and OTA

andig, 2016-01-22:
When OTA starts it blocks the synchronous web server. The async server however seems to keep running according to my logs. As both async server and ota may e.g. update spiffs that seems to call for trouble.
Should async server be stopped on OTA start? How?

me-no-dev, 2016-01-22:
I see no problem to run both as I have done it and have downloaded/uploaded at the same time. No problem ;)

andig, 2016-01-22:
I was more thinking about OTA-updating SPIFFS?

me-no-dev, 2016-01-22:
first of all it does not update spiffs :) it writes linearly and much faster than spiffs.
second of all I have run concurrent upload and downloads, which are much more demanding than OTA.
Stay assured OTA is fine ;) There are many checks performed before the go ahead is given for the firmware to be updated. There is no chance the data got there and is corrupted and things went ahead.
As a matter of fact I will move OTA to async soon so everything can work as usual without interrupting services until successful OTA image has been written and ready to restart.

andig, 2016-01-22:
Excellent. Got my sketch working now and wifi performance is really overwhelming. Thank you so much!

andig, 2016-01-22:
> first of all it does not update spiffs

Ah, of course. It writes below spiffs, the stops all processes, then copies over. I forgot...

closed 2016-01-22 by andig

kiralikbeyin, 2016-07-28:
@andig Please can you share OTA code with html page if possible? Thanks.

andig, 2016-07-29:
@kiralikbeyin not sure what you're looking for- just check the OTA examples in the esp8266 repo?

kiralikbeyin, 2016-07-29:
I am trying to convert https://github.com/gmag11/FSBrowser/ to ESPAsyncWebServer
It uses normal esp web server lib and there is an update firmware page with these codes;

How to make it work with ESPAsyncWebServer, this is what i search...

```
void setUpdateMD5() {
  browserMD5 = "";
#ifdef DEBUG_WEBSERVER
  DBG_OUTPUT_PORT.printf("Arg number: %d\n", server.args());
#endif // DEBUG_WEBSERVER
  if (server.args() > 0)  // Read hash
  {
    //String temp = "";
    for (uint8_t i = 0; i < server.args(); i++) {
#ifdef DEBUG_WEBSERVER
      DBG_OUTPUT_PORT.printf("Arg %s: %s\n", server.argName(i).c_str(), server.arg(i).c_str());
#endif // DEBUG_WEBSERVER
      if (server.argName(i) == "md5") {
        browserMD5 = urldecode(server.arg(i));
        Update.setMD5(browserMD5.c_str());
        continue;
      } if (server.argName(i) == "size") {
        updateSize = server.arg(i).toInt();
#ifdef DEBUG_WEBSERVER
        DBG_OUTPUT_PORT.printf("Update size: %u\n", server.arg(i).toInt());
#endif // DEBUG_WEBSERVER
        continue;
      }
    }
    server.send(200, "text/html", "OK --> MD5: " + browserMD5);
  }

}

void updateFirmware () {
  // handler for the file upload, get's the sketch bytes, and writes
  // them through the Update object
  HTTPUpload& upload = server.upload();
  if (upload.status == UPLOAD_FILE_START) {
    WiFiUDP::stopAll();
#ifdef DEBUG_WEBSERVER
    DBG_OUTPUT_PORT.printf("Update: %s\n", upload.filename.c_str());
#endif // DEBUG_WEBSERVER
    //uint32_t maxSketchSpace = (ESP.getSketchSize() - 0x1000) & 0xFFFFF000;
    uint32_t maxSketchSpace = ESP.getSketchSize();
#ifdef DEBUG_WEBSERVER
    //uint32_t oldValue = (ESP.getSketchSize() - 0x1000) & 0xFFFFF000;
    DBG_OUTPUT_PORT.printf("Max free scketch space: %u\n", maxSketchSpace);
    DBG_OUTPUT_PORT.printf("New scketch size: %u\n", updateSize);
    //DBG_OUTPUT_PORT.printf("Old value: %u\n", oldValue);
#endif // DEBUG_WEBSERVER
    if (browserMD5 != NULL && browserMD5 != "") {
      Update.setMD5(browserMD5.c_str());
#ifdef DEBUG_WEBSERVER
      DBG_OUTPUT_PORT.printf("Hash from client: %s\n", browserMD5.c_str());
#endif // DEBUG_WEBSERVER
    }
    if (!Update.begin(updateSize)) {//start with max available size
#ifdef DEBUG_WEBSERVER
      Update.printError(DBG_OUTPUT_PORT);
#endif // DEBUG_WEBSERVER
    }
  }
  else if (upload.status == UPLOAD_FILE_WRITE) {
#ifdef DEBUG_WEBSERVER
    DBG_OUTPUT_PORT.print(".");
#endif // DEBUG_WEBSERVER
    if (Update.write(upload.buf, upload.currentSize) != upload.currentSize) {
#ifdef DEBUG_WEBSERVER
      Update.printError(DBG_OUTPUT_PORT);
#endif // DEBUG_WEBSERVER
    }
  }
  else if (upload.status == UPLOAD_FILE_END) {
    String updateHash;
    if (Update.end(true)) { //true to set the size to the current progress
#ifdef DEBUG_WEBSERVER
      updateHash = Update.md5String();
      DBG_OUTPUT_PORT.printf("Upload finished. Calculated MD5: %s\n", updateHash.c_str());
      DBG_OUTPUT_PORT.printf("Update Success: %u\nRebooting...\n", upload.totalSize);
#endif // DEBUG_WEBSERVER
    }
    else {
#ifdef DEBUG_WEBSERVER
      updateHash = Update.md5String();
      DBG_OUTPUT_PORT.printf("Upload failed. Calculated MD5: %s\n", updateHash.c_str());
      Update.printError(DBG_OUTPUT_PORT);
#endif // DEBUG_WEBSERVER
    }
  }
  else if (upload.status == UPLOAD_FILE_ABORTED) {
    Update.end();
#ifdef DEBUG_WEBSERVER
    DBG_OUTPUT_PORT.println("Update was aborted");
#endif // DEBUG_WEBSERVER
  }
  delay(2);
}

******************

  server.on("/update", HTTP_POST, [&]() {
    if (!checkAuth())
      return server.requestAuthentication();
    server.sendHeader("Connection", "close");
    server.sendHeader("Access-Control-Allow-Origin", "*");
    String gonderabi = String("http://") + config.DeviceName.c_str() + String(".local");
    server.sendHeader("Location", gonderabi, true);
    server.send ( 301, "text/plain", "");
    server.send(200, "text/html", (Update.hasError()) ? "FAIL" : String("<META http-equiv=\"refresh\" content=\"15;URL=/\">Ayarlar kaydediliyor... http://") + gonderabi );

    delay(50);
    //	server.send(200, "text/html", (Update.hasError()) ? "FAIL" : "<META http-equiv=\"refresh\" content=\"15;URL=/update\">Update correct. Restarting...");
    ESP.restart();
  }, updateFirmware);
```

me-no-dev, 2016-07-29:
do not delay in the callbacks and other than that what is the problem? converting it to async will require you to change a few lines but that's about it
again all well explained in README. Do not restart from the callbacks either ;) set a flag and do it from the loop.

kiralikbeyin, 2016-07-29:
@me-no-dev Where is my mistake can you see?

Serial output: 

END Setup
pm open,type:2 0
GET[md5]: 71a8eef2e4c53460ec0abaf7c79677a4
GET[size]: 334192
ARG[md5]: 71a8eef2e4c53460ec0abaf7c79677a4
bak md5
bakbakbakbakbak md5
AAAAAAA
ARG[size]: 334192
bak size
bakbakbakbakbak size

Update size: 334192
UploadStart: FSBrowser-SON.ino.nodemcu.bin
UploadStart: FSBrowser-SON.ino.nodemcu.bin,B
UploadStart:  1073693836 B
UploadStart: FSBrowser-SON.ino.nodemcu.bin, 1460 B
UploadEnd: FSBrowser-SON.ino.nodemcu.bin, 334192 B
Max free scketch space: 327680
New scketch size: 334192
Hash from client: 71a8eef2e4c53460ec0abaf7c79677a4
sleep disable
Upload failed. Calculated MD5: d41d8cd98f00b204e9800998ecf8427e
ERROR[7]: MD5 Check Failed
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl

```
server.on("/setmd5", HTTP_GET, [](AsyncWebServerRequest * request) {
    if (!checkAuth())
      return checkAuth();
    browserMD5 = "";

    int params = request->params();
    for (int i = 0; i < params; i++) {
      AsyncWebParameter* p = request->getParam(i);
      if (p->isFile()) { //p->isPost() is also true
        Serial.printf("FILE[%s]: %s, size: %u\n", p->name().c_str(), p->value().c_str(), p->size());
      } else if (p->isPost()) {
        Serial.printf("POST[%s]: %s\n", p->name().c_str(), p->value().c_str());
      } else {
        Serial.printf("GET[%s]: %s\n", p->name().c_str(), p->value().c_str());
      }
    }

    int args = request->args();
    for (int i = 0; i < args; i++) {
      Serial.printf("ARG[%s]: %s\n", request->argName(i).c_str(), request->arg(i).c_str());
      String bak = request->argName(i).c_str();
      String bakto = request->arg(i).c_str();
      Serial.println("bak " + bak);
      //  Serial.print("bakto "+bakto);
      String bak2 = request->argName(i);
      String bakto2 = request->arg(i);
      Serial.println("bakbakbakbakbak " + bak2);
      //  Serial.print("bakto "+bakto2);

      if ( bak == "md5") {
        //
        //      browserMD5 = urldecode(request->arg(i).c_str());
        //      Update.setMD5(browserMD5.c_str());
        browserMD5 = urldecode(request->arg(i));
        Update.setMD5(browserMD5.c_str());
        Serial.println("AAAAAAA");

      } if (bak == "size") {
        updateSize = request->arg(i).toInt();

        Serial.println("ahahahahha bok " + updateSize);
#ifdef DEBUG_WEBSERVER
        DBG_OUTPUT_PORT.printf("Update size: %u\n", updateSize);
#endif // DEBUG_WEBSERVER
      }
 
    }
 
    // server.send(200, "text/html", "OK --> MD5: " + browserMD5);
    request->send(200, "text/html", "OK --> MD5: " + browserMD5);
 
  });
```

```
  server.on("/update", HTTP_POST, [](AsyncWebServerRequest * request) {

    if (!checkAuth())
      return checkAuth();

    AsyncWebServerResponse *response = request->beginResponse(200); //Sends 404 File Not Found
    response->addHeader("Connection", "close");
    response->addHeader("Access-Control-Allow-Origin", "*");
    request->send(response);

    request->redirect("/");

    //  server.send(200, "text/html", (Update.hasError()) ? "FAIL" : "<META http-equiv=\"refresh\" content=\"15;URL=/update\">Update correct. Restarting...");
   // ESP.restart();
  }, handleUploadme);

```

```

void handleUploadme(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
  if (!index) {
    Serial.printf("UploadStart: %s\n", filename.c_str());
    Serial.printf("UploadStart: %s,B\n", filename.c_str(), index );
    Serial.printf("UploadStart:  %u B\n", filename.c_str(),  len);
    Serial.printf("UploadStart: %s, %u B\n", filename.c_str(), index + len);

  }
  //  for(size_t i=0; i<len; i++){
  //    Serial.write(data[i]);
  //  }
  if (final) {

    Serial.printf("UploadEnd: %s, %u B\n", filename.c_str(), index + len);
    uint32_t maxSketchSpace =  (ESP.getSketchSize() - 0x1000) & 0xFFFFF000;

    if (maxSketchSpace == maxSketchSpace)
    {

      String updateHash;

      updateHash = Update.md5String();
      Update.setMD5(browserMD5.c_str());

#ifdef DEBUG_WEBSERVER
      //uint32_t oldValue = (ESP.getSketchSize() - 0x1000) & 0xFFFFF000;
      DBG_OUTPUT_PORT.printf("Max free scketch space: %u\n", maxSketchSpace);
      DBG_OUTPUT_PORT.printf("New scketch size: %u\n", updateSize);
      //DBG_OUTPUT_PORT.printf("Old value: %u\n", oldValue);
#endif // DEBUG_WEBSERVER
      if (browserMD5 != NULL && browserMD5 != "") {
        Update.setMD5(browserMD5.c_str());
#ifdef DEBUG_WEBSERVER
        DBG_OUTPUT_PORT.printf("Hash from client: %s\n", browserMD5.c_str());
#endif // DEBUG_WEBSERVER
      }
      if (!Update.begin(updateSize)) {//start with max available size
#ifdef DEBUG_WEBSERVER
        Update.printError(DBG_OUTPUT_PORT);
#endif // DEBUG_WEBSERVER
      }
      if (Update.end(true)) { //true to set the size to the current progress
#ifdef DEBUG_WEBSERVER
        updateHash = Update.md5String();
        DBG_OUTPUT_PORT.printf("Upload finished. Calculated MD5: %s\n", updateHash.c_str());
        // DBG_OUTPUT_PORT.printf("Update Success: %u\nRebooting...\n", upload.totalSize);
#endif // DEBUG_WEBSERVER
      }
      else {
#ifdef DEBUG_WEBSERVER
        updateHash = Update.md5String();
        DBG_OUTPUT_PORT.printf("Upload failed. Calculated MD5: %s\n", updateHash.c_str());
        Update.printError(DBG_OUTPUT_PORT);
#endif // DEBUG_WEBSERVER
      }
    }
  }
}
```

me-no-dev, 2016-07-29:
wrong MD5, why... I can not tell

probonopd, 2017-07-27:
Let's provide an example sketch like WebUpdate but for ESPAsyncWebServer. While we are at it, why not convert over all the other webserver examples from the non-Async version, too.

pit001, 2017-08-23:
A++

mcmega, 2017-12-06:
Did anyone get an update?

JMishou, 2017-12-30:
If anyone comes across this looking for an ESPAsyncWebServer WebUpdate I have a basic example here...

https://gist.github.com/JMishou/60cb762047b735685e8a09cd2eb42a60

tablatronix, 2020-12-15:
This example was never created, afaik

pedros89, 2021-07-19:
These two libraries cannot work together

#include "ESPAsyncWebServer.h"
#include "HttpsOTAUpdate.h"
https://github.com/espressif/arduino-esp32/tree/master/libraries/Update/examples/HTTPS_OTA_Update

I need to connect to my website and do the update.

I need to update my ESP and also to send and receive data. I managed to do the things separately but when I put them together the two libraries clashes

tsctrl, 2022-05-16:
yes, same issue with me here https://github.com/me-no-dev/ESPAsyncWebServer/issues/1159