me-no-dev/ESPAsyncWebServer #4: gzipped file only serverd if uncompressed file exists, too

andig, 2016-01-23:
Problem is that .gz existence is checked in two places. See https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/AsyncWebServerHandlerImpl.h#L24 

Needs change similar to this or better:

      _isFile = _fs.exists(path) || _fs.exists((String(path)+".gz").c_str());

me-no-dev, 2016-01-23:
done! and thanks :) https://github.com/me-no-dev/ESPAsyncWebServer/commit/3c658dcae864daf2d3d7f194281d016cce148830

andig, 2016-01-23:
> done! and thanks :) 3c658dc

I have to thank you. The async server is from another dimension ;)

andig, 2016-01-23:
:)

closed 2016-01-23 by andig

me-no-dev, 2016-01-23:
I have been dreaming about it for a while :) it took some time to take on the task to parse partially received data and to send stuff in chunks :0 today I stumbled upon chunked output which can help much in streaming unknown length content