me-no-dev/ESPAsyncWebServer #5: code structure

andig, 2016-01-25:
During debugging I kept being confused by AsyncWebServerHandlerImpl ws WebHandlers, the sooner only containing static and callback handlers. Wouldn't it make sense to move those to the same file?

me-no-dev, 2016-01-25:
AsyncWebServerHandlerImpl keeps all implementations of the AsyncWebHandler (thus the name) the cpp file has their source. It should not be needed for inclusion unless you extend one of the classes there. They all look to the server as AsyncWebResponse and the server treats them as such (like Print implementations)

andig, 2016-01-25:
> AsyncWebServerHandlerImpl keeps all implementations of the AsyncWebHandler

I found it confusion as part (and only part, e.g. not the constructor) of the AsyncStaticWebHandler implementation in turn is in WebHandlers.cpp. Also those are the only two files prefixed with Async.

But definitely a minor thing, please feel free to close.

me-no-dev, 2016-01-25:
all headers have Async in their name (apart from the StringArray). Source files do not because they can not be mixed up with other source files (you include the headers and not the source) :)
Some names will probably change in a very near future :) @sticilface has done some major work to make ArduinoJson async as well :)

andig, 2016-02-10:
Fixed by https://github.com/me-no-dev/ESPAsyncWebServer/commit/40d5d4415e7e8f333a100ce6795b4d0ef2163140 I would say- thank you!

closed 2016-02-10 by andig

me-no-dev, 2016-02-10:
:)