me-no-dev/ESPAsyncWebServer #6: Error when build

VNArmy, 2016-01-26:
Hello,
I got following error when build the ESP_AsyncFSBrowser example

Arduino: 1.6.7 (Windows 7), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck"

Build options changed, rebuilding all
In file included from C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:0,

                 from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:32:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/include/wl_definitions.h:73:17: error: redeclaration of 'CLOSED'

   CLOSED      = 0,

                 ^

In file included from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/lwip/tcp.h:128:3: note: previous declaration 'tcp_state CLOSED'

   CLOSED      = 0,

   ^

In file included from C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:0,

                 from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:32:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/include/wl_definitions.h:74:17: error: redeclaration of 'LISTEN'

   LISTEN      = 1,

                 ^

In file included from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/lwip/tcp.h:129:3: note: previous declaration 'tcp_state LISTEN'

   LISTEN      = 1,

   ^

In file included from C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:0,

                 from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:32:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/include/wl_definitions.h:75:17: error: redeclaration of 'SYN_SENT'

   SYN_SENT    = 2,

                 ^

In file included from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/lwip/tcp.h:130:3: note: previous declaration 'tcp_state SYN_SENT'

   SYN_SENT    = 2,

   ^

In file included from C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:0,

                 from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:32:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/include/wl_definitions.h:76:17: error: redeclaration of 'SYN_RCVD'

   SYN_RCVD    = 3,

                 ^

In file included from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/lwip/tcp.h:131:3: note: previous declaration 'tcp_state SYN_RCVD'

   SYN_RCVD    = 3,

   ^

In file included from C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:0,

                 from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:32:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/include/wl_definitions.h:77:17: error: redeclaration of 'ESTABLISHED'

   ESTABLISHED = 4,

                 ^

In file included from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/lwip/tcp.h:132:3: note: previous declaration 'tcp_state ESTABLISHED'

   ESTABLISHED = 4,

   ^

In file included from C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:0,

                 from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:32:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/include/wl_definitions.h:78:17: error: redeclaration of 'FIN_WAIT_1'

   FIN_WAIT_1  = 5,

                 ^

In file included from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/lwip/tcp.h:133:3: note: previous declaration 'tcp_state FIN_WAIT_1'

   FIN_WAIT_1  = 5,

   ^

In file included from C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:0,

                 from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:32:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/include/wl_definitions.h:79:17: error: redeclaration of 'FIN_WAIT_2'

   FIN_WAIT_2  = 6,

                 ^

In file included from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/lwip/tcp.h:134:3: note: previous declaration 'tcp_state FIN_WAIT_2'

   FIN_WAIT_2  = 6,

   ^

In file included from C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:0,

                 from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:32:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/include/wl_definitions.h:80:17: error: redeclaration of 'CLOSE_WAIT'

   CLOSE_WAIT  = 7,

                 ^

In file included from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/lwip/tcp.h:135:3: note: previous declaration 'tcp_state CLOSE_WAIT'

   CLOSE_WAIT  = 7,

   ^

In file included from C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:0,

                 from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:32:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/include/wl_definitions.h:81:17: error: redeclaration of 'CLOSING'

   CLOSING     = 8,

                 ^

In file included from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/lwip/tcp.h:136:3: note: previous declaration 'tcp_state CLOSING'

   CLOSING     = 8,

   ^

In file included from C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:0,

                 from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:32:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/include/wl_definitions.h:82:17: error: redeclaration of 'LAST_ACK'

   LAST_ACK    = 9,

                 ^

In file included from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/lwip/tcp.h:137:3: note: previous declaration 'tcp_state LAST_ACK'

   LAST_ACK    = 9,

   ^

In file included from C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:28:0,

                 from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:32:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/include/wl_definitions.h:83:17: error: redeclaration of 'TIME_WAIT'

   TIME_WAIT   = 10

                 ^

In file included from C:\Program Files (x86)\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:

C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/lwip/tcp.h:138:3: note: previous declaration 'tcp_state TIME_WAIT'

   TIME_WAIT   = 10

   ^

exit status 1
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

me-no-dev, 2016-01-26:
what version ESP core?
needs to be the git master :)
you look to be at 2.0.0 and you will have loots of problems there

mkeyno, 2016-03-01:
hi @me-no-dev 
I have minor problem and it seems  ESP8266WebServer.h  not bind totally with ESPAsyncWebServer.h and I have following error and when I include the  ESP8266WebServer.h my sketch freeze
my platform 
last esp staging version 2 on ardino 1.6.4 

sketch_mar01b.ino: In function 'void setup()':
sketch_mar01b.ino:24:3: error: 'WiFi' was not declared in this scope
sketch_mar01b.ino:24:13: error: 'WIFI_STA' was not declared in this scope
sketch_mar01b.ino:26:38: error: 'WL_CONNECTED' was not declared in this scope
sketch_mar01b.ino:39:6: error: 'handleUpload' was not declared in this scope
'WiFi' was not declared in this scope

me-no-dev, 2016-03-01:
the ESP8266WebServer and the Async web server are not made to work together. Because of some structure decisions in the ESP Arduino Core, it's impossible to have both included.

mkeyno, 2016-03-01:
hi @me-no-dev 
so whats make this error, I can not even compile the ESPAsyncTCP example and have following error

 \arduino-1.6.4\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp: In member function 'size_t AsyncPrinter::_sendBuffer()':
 \arduino-1.6.4\libraries\ESPAsyncTCP\src\AsyncPrinter.cpp:116:34: error: 'class cbuf' has no member named 'available'
   size_t available = _tx_buffer->available();
                                  ^
Error compiling

Van4ik, 2016-03-16:
Hi, I have the same error:
...\2.1.0-rc2\libraries\ESPAsyncTCP-master\src\AsyncPrinter.cpp: In member function 'size_t AsyncPrinter::_sendBuffer()':
...\2.1.0-rc2\libraries\ESPAsyncTCP-master\src\AsyncPrinter.cpp:116:34: error: 'class cbuf' has no member named 'available'
   size_t available = _tx_buffer->available();
                                  ^
Error compiling.

Can you help us? Thanks.

me-no-dev, 2016-03-17:
you guys need to use the git versions of the esp8266 arduino repository.
I imagine your cores are older versions and that is why you get that error.
The method in question was added recently

Van4ik, 2016-03-17:
Thanks. I removed Esp8266 board in Arduino and followed by this link https://github.com/esp8266/Arduino#using-git-version
Now, I uploaded build into Esp. But in your sample ESP_AsyncFSBrowser works only /heap
Url /edit  works sometimes and i see authorisation window. But nothing more. I will continue to test, need more times

me-no-dev, 2016-03-18:
do you have the files on the SPIFFS? the example is using files that it is serving from it.
If you do not know how to upload the files let me know.
The user pass that it is asking you for is configurable in the sketch (you can remove it also)

Van4ik, 2016-03-20:
No, I have not the files on the SPIFFS. I don't know how upload files from folder DATA. I guess i must compile binary file from my contents and then upload it into the flash (ESP8266-12 q). I am not understand how upload files in SPIFFS. I want use  some files .html and .css. and json queries.
In current version of my project the barometer (based on Advanced Webserver and bmp180)  i am using variables, which sizes is near 7kb. Periodically the device reboots due to system exceptions when I try to make a request. Maybe you can give some advice me how i upload folder Data into the flash.
I think I need compile tools https://github.com/igrr/mkspiffs but i am get error in windows. 
I am attach project based on your ESPAsyncWebServer code.

When trigger triggering i get error
tmr.attach(3,updateData);
....
void updateData(){
  bmp180temp = bmp.readTemperature();
  bmp180pres = bmp.readPressure() * 0.0075006375541921,2;
  bmp180height = bmp.readAltitude();
  updateUpTime();
  Serial.println(ESP.getFreeHeap());
}

[ESP_AsyncFSBrowser_my.txt](https://github.com/me-no-dev/ESPAsyncWebServer/files/181507/ESP_AsyncFSBrowser_my.txt)

VNArmy, 2016-03-21:
Go to IP-Address/edit to upload.

Van4ik, 2016-03-21:
When i navigate my device by url http://192.168.4.200/edit
I get message in console
NOT_FOUND: GET http://192.168.4.200/edit

When i navigate IP-adress/list i am get authorisation window in my browser. After I putting admin/admin a get long waiting and nothing more.
In both cases, I get a blank page. I am use Chrome browser.
Some times in console appear next messages
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)
wdt reset

One time i uncomment the next line
 //Serial.printf("format start\n"); SPIFFS.format();  Serial.printf("format end\n");
I thought it would prepare a flash. It did not help.

me-no-dev, 2016-03-21:
https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md#uploading-files-to-file-system
follow the instructions, get the tool and it will upload the contents of the sketch data subfolder to the filesystem

Van4ik, 2016-03-21:
Thank you me-no-dev. Your version of the code is stable. I many times pressed the F5 key, and the web server responds with confidence, because when I uploaded the html from the variable, device often reboots. Great!
I still have question. A little bit later.

me-no-dev, 2016-03-21:
Using this server I have been able to upload/download and display pages at the same time. Depending on the browser, you can have up to 5 concurrent connections (limited by the ESP's network stack).
The regular server on the other hand, handles the connections one by one, so having too many connections while the server is busy is usually resulting in reboots because of network overflow and other issues.

closed 2016-04-27 by me-no-dev