me-no-dev/ESPAsyncWebServer #7: async server shows crash with heap4c

andig, 2016-01-27:
Trying to track down some crashes I've implemented heap4c and a simple sketch:

	#include <ESP8266WiFi.h>
	#include <ESPAsyncWebServer.h>

	AsyncWebServer g_server(80);

	void setup() {
	  Serial.begin(115200);
	  WiFi.begin();
	  g_server.begin();
	}

	void loop() {
	  delay(100);
	}

This will crash with heap4c present and run fine for ages (as does a more complex sketch) as long as g_server is not started. As you can see there are no handlers attached, presence of g_server alone is enough to cause the crash:

	m 40 88 33 <
	m 12 64 34 <
	m 32 80 35 <
	m 76 128 36 <
	f 64 34 <
	f 80 35 <
	f 128 36 <
	f 88 33 <
	f 64 32 <
	m 16 64 37 <
	f 64 37 <
	m 16 64 38 <
	m 40 88 39 <
	m 12 64 40 <
	m 32 80 41 <
	m 76 128 42 <
	f 64 40 <
	f 80 41 <
	f 128 42 <
	f 88 39 <
	f 64 38 <
	m 16 64 43 <
	m 40 88 44 <
	m 12 64 45 <
	m 32 80 46 <
	m 76 128 47 <
	f 64 45 <
	f 80 46 <
	f 128 47 <
	f 88 44 <
	f 64 3 <
	m 16 64 48 <
	f 64 43 <
	m 16 64 49 <
	f 64 49 <
	m 16 64 50 <
	m 40 88 51 <
	m 12 64 52 <
	m 32 80 53 <
	m 76 128 54 <
	f 64 52 <
	f 80 53 <
	f 128 54 <
	f 88 51 <
	f 64 50 <
	m 16 64 55 <
	m 40 88 56 <
	m 12 64 57 <
	m 32 80 58 <
	m 76 128 59 <
	f 64 57 <
[... 695 lines cut ...]

	Exception (28):
	epc1=0x4021dec7 epc2=0x00000000 epc3=0x00000000 excvaddr=0x007e8000 depc=0x00000000

	ctx: sys 
	sp: 3ffffd70 end: 3fffffb0 offset: 01a0

	>>>stack>>>
	3fffff10:  4021bee2 3fff0d20 3fff1358 3ffee820  
	3fffff20:  3ffee844 4021baf1 3fff0d20 00000000  
	3fffff30:  3ffee844 3ffee820 0000cccc 4021bae0  
	3fffff40:  69b13f15 000019dc 00000001 00000011  
	3fffff50:  00000000 00000000 4021a8f6 3fff0cd8  
	3fffff60:  3fff0b98 3ffe9be6 3fff0b98 4021968b  
	3fffff70:  3fff0b98 00000014 40219c36 3fff0cd8  
	3fffff80:  3fff0b98 3fffdc80 3fff0c38 00000001  
	3fffff90:  402255ef 3fff0cd8 00000000 40205bdb  
	3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
	<<<stack<<<

	 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

	 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

	wdt reset

me-no-dev, 2016-01-27:
so you are testing how the server performs when there are no handlers attached?
Do you wait for the WiFi to start?
Also can you please use addr2line when such exception occurs and get where it happened?
Interesting addresses above are:
0x4021dec7
4021bee2
4021baf1

andig, 2016-01-27:
> Do you wait for the WiFi to start?

will add.

> Also can you please use addr2line when such exception occurs and get where it happened?

how to (on windows)?

andig, 2016-01-27:
No success (new dump, new addresses):

	C:\andi\arduino\hardware\eps8266com\esp8266\tools\xtensa-lx106-elf\bin>xtensa-lx106-elf-addr2line -e C:\Users\xx\AppData\Local\Temp\build662bf6597004a59e628e0f8ceba459f9.tmp\async.ino.elf 0x4021befa 0x4021bb09 0x4021dedf 0x4021baf8
	??:?
	??:?
	??:?
	??:?

addr2line needs be run on the .elf, right?

andig, 2016-01-27:
> so you are testing how the server performs when there are no handlers attached?

I'm trying to build a minimal test case for the crashes I've seen. After removing all my code and then all handlers it still crashed if there's http requests coming in. I'm suspecting the async server itself (no better idea with the minimal sketch above..)

me-no-dev, 2016-01-27:
just do 
```bash
C:\andi\arduino\hardware\eps8266com\esp8266\tools\xtensa-lx106-elf\bin>xtensa-lx106-elf-addr2line C:\Users\xx\AppData\Local\Temp\build662bf6597004a59e628e0f8ceba459f9.tmp\async.ino.elf
```
then enter the addresses (they change with the code that's fine) just look for the addresses starting with 402 in the dump (some 401 might show as well)

me-no-dev, 2016-01-27:
the server should return 500 on request. How often does your curl command send requests?

me-no-dev, 2016-01-27:
does it wait for them to finish before starting a new one?

andig, 2016-01-27:
> just do 

error: `xtensa-lx106-elf-addr2line: 'a.out': No such file`. I need the -e

> then enter the addresses (they change with the code that's fine) just look for the addresses starting with 402 in the dump (some 401 might show as well)

Same result. Just ??.

> the server should return 500 on request. How often does your curl command send requests? Does it wait for them to finish before starting a new one?

Even a single request is enough to crash if heap4c is compiled in.

me-no-dev, 2016-01-27:
and what if not using heap4c?
and you were correct. I actually execute xtensa-lx106-elf-addr2line -aipfC -e [elf]

andig, 2016-01-27:
> and what if not using heap4c?

no crash, runs for ages. But heap4c does heap poisoning as I've understood.

andig, 2016-01-27:
I've gone back to an old version of my sketch with sync server + heap4c. No crashes.

andig, 2016-01-27:
For what it's worth: naked AsyncServer with heap4c doesn't crash. AyncWebServer does.

me-no-dev, 2016-01-27:
maybe stop printing from heap4 ? and you need to understand that I can not support heap4 nor alter it's code as we are not supposed to use it at all

me-no-dev, 2016-01-27:
you are printing from interrupt using HardwareSerial that uses interrupts. Many shit can go wrong
the other one you print from the loop
And why test naked server? Who will ever use a Web server without a handler? There is TCP server for that

andig, 2016-01-27:
> And why test naked server? Who will ever use a Web server without a handler? There is TCP server for that

Nobody. Trying to limit the number of variables. Don't have better ideas without gdb :(

> you are printing from interrupt using HardwareSerial that uses interrupts. Many shit can go wrong
the other one you print from the loop

ok. I could test get rid of printing and leave the lib in.

andig, 2016-01-27:
btw, this would need update for latest git https://github.com/me-no-dev/ESPAsyncWebServer/blob/8fed4c42f34a4693b6529f051fbb6840d886a1b4/src/WebResponses.cpp#L362

andig, 2016-01-27:
> ok. I could test get rid of printing and leave the lib.

Still crashes :(

me-no-dev, 2016-01-27:
I'm trying to write a tool to decode the dumps for you (through the IDE tools).
Maybe we'll get a better idea of what and why it happens. But since it does not crash with the regular memory management, I doubt it's the server's fault really... I checked the code and it all goes as planned even if no handlers are attached. As I said, will return 500 :)
I updated the git for the changes in the core ;)

me-no-dev, 2016-01-27:
so I tested your sketch with umm_malloc and have no issues either :)
if you want to give it a shot, my git fork of the ESP repo is running it

andig, 2016-01-28:
> so I tested your sketch with umm_malloc and have no issues either :)

so its working with heap4 and working with umm_alloc, both with traffic, right?

> if you want to give it a shot, my git fork of the ESP repo is running it

the 31b or the arduino? main branch?

I'm wondering what other differences there could be. Maybe the toolchain on Windows has problems?

me-no-dev, 2016-01-28:
my ESP8266 Arduino branch is running on umm_malloc.
I did not try heap4 to be honest. Had already deleted the contents of the file :)

andig, 2016-01-28:
Ok. Crash with heap4 is this:

	0x4021dfe7: tcp_output at ??:?
	0x4021dfe7: tcp_output at ??:?
	0x4021c002: tcp_input at ??:?
	0x4021bc11: tcp_input at ??:?
	0x4021bc00: tcp_input at ??:?
	0x4021aa16: ip_input at ??:?
	0x402197ab: etharp_find_addr at ??:?
	0x40219d56: ethernet_input at ??:?
	0x4022570f: ets_snprintf at ??:?
	0x40205cf3: loop_task at C:\andi\arduino\hardware\eps8266com\esp8266\cores\esp8266/core_esp8266_main.cpp:43
	0x40000f49: ?? ??:0
	0x40000f49: ?? ??:0

As it does not crash with your umm* branch I'm closing as invalid. Thanks to the tool I can go back to the original sketch now and see how that behaves...

closed 2016-01-28 by andig