me-no-dev/ESPAsyncWebServer #15: Feature Request: make full use of lwip's buffer size

nouser2013, 2016-02-09:
Just for completeness, I'd like to request a feature to make AsyncServer better integrate with lwIP's stack functionality. Web browsers will immediately close a connection as soon as `Content-Length` bytes have been received.

Can we have AsyncServer send as much data as lwip `tcp_snd_buf` has currently room? Currently, the HTTP header is sent as separate packet, delaying Windows' ACK for this:
[image]
Also in the image above, the last two packets should have been sent in one go without waiting for the ACK... It would be very nice to have this for performance reasons since now a transfer takes 0.5s (with my setting and 5k HTML data) to Windows clients, which we could easily get down to under 100ms.

nouser2013, 2016-02-09:
Note: for callback requests this would probably go like so:

1. enable `tcp_nagle` (have LWIP manage send buffer and times)
1. compose header
1. `tcp_write()` composed header (no call to `tcp_output()`)
1. call tcp_buffer_free_size()
1. have callback fill a sendbuffer with so many bytes
1. call `tcp_write()` on filled sendbuffer 
1. `repeat until (callback return) < bufferMaxLen`

This should put everything in the hands of LWIP. hope i didnt say complete bs =)

me-no-dev, 2016-02-09:
will look into it. there are other ways to do it also ;) the basic response for example is already sending as much as it can, but the other responses were a bit different... so :) will see what I can do.
Other than that I send as much as the tcp tells me that it can take. That can be any number up to 2 packets (that is how much I have seen most).

me-no-dev, 2016-02-09:
I'll rework the way that the AbstractResonse sends data and that will fix it for all response types.

me-no-dev, 2016-04-08:
it's done :) sorry for the long wait

closed 2016-04-08 by me-no-dev