me-no-dev/ESPAsyncWebServer #19: use ESPAsyncWebServer  and websocket to stream video

mkeyno, 2016-04-16:
hi ,can I have your comments and opinion about   if ESP module can handle the video stream from browser to the module . I think Arducam made good library to stream the video to client browser but I don't know whether reverse action is possible or not , if it is what would be optimum fps and frame size

me-no-dev, 2016-04-16:
I'm confused by "from browser to the module"
do you mean to show the arducam image using the ESP as a server+socket?
I do not have the proper kind of arducam to test but the WebSocket lib has async implementation also :)
so I see no problem to be done.

You need the async branch of the WebSocket git and to change a define in the header.

mkeyno, 2016-04-16:
thanks for quick respond , I mean right now people test the [video stream from ESP+Arducam  to browser](https://github.com/luc-github/ESP8266/issues/82#issuecomment-210101801) and I'm just curious about reverse action and send video frame stream to the ESP so it can process it for example  store in the SD card

me-no-dev, 2016-04-27:
ok I have integrated a websocket plugin that can be used in that way and receive the data/image there :)
Check the latest changes and the WebSocket info towards the end of the README for more info on how to use it, with it you can accept any WebSocket message that conforms to the standard :)

mkeyno, 2016-04-28:
dear @me-no-dev do you have any example code ?can we resize the websocket buffer to video frame that we intended uploaded to ESP

me-no-dev, 2016-04-28:
the web socket in my implementation does not use a buffer to receive the data. You get every incoming packet and info, so you can put them back together and you know which one is first and last.
Works much like the file uploads :)

closed 2016-05-27 by me-no-dev