me-no-dev/ESPAsyncWebServer #20: Does Async TCP break HTTP Client?

kaeferfreund, 2016-04-23:
Hi,
seems to me that Async TCP breaks HTTP Client.

Is there a way to submit requests to a url? Since on the basic SyncClient this does not work in my case...

Apart from google-url, only IP based connections work with syncClient

Thanks in advance

me-no-dev, 2016-04-23:
how is the Async TCP breaking the HTTP Client? I'm a bit confused :) do you have any log or error that I can work with?
Also are you asking for a way to make a web request using just the SyncClient? What request do you want to send?

kaeferfreund, 2016-04-23:
sorry for the confusing post. Sometimes a little sleep helps with thinking (and tinkering too :D )
 The problem is not related to the async stuff. This works like a charm. Problem is in the current master branch of ESPArduino... Trying to figure it out now...

closed 2016-04-23 by kaeferfreund