me-no-dev/ESPAsyncWebServer #21: WS2812B RGB LED with NeoPixelBus Animation now flickering

hallard, 2016-05-02:
Hi @me-no-dev 
I've just migrated my project to AsyncLib, all is working fine and it's fast. I've got one question, I had some WS2812B led breathing and it was working fine with the classic WebServer (no async) but with the async version, LED are flickering, of course I understand Async does job under the hood but is there any option to slow down this kind of network "loop task" ? I just would like testing to see if it's due to some timing problem since these LED are very stric on this point.

By the way since we do nothing in loop nor call to async, how the TCP stuff is handled (interrupts, timer, ...), just curious  ?

me-no-dev, 2016-05-02:
the async lib is connected directly to the network stack and it gets it's data through callbacks. I do not think that those callbacks are interrupt based or we would not have had the need to call yield.
As for the WS2812B, what lib are you using and how is it pumping the data to the pixels? I2S, UART or bitbang?
The regular stuff (non-async) works much the same, but it's waiting for all the data to come instead of getting it as it comes, so maybe some yield is causing the network to spend time in a moment where the pixels should be getting data.

hallard, 2016-05-02:
Thanks, good to know about stack.
I'm using latest neopixelbus with bitbang800KHz for WS2812. I will investigate on this issue later, something to learn about Async since I'm using the whole, including, WebSocket, ArduinoJSON, SPIFFS, ....

By the way not sure if it's related but now calling WifI.scanNetworks() returns 0, did you noticed that point  ? (But I'm connecting to my AP without any problem)

```arduino
  int n = WiFi.scanNetworks();
  // Just to debug where we are
  Debugf("found %d networks!",n);
```
Another question, from WebSocket handler, how to get the client IP from AsyncWebSocketClient * client ?

me-no-dev, 2016-05-03:
yeah... need to add access to the client IP/Port...
as for WiFi.scanNetworks() do not call it inside async callbacks or run it in async mode and check for the result. If you are not calling it from there, then I have no clue

me-no-dev, 2016-05-03:
here is access to  IP/Port
https://github.com/me-no-dev/ESPAsyncWebServer/commit/8dfd3286277da1e51d2bd1dfa83b4852c00b250e

me-no-dev, 2016-05-03:
maybe you should try Uart or DMA mode for the pixels, since they have buffers to smooth things out :)

hallard, 2016-05-03:
yeah, but with GPIO of my project I can't use DMA or Uart.
But anyway, in fact LED are working fine (controlling them) it's just the animation (breathing) that seems impacted.
NeoPixel animation is run in main loop lot of time and I suspect Async task could sometime slow down main task and miss some breathing steps. I need to check this point, mainly the example I'm using is this one from NeoPixel Examples.
https://github.com/Makuna/NeoPixelBus/blob/master/examples/NeoPixelFunFadeInOut/NeoPixelFunFadeInOut.ino
I will merge basic AsyncWebServer into this sample just to see. My main project is so big and doing so much stuff that's it's difficult to isolate such kind of issues.
Anyway, thank you very much for you help

me-no-dev, 2016-05-03:
yes, I suspect the animation is yielding somewhere and that switches the context and causes delay in between commands

hallard, 2016-05-06:
@me-no-dev 
I've done some test, that's strange, on the main loop I'm calling a function so set builtin led and read on board switch (setting GPIO0 input, read it and setting back to output).
On board switch is on GPIO0 shared with WS2812, until async no flickering pb. 
I tried to start/stop animation when realing GPIO as also disable/enable interrupts, but nothing really worked.
Anyway, not really an issue I've commented out the code that was reading GPIO0 and now all works fine with RGB Led and AsyncWebServer
Not sure to understand what the source of the problem but always not really good idea to share pin

Here my code for information
```arduino
/* ======================================================================
Function: handle_key_led 
Purpose : Manage on board Led and Push button of NodeMCU
Input   : true to light on the led, false to light off
Output  : true if button is pressed
Comments: on NodeMCU BUILTIN_LED is GPIO16 and switch (Flash) on GPIO 0
====================================================================== */
uint8_t handle_key_led( uint8_t level)
{
  uint8_t btn=true;
  digitalWrite(BUILTIN_LED, 1);
  pinMode(BUILTIN_LED, OUTPUT);
  digitalWrite(BUILTIN_LED, !level);

  //animations.Pause();
  //noInterrupts(); // Need 100% focus on instruction timing
  // Flash is on GPIO0, so RGB LED
  //pinMode(0, INPUT);
  //btn = digitalRead(0);
  //pinMode(0, OUTPUT); // set back to output
  //interrupts();
  //animations.Resume();
  return !btn;
}

```

closed 2016-05-06 by hallard