me-no-dev/ESPAsyncWebServer #22: WebSocket Buffer Len

hallard, 2016-05-02:
When receiving WS_TEXT from websocket handler, if my text is for example foo the associated len received is 3. Does the buffer as 1 more size not counted by len ? because setting `data[len] = 0;` may have unpredictable effects ?

me-no-dev, 2016-05-03:
regardless of what you receive, the buffer that is passed is the underlying TCP buffer from lwip and I can not guarantee that it's 0 terminated :) this is all done to conserve as much memory as possible. You can always copy that to a buffer (like the web server does when handling it's requests).

hallard, 2016-05-03:
@me-no-dev 
thanks for the head up, yeah I understand, and I don't think the '\0' is there and I'm not sure there is place for it in the buffer. Best deal if we want to treat as string if to allocate a buffer of size (len+1) copy data and put the \0 at the end
Another is to do byte compare but not string compare, that can also be the deal ;-)

closed 2016-05-13 by me-no-dev