me-no-dev/ESPAsyncWebServer #27: serveStatic("/" ...) results in HTTP 500

mfalkvidd, 2016-05-08:
readme.md says:
```
  // attach filesystem root at URL /fs 
  server.serveStatic("/fs", SPIFFS, "/");
```
I wanted to serve the root uri from /html/ on SPIFFS, so I used
`  server.serveStatic("/", SPIFFS, "/html/");`
but this results in HTTP 500. The following syntax works though:
`  server.serveStatic("", SPIFFS, "/html/");`

vitotai, 2016-05-11:
I might have the same issue as you.
I solved it by modifying WebHandlers.cpp
String AsyncStaticWebHandler::_getPath(AsyncWebServerRequest *request)

    if (!baserequestUrl.length()) {
      baserequestUrl += "/";
    }

change to 
	
    if (!baserequestUrl.length()) {
    	if( _uri != "/")
      		baserequestUrl += "/";
    }

closed 2016-06-24 by me-no-dev

ccoenen, 2018-02-18:
If anyone else lands here with SPIFFS / serveStatic and HTTP 500 errors, one thing that was not immediately obvious to me: you have to call `SPIFFS.begin()` before you can server anything successfully.