me-no-dev/ESPAsyncWebServer #28: PROGMEM expamle with binary array

Zeron, 2016-05-09:
Example code here:  http://www.pastebin.ca/3595768

When using the PROGMEM example, it works great when the PROGMEM variable is declared like:
`const char DataText[] PROGMEM = R"=====(Blah blah Blah)=====";`

I have had no issue with it for data over 100Kb.

If however the PROGMEM variable lie like this:
`const char power_png[] PROGMEM = {
    0x89, 0x50, 0x4e, 0x47, 0x0d, 0x0a, 0x1a, 0x0a, 0x00, 0x00, 0x00, 0x0d, 
    0x49, 0x48, 0x44, 0x52, 0x00, 0x00, 0x00, 0xc8, 0x00, 0x00, 0x00, 0xc8
}
`
the example fails because strlen_P always returns 9

If I replace the strlen_P with sizeof, it will work some times, but take 5-7minutes to transfer.

Is there a way to be able to use PROGMEM arrays for binary data?

me-no-dev, 2016-05-10:
the problem with your progmem is that you have zeroes inside which is a string termination. You need to know beforehand what is the size of the progmem data and memcpy_P it to a normal uint8_t* buffer and send it that way. strlen will always return up to the first 0x00

me-no-dev, 2016-05-12:
```cpp
#define PNG_LEN 123456 //size of the progmem png in bytes
void GetPowerPng(AsyncWebServerRequest *request) {
  AsyncWebServerResponse *response = request->beginResponse(
    String("image/png"),
    PNG_LEN,
    [](uint8_t *buffer, size_t maxLen, size_t alreadySent) -> size_t {
      if ((PNG_LEN - alreadySent)>maxLen) {
        // We have more to read than fits in maxLen Buffer
        memcpy_P((char*)buffer, power_png+alreadySent, maxLen);
        return maxLen;
      }
      // Ok, last chunk
      memcpy_P((char*)buffer, power_png+alreadySent, (PNG_LEN - alreadySent));
      return (PNG_LEN - alreadySent); // Return from here to end of indexhtml
    }
  );
  request->send(response);  
}
```

Zeron, 2016-05-24:
No need to #define PNG_LEN as you can use sizeof to get the size.

```
void JavascriptHome(AsyncWebServerRequest *request) {
  
  AsyncWebServerResponse *response = request->beginResponse(
    String(F("text/javascript")),
    sizeof(home_js_gz),
    [](uint8_t *buffer, size_t maxLen, size_t alreadySent) -> size_t {

      size_t ReadAmount = 0;
      
      if ( (sizeof(home_js_gz) - alreadySent) > maxLen ) {
        // We have more to read than fits in maxLen Buffer
        ReadAmount = maxLen;
      } else {
        ReadAmount = sizeof(home_js_gz) - alreadySent;
      }

      memcpy_P((char*)buffer, home_js_gz + alreadySent, ReadAmount);
     
     return ReadAmount;
    }
  );
  response->addHeader(F("Content-Encoding"), F("gzip"));
  request->send(response);  
}
```

closed 2016-05-24 by Zeron