me-no-dev/ESPAsyncWebServer #29: System reset when trying to upload a large file with size 300k

vitotai, 2016-05-12:
I've tried for a while.
The example behaves exactly the same, while the synchronous ESP8266WebServer has no such issue.

me-no-dev, 2016-05-12:
that is caused by SPIFFS being filled and needing to wipe blocks in order to fit the new file.
when it hits that point, the process of rearranging and wiping blocks takes enough to trigger the watchdog and reset the sketch. In the regular web server you are running from the loop and the process yields, in async there is no such thing :)
if you upload a clean SPIFFS image to the flash, you will find the upload working fine (till you fill it up again by writing to it)

vitotai, 2016-05-12:
Thank you for your response.

I've tried to isolate the issue and used the following empty handleUpload() to test.
ESP8266 will hang or stop when the index went over 200,000. 
There is no useful output but a strange string "$T0b#e6".
I am trying to make GDBSTUB work, so I have no idea what's going on. 

```
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final)
{
    if(!index){ // initial start
        if(!_username.length() || request->authenticate(_username.c_str(),_password.c_str()))
        	_uploadAuthenticated = true;

        WiFiUDP::stopAll();
    	DEBUGF("Update: %s size:%d\n", filename.c_str());
        
    }
    if(_uploadAuthenticated && len){
//        ESP.wdtDisable(); request->_tempFile.write(data,len); ESP.wdtEnable(10);
		DEBUGF("Update:%d - %d\n",index,len);
    }
    if(_uploadAuthenticated && final){
        DEBUGF("Update End.\n");
	}
}

```

me-no-dev, 2016-05-12:
I'll give this exact code a go. Why do you stop UDP?

me-no-dev, 2016-05-12:
ok I can confirm some sort of issue
depending on the file, the upload hangs at the exact same progress for each file. Hmmmm

vitotai, 2016-05-12:
I was trying to port ESP8266HTTPUpdateServer and had this issue.
The code is copied from that porting. I wanted to make it clear so delete the commented-out code.

me-no-dev, 2016-05-12:
I'll figure it out :) interesting thing is that it was working fine for a long time and nothing related has changed recently...

me-no-dev, 2016-05-12:
please try the latest git :)

vitotai, 2016-05-13:
It's working perfectly now.
Thank you for your effort to create such a wonderful library.

closed 2016-05-13 by vitotai