me-no-dev/ESPAsyncWebServer #30: Can't build the example

hazarkarabay, 2016-05-13:
Hi,
I got following error when build the ESP_AsyncFSBrowser example.

I'm using PlatformIO 2.9.1, checked that Platformio ESP core == https://github.com/esp8266/Arduino master branch.

```
(penv) r:\test>pio run
[05/13/16 20:26:00] Processing d1_mini (platform: espressif, board: d1_mini, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------
xtensa-lx106-elf-ar rcs .pioenvs\d1_mini\libFrameworkArduinoVariant.a
xtensa-lx106-elf-g++ -o .pioenvs\d1_mini\src\tmp_ino_to.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -f
function-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=20200 -DP
LATFORMIO=020901 -U__STRICT_ANSI__ -IC:\users\hazar\.platformio\packages\framework-arduinoespressif\tools\sdk\include -IC:\users\hazar\.platformio\packages\fram
ework-arduinoespressif\tools\sdk\lwip\include -I.pioenvs\d1_mini\FrameworkArduino -I.pioenvs\d1_mini\FrameworkArduinoVariant -I.pioenvs\d1_mini\ESP8266WiFi -I.p
ioenvs\d1_mini\ESP8266mDNS -I.pioenvs\d1_mini\ArduinoOTA -I.pioenvs\d1_mini\Hash -I.pioenvs\d1_mini\ESPAsyncTCP_ID305 -I.pioenvs\d1_mini\ESPAsyncWebServer_ID306
 src\tmp_ino_to.cpp
xtensa-lx106-elf-g++ -o .pioenvs\d1_mini\FrameworkArduino\Esp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-function
s=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=202
00 -U__STRICT_ANSI__ -IC:\users\hazar\.platformio\packages\framework-arduinoespressif\tools\sdk\include -IC:\users\hazar\.platformio\packages\framework-arduinoe
spressif\tools\sdk\lwip\include -I.pioenvs\d1_mini\FrameworkArduino -I.pioenvs\d1_mini\FrameworkArduinoVariant .pioenvs\d1_mini\FrameworkArduino\Esp.cpp
xtensa-lx106-elf-g++ -o .pioenvs\d1_mini\FrameworkArduino\FS.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions
=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=2020
0 -U__STRICT_ANSI__ -IC:\users\hazar\.platformio\packages\framework-arduinoespressif\tools\sdk\include -IC:\users\hazar\.platformio\packages\framework-arduinoes
pressif\tools\sdk\lwip\include -I.pioenvs\d1_mini\FrameworkArduino -I.pioenvs\d1_mini\FrameworkArduinoVariant .pioenvs\d1_mini\FrameworkArduino\FS.cpp
xtensa-lx106-elf-g++ -o .pioenvs\d1_mini\FrameworkArduino\HardwareSerial.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -fali
gn-functions=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -D
ARDUINO=20200 -U__STRICT_ANSI__ -IC:\users\hazar\.platformio\packages\framework-arduinoespressif\tools\sdk\include -IC:\users\hazar\.platformio\packages\framewo
rk-arduinoespressif\tools\sdk\lwip\include -I.pioenvs\d1_mini\FrameworkArduino -I.pioenvs\d1_mini\FrameworkArduinoVariant .pioenvs\d1_mini\FrameworkArduino\Hard
wareSerial.cpp
xtensa-lx106-elf-g++ -o .pioenvs\d1_mini\FrameworkArduino\IPAddress.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-fu
nctions=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUI
NO=20200 -U__STRICT_ANSI__ -IC:\users\hazar\.platformio\packages\framework-arduinoespressif\tools\sdk\include -IC:\users\hazar\.platformio\packages\framework-ar
duinoespressif\tools\sdk\lwip\include -I.pioenvs\d1_mini\FrameworkArduino -I.pioenvs\d1_mini\FrameworkArduinoVariant .pioenvs\d1_mini\FrameworkArduino\IPAddress
.cpp
xtensa-lx106-elf-g++ -o .pioenvs\d1_mini\FrameworkArduino\MD5Builder.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-f
unctions=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDU
INO=20200 -U__STRICT_ANSI__ -IC:\users\hazar\.platformio\packages\framework-arduinoespressif\tools\sdk\include -IC:\users\hazar\.platformio\packages\framework-a
rduinoespressif\tools\sdk\lwip\include -I.pioenvs\d1_mini\FrameworkArduino -I.pioenvs\d1_mini\FrameworkArduinoVariant .pioenvs\d1_mini\FrameworkArduino\MD5Build
er.cpp
xtensa-lx106-elf-g++ -o .pioenvs\d1_mini\FrameworkArduino\Print.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functi
ons=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=2
0200 -U__STRICT_ANSI__ -IC:\users\hazar\.platformio\packages\framework-arduinoespressif\tools\sdk\include -IC:\users\hazar\.platformio\packages\framework-arduin
oespressif\tools\sdk\lwip\include -I.pioenvs\d1_mini\FrameworkArduino -I.pioenvs\d1_mini\FrameworkArduinoVariant .pioenvs\d1_mini\FrameworkArduino\Print.cpp
xtensa-lx106-elf-g++ -o .pioenvs\d1_mini\FrameworkArduino\Stream.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-funct
ions=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=
20200 -U__STRICT_ANSI__ -IC:\users\hazar\.platformio\packages\framework-arduinoespressif\tools\sdk\include -IC:\users\hazar\.platformio\packages\framework-ardui
noespressif\tools\sdk\lwip\include -I.pioenvs\d1_mini\FrameworkArduino -I.pioenvs\d1_mini\FrameworkArduinoVariant .pioenvs\d1_mini\FrameworkArduino\Stream.cpp
xtensa-lx106-elf-g++ -o .pioenvs\d1_mini\FrameworkArduino\StreamString.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign
-functions=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DAR
DUINO=20200 -U__STRICT_ANSI__ -IC:\users\hazar\.platformio\packages\framework-arduinoespressif\tools\sdk\include -IC:\users\hazar\.platformio\packages\framework
-arduinoespressif\tools\sdk\lwip\include -I.pioenvs\d1_mini\FrameworkArduino -I.pioenvs\d1_mini\FrameworkArduinoVariant .pioenvs\d1_mini\FrameworkArduino\Stream
String.cpp
R:/test/src/ESP_AsyncFSBrowser.ino:18:6: error: 'bool SPIFFSEditor::canHandle(AsyncWebServerRequest*)' cannot be overloaded
if(request->method() == HTTP_GET && request->url() == "/edit" && (SPIFFS.exists("/edit.htm") || SPIFFS.exists("/edit.htm.gz")))
^
R:/test/src/ESP_AsyncFSBrowser.ino:17:10: error: with 'bool SPIFFSEditor::canHandle(AsyncWebServerRequest*)'
bool canHandle(AsyncWebServerRequest *request){
^
R:/test/src/ESP_AsyncFSBrowser.ino:34:10: error: 'void SPIFFSEditor::handleRequest(AsyncWebServerRequest*)' cannot be overloaded
if(_username.length() && (request->method() != HTTP_GET || request->url() == "/edit" || request->url() == "/list") && !request->authenticate(_username.c_str(),_
password.c_str()))
^
R:/test/src/ESP_AsyncFSBrowser.ino:19:10: error: with 'void SPIFFSEditor::handleRequest(AsyncWebServerRequest*)'
return true;
^
R:/test/src/ESP_AsyncFSBrowser.ino:99:10: error: 'void SPIFFSEditor::handleUpload(AsyncWebServerRequest*, String, size_t, uint8_t*, size_t, bool)' cannot be ove
rloaded
if(!index){
^
R:/test/src/ESP_AsyncFSBrowser.ino:21:10: error: with 'void SPIFFSEditor::handleUpload(AsyncWebServerRequest*, String, size_t, uint8_t*, size_t, bool)'
return true;
^
scons: *** [.pioenvs\d1_mini\src\tmp_ino_to.o] Error 1
xtensa-lx106-elf-g++ -o .pioenvs\d1_mini\FrameworkArduino\Tone.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functio
ns=4 -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=20
200 -U__STRICT_ANSI__ -IC:\users\hazar\.platformio\packages\framework-arduinoespressif\tools\sdk\include -IC:\users\hazar\.platformio\packages\framework-arduino
espressif\tools\sdk\lwip\include -I.pioenvs\d1_mini\FrameworkArduino -I.pioenvs\d1_mini\FrameworkArduinoVariant .pioenvs\d1_mini\FrameworkArduino\Tone.cpp
================================================================= [ ERROR ] Took 1.71 seconds =================================================================

me-no-dev, 2016-05-13:
I am personally not a platformio user but have many such users and I know the lib compiles fine.
What is really strange though is the errors you are getting:
```cpp
error: 'bool SPIFFSEditor::canHandle(AsyncWebServerRequest*)' cannot be overloaded with 'bool SPIFFSEditor::canHandle(AsyncWebServerRequest*)'
```
which makes no sense at all!
there is only one such method in the sketch :) how can it try to overwrite itself?
maybe there is something else wrong?
@ivankravets can maybe shed some light?

hazarkarabay, 2016-05-14:
I can confirm that example compiles fine in freshly installed Arduino 1.6.9. 

Sorry for taking your time.

closed 2016-05-14 by hazarkarabay

me-no-dev, 2016-05-14:
it's good you did :) if someone else comes with something like this, we will see a pattern and will be able to act better on it ;)

ivankravets, 2016-05-16:
I'll take a look on it. This is a problem with converting INO -> CPP.

@hazarkarabay please rename `ESP_AsyncFSBrowser.ino` to `ESP_AsyncFSBrowser.cpp`, append `#include <Arduino.h>` on the top of file, declare prototypes and try with PlatformIO.