me-no-dev/ESPAsyncWebServer #32: AsyncWebSocket memory leak?

hazarkarabay, 2016-05-15:
Each time I connected then disconnected to the Websocket server, ESP.getFreeHeap() decreases by 184.

Example code: ESPAsyncWebServer/examples/ESP_AsyncFSBrowser/ESP_AsyncFSBrowser.ino with one modification to demonstrate problem:

Line 121: 
Change 
`client->printf("Hello Client %u :)", client->id());`

To
`client->printf("Hello Client %u :), %u", client->id(), ESP.getFreeHeap());`

Websocket client log:
```
CONNECTED

RESPONSE: Hello Client 1 :), 38696

DISCONNECTED

CONNECTED

RESPONSE: Hello Client 2 :), 38512

DISCONNECTED

CONNECTED

RESPONSE: Hello Client 3 :), 38328

DISCONNECTED

...
```

me-no-dev, 2016-05-16:
totally possible... though I did take special care not to miss to free any object, but :D

me-no-dev, 2016-05-19:
I did some tests, the heap comes back after a minute, which means that the tcp is not cleared right away... will see why

me-no-dev, 2016-05-27:
I spent one night hunting this leak. I went back and fourth, measured sizes of objects, did tcpdump to make sure connections are properly cleared... and so far nothing...
It is something in the network stack, but do not know what.... no combination of object sizes can amount to 184 bytes also.
Will post more when I know a bit more.

me-no-dev, 2016-05-28:
fixed :)
https://github.com/me-no-dev/ESPAsyncWebServer/commit/bda2cd637a292bad736c719bdf057736aadde493

closed 2016-05-28 by me-no-dev