me-no-dev/ESPAsyncWebServer #34: sync connectin with GPRS

mkeyno, 2016-05-29:
hi @me-no-dev ,sorry for trigger issue here but actually , I'm trying to establish websocketclient from GPRS  connection to update remote website data, unfortunately Markus not available today's  , and that's why I put my problem here, I find only [this code in this github](https://github.com/adik/webgsmcontrol/blob/master/src/main.cpp) , which is no suitable for my module (SIM800L), you may know we only have AT command to control the GSM module whereas I found only one command to change HTTP parameters (AT+HTTPPARA="CONTENT",<HTTPParamValue>), is it possible to run websocket connection with just this command?
can you guide me to write the simple websocket client ?

thanks

me-no-dev, 2016-05-29:
could you please explain better what you are trying to achieve as final result?
you want to have a persistent TCP connection to a server and pipe that data through the web socket? Or you want to make a transparent bridge to the SIM800L?

me-no-dev, 2016-05-29:
also if you can jump on this repo's gitter chat, we can speed up the help process much :)

mkeyno, 2016-05-29:
thanks for quick respond , actually I don't know which way is best , I want establish somethings like websocket client, so I could on-event send my new data to server.
 you may aware about Arduino GPRS module which is useful for remote control and monitor , but those module work on AT commands and so   unreliable , as it mentioned it only have one [AT command to change HTTP parameters](http://www.seeedstudio.com/wiki/images/5/58/SIM800_Series_AT_Command_Manual_V1.01.pdf) (AT+HTTPPARA="CONTENT",), but @Adik somehow upgraded HTTP connection to the websocket service 
can you check his code and guide me how can do the same
my best regards

closed 2016-09-23 by me-no-dev