me-no-dev/ESPAsyncWebServer #35: Update firmware over the FILE Upload handling.

EduardUA, 2016-06-07:
I have a problem with updating firmware.

`void setup()
{server.on("/upload", HTTP_POST, [](AsyncWebServerRequest *request){
		request->send(200);
	}, handleUpload);
}`

me-no-dev, 2016-06-07:
This is the basic of it, but you need the latest git version of the ESP core as well (Updater.runAsync was added yesterday).
```cpp
static void handle_update_progress_cb(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
  uint32_t free_space = (system_free_sketch_space() - 0x1000) & 0xFFFFF000;
  if (!index){
    Updater.runAsync(true);
    if (!Update.begin(free_space)) {
      Update.printError(Serial);
    }
  }

  if (Update.write(data, len) != len) {
    Update.printError(Serial);
  }

  if (final) {
    if (!Update.end(true)){
      Update.printError(Serial);
    }
  }
}
```

closed 2016-06-07 by EduardUA

tretyakovsa, 2017-10-02:
Please show the full example of the sketch for the update!

klaasdc, 2017-11-06:
Here is my complete working example, in case it is useful for someone:

```arduino
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

char ssid[] = "YOURSSID";
char pass[] = "YOURPASS";

WiFiClient client;

const char upload_page[] PROGMEM = R"=====(
<!DOCTYPE HTML>
<HTML>
  <HEAD>
    <TITLE>Firmware upload example</TITLE>
  </HEAD>
  <BODY>
    <H1>Choose .ino.bin file</H1>
    <form id="uploadbanner" enctype="multipart/form-data" method="post" action="/upload">
       <input id="fileupload" name="myfile" type="file" />
       <input type="submit" value="submit" id="submit" />
    </form>
  </BODY>
</HTML>
)=====";

void wifiInit(){
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, pass);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void indexrequest(AsyncWebServerRequest *request){
  request->send_P(200, "text/html", upload_page);
}

static int restartNow = false;

static void handle_update_progress_cb(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
  uint32_t free_space = (ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000;
  if (!index){
    Serial.println("Update");
    Update.runAsync(true);
    if (!Update.begin(free_space)) {
      Update.printError(Serial);
    }
  }

  if (Update.write(data, len) != len) {
    Update.printError(Serial);
  }

  if (final) {
    if (!Update.end(true)){
      Update.printError(Serial);
    } else {
      restartNow = true;//Set flag so main loop can issue restart call
      Serial.println("Update complete");
    }
  }
}

void serverInit(){
  server.on("/", HTTP_GET, indexrequest);

  // handler for the /update form POST (once file upload finishes)
  server.on("/upload", HTTP_POST, [](AsyncWebServerRequest *request){
      request->send(200);
    }, handle_update_progress_cb);

  server.onNotFound([](AsyncWebServerRequest *request){
    request->send(404);
  });
  server.begin();
}

void setup() {
  Serial.begin(115200);
  wifiInit();
  serverInit();
}

void loop() {
  if (restartNow){
    Serial.println("Restart");
    ESP.restart();
  }
  yield();
}
```

MkLHX, 2019-04-26:
Hi @klaasdc 
Thanks for shared code. But @ this time Update.runAsync() is not callable on platformio project 

Here my code
`
#include "Arduino.h"
#include "WiFi.h"
#include "ESPAsyncWebServer.h"
#include "Update.h"

AsyncWebServer server_http(80);

String updateView = "<h1>Choose .ino or .bin file for update</h1>"
					"<form id='form' method='POST' action='/process' enctype='multipart/form-data'>"
					"<input type='file' name='file' id='file'>"
					"<br><input type='submit' class='btn' value='Update It'></form>";

void updateRequest(AsyncWebServerRequest *request){
    request->send(200,"text/html",updateView);
}

static int restartNow = false;
void httpProcessUpdate(AsyncWebServerRequest *request, const String &filename, size_t index, uint8_t *data, size_t len, bool final)
{
	uint32_t free_space = (ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000;
	if (!index)
	{
		Serial.println("process update");
		Update.runAsync(true);
		if (!Update.begin(free_space))
		{
			Update.printError(Serial);
		}
		// Serial.printf("UploadStart: %s\n", filename.c_str());
	}

	if (Update.write(data, len) != len)
	{
		Update.printError(Serial);
	}

	if (final)
	{
		if (!Update.end(true))
		{
			Update.printError(Serial);
		}
		else
		{
			restartNow = true; //Set flag so main loop can issue restart call
			Serial.println("Update complete");
		}
	}
}
void setup(){
        Serial.begin(115200);
	//WIFI AP
	int timeout = 0;
	while (!WiFi.softAP("myesp32", "12345678"))
	{
		Serial.println(F("."));
		delay(250);
		if (timeout > 40) //delay 250ms so 10s = 40*250ms
		{
			Serial.println(F("[Wifi AP]... not start"));
		}
		timeout++;
	};
	Serial.println(F("[Wifi AP]... started"));
	Serial.println("[Wifi AP]... ip : " + WiFi.softAPIP().toString());

	//HTTP SERVER
	server_http.on("/", HTTP_GET, updateRequest);
	server_http.on("/process", HTTP_POST, [](AsyncWebServerRequest *request) {
		request->send(200);
	}, httpProcessUpdate);
	server_http.begin();
}
void loop(){
	if (restartNow)
	{
		Serial.println("Restart");
		ESP.restart();
	}
	yield();
}`

GeorgeFlorian, 2019-07-02:
Did you manage to make this work on ESP32 ?

GeorgeFlorian, 2019-07-02:
> Hi @klaasdc
> Thanks for shared code. But @ this time Update.runAsync() is not callable on platformio project
> 
> i'm looking for a way to update firmware in AP mode only directly connected to the ESP32 AP
> 
> Here my code
> 
> ```
> #include "Arduino.h"
> #include "WiFi.h"
> #include "ESPAsyncWebServer.h"
> #include "Update.h"
> 
> AsyncWebServer server_http(80);
> 
> String updateView = "<h1>Choose .ino or .bin file for update</h1>"
> 				"<form id='form' method='POST' action='/process' enctype='multipart/form-data'>"
> 				"<input type='file' name='file' id='file'>"
> 				"<br><input type='submit' class='btn' value='Update It'></form>";
> 
> void updateRequest(AsyncWebServerRequest *request){
>     request->send(200,"text/html",updateView);
> }
> 
> static int restartNow = false;
> void httpProcessUpdate(AsyncWebServerRequest *request, const String &filename, size_t index, uint8_t *data, size_t len, bool final)
> {
> 	uint32_t free_space = (ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000;
> 	if (!index)
> 	{
> 		Serial.println("process update");
> 		Update.runAsync(true);
> 		if (!Update.begin(free_space))
> 		{
> 			Update.printError(Serial);
> 		}
> 		// Serial.printf("UploadStart: %s\n", filename.c_str());
> 	}
> 
> 	if (Update.write(data, len) != len)
> 	{
> 		Update.printError(Serial);
> 	}
> 
> 	if (final)
> 	{
> 		if (!Update.end(true))
> 		{
> 			Update.printError(Serial);
> 		}
> 		else
> 		{
> 			restartNow = true; //Set flag so main loop can issue restart call
> 			Serial.println("Update complete");
> 		}
> 	}
> }
> void setup(){
>         Serial.begin(115200);
> 	//WIFI AP
> 	int timeout = 0;
> 	while (!WiFi.softAP("myesp32", "12345678"))
> 	{
> 		Serial.println(F("."));
> 		delay(250);
> 		if (timeout > 40) //delay 250ms so 10s = 40*250ms
> 		{
> 			Serial.println(F("[Wifi AP]... not start"));
> 		}
> 		timeout++;
> 	};
> 	Serial.println(F("[Wifi AP]... started"));
> 	Serial.println("[Wifi AP]... ip : " + WiFi.softAPIP().toString());
> 
> 	//HTTP SERVER
> 	server_http.on("/", HTTP_GET, updateRequest);
> 	server_http.on("/process", HTTP_POST, [](AsyncWebServerRequest *request) {
> 		request->send(200);
> 	}, httpProcessUpdate);
> 	server_http.begin();
> }
> void loop(){
> 	if (restartNow)
> 	{
> 		Serial.println("Restart");
> 		ESP.restart();
> 	}
> 	yield();
> }
> ```

Did you manage to make this work on the ESP32 ?

MkLHX, 2019-07-02:
hi @GeorgeFlorian, yep i manage it!
https://gist.github.com/MkLHX/295e79a407db6a553085279f9548954b

GeorgeFlorian, 2019-07-02:
> hi @GeorgeFlorian, yep i manage it!
> https://gist.github.com/MkLHX/295e79a407db6a553085279f9548954b

Great ! Thank you !
With your example and @Ibernstone 's [example](https://github.com/lbernstone/asyncUpdate/blob/master/AsyncUpdate.ino) I will surely be able to make my own WEB OTA.

GeorgeFlorian, 2019-07-02:
> hi @GeorgeFlorian, yep i manage it!
> https://gist.github.com/MkLHX/295e79a407db6a553085279f9548954b

> hi @GeorgeFlorian, yep i manage it!
> https://gist.github.com/MkLHX/295e79a407db6a553085279f9548954b

Great ! Thank you !
With your example and @Ibernstone 's [example](https://github.com/lbernstone/asyncUpdate/blob/master/AsyncUpdate.ino) I will surely be able to make my own WEB OTA.

Also, what kind of files do you upload to perform the update ? What extension do they need to have ? Will a simple .cpp suffice ?

MkLHX, 2019-07-02:
> Also, what kind of files do you upload to perform the update ? What extension do they need to have ? Will a simple .cpp suffice ?

nope ! you have to send .bin file

GeorgeFlorian, 2019-07-02:
> > Also, what kind of files do you upload to perform the update ? What extension do they need to have ? Will a simple .cpp suffice ?
> 
> nope ! you have to send .bin file

Wait, what ? Isn't this a method to update the code that's on the ESP32 ? There seems something that I don't get.
How do I go from a .cpp to a .bin file ?

Pablo2048, 2019-07-02:
In Arduino IDE menu -> Sketch -> Export Compiled Binary (https://arduino.stackexchange.com/questions/48431/how-to-get-the-firmware-hex-file-from-a-ino-file-containing-the-code/48564) - Use the Google, Luke...

MkLHX, 2019-07-02:
What is your dev env?
you know, .cpp or .ino files are compiled into a .bin or .hex files to run firmware flash.

For example in platformio when you launch "pio run"command, all files are parsed and compiled to make .bin file and then .bin file is flash in esp32.

The ESP32 doesn't do anything about compiling his firmware.

GeorgeFlorian, 2019-07-02:
> What is your dev env?
> you know, .cpp or .ino files are compiled into a .bin or .hex files to run firmware flash.
>
> For example in platformio when you launch "pio run"command, all files are parsed and [...]

I am using Platformio IDE on VSCode.
I have never used commands though.
I have only built and uploaded the code, but it always stayed in .cpp form.

> In Arduino IDE menu -> Sketch -> Export Compiled Binary (https://arduino.stackexchange.com/questions/48431/how-to-get-the-firmware-hex-file-from-a-ino-file-containing-the-code/48564) - Use the Google, Luke...

Thank you, mate !
I am using Google. I have google how to update firmware using ESP32 and I ended up here, where I found a man willing to help me and point me in the right direction. And while I am conversing with you both on this thread I am also adapting the OTA code to my suit my project and searching on Google on how to transform .cpp into .bin.

MkLHX, 2019-07-02:
So you are looking for OTA Update or  to Upload firmware through handle file from webserver?
This is not the same thing....
Here we talk about update through webserver page with html form like this: 

> https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/OTAWebUpdater/OTAWebUpdater.ino

To perform OTA update through your network you need do this thing :

> https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino

Pablo2048, 2019-07-02:
Ok, so if you are using PIO you have to add extra script to your platformio.ini. This script can (for example) look like this:
```python
Import("env")
import shutil
import os
#
# Dump build environment (for debug)
#print env.Dump()
#

#
# Upload actions
#

def after_build(source, target, env):
  shutil.copy(firmware_source, 'bin/firmware.bin')
  # env.Execute("pio run -t buildfs")
  # shutil.copy(spiffs_source, 'bin/spiffs.bin')

env.AddPostAction("buildprog", after_build)

firmware_source = os.path.join(env.subst("$BUILD_DIR"), "firmware.bin")
spiffs_source = os.path.join(env.subst("$BUILD_DIR"), "spiffs.bin")
```
I'm using this with my ESP8266 PIO target, but I didn't see why this shouldn't work for ESP32 too...

MkLHX, 2019-07-02:
> I am using Platformio IDE on VSCode.
Which framework are you use?

GeorgeFlorian, 2019-07-02:
> So you are looking for OTA Update or to Upload firmware through handle file from webserver?
> This is not the same thing....
> Here we talk about update through webserver page with html form like [...]

Shouldn't both methods end up with the same result: updating/changing the code that's running on the ESP32 ?
OTA Update is made by using Arduino IDE, right ?
I want the end-user to be able to update the code using a web-page and a provided file.

> > I am using Platformio IDE on VSCode.
> 
> Which framework are you use?

```
[env:esp32-evb]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32-evb
framework = arduino
board_build.flash_mode = qio
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
monitor_speed = 115200
upload_speed = 115200
build_flags = -DCORE_DEBUG_LEVEL=5

#libraries
lib_deps =  https://github.com/me-no-dev/ESPAsyncWebServer.git
            https://github.com/me-no-dev/AsyncTCP.git
```

Pablo2048, 2019-07-02:
Try to add this line into your platformio.ini
```
extra_scripts = post:scripts/pio_copy.py
```
save my Python code from my previous message to scripts/pio_copy.py and try to compile your sketch. After compiling there has to be bin/ directory in your project directory with firmware.bin file...

MkLHX, 2019-07-02:
The 2 methods have the same result: update the esp32 firmware. But they have to different workflow.

With OTA, you can run firmware update platformio IDE like arduino IDE by using this command
```pio run -t upload --upload-port 192.168.0.2 //look your esp32 network ip adress```
(you need to add OTA code and lib to succes it)

if you want the end-user perform update through web view by uploading file through html form you need to give the .bin file access for the end-user, or you need to give access to the source-code and the right to perform firmware build by run command ```pio run``` or something like this.

GeorgeFlorian, 2019-07-02:
> Try to add this line into your platformio.ini
> 
> ```
> extra_scripts = post:scripts/pio_copy.py
> ```
> 
> save my Python code from my previous message to scripts/pio_copy.py and try to compile your sketch. After compiling there has to be bin/ directory in your project directory with firmware.bin file...

So I have added `extra_scripts = post:scripts/pio_copy.py` to the end of the platformio.ini file.
Then I've made `scripts` directory inside my project's directory and there I've made a file named `pio_copy.py` and I've copied :
```
Import("env")
import shutil
import os
#
# Dump build environment (for debug)
#print env.Dump()
#

#
# Upload actions
#

def after_build(source, target, env):
  shutil.copy(firmware_source, 'bin/firmware.bin')
  # env.Execute("pio run -t buildfs")
  # shutil.copy(spiffs_source, 'bin/spiffs.bin')

env.AddPostAction("buildprog", after_build)

firmware_source = os.path.join(env.subst("$BUILD_DIR"), "firmware.bin")
spiffs_source = os.path.join(env.subst("$BUILD_DIR"), "spiffs.bin")
```
Then I've hit `CTRL+ALT+B` which means `Platformio: Build` and it returns an error:
```
*** [buildprog] bin/firmware.bin: No such file or directory
```

So I've created a `bin` directory inside my project's directory and now it doesn't return an error, but it doesn't make a .bin file either.

MkLHX, 2019-07-02:
on platformio project, you can find the .bin file into
(i'm on windows 10)
```your-project-path\project-folder\.pioenvs\you-env-board-in-your-case: esp32-evb```
the firmware.bin is the compiled firmware you have to upload through html form.

Pablo2048, 2019-07-02:
I've tested this just now and it works. My platformi.ini looks like:
```ini
; PlatformIO Project Configuration File

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
  http://git.xpablo.cz/xPablo.cz/encipheredEEPROMStrings.git
  http://git.xpablo.cz/pablo2048/Interval.git
  https://github.com/me-no-dev/AsyncTCP.git
  https://github.com/me-no-dev/ESPAsyncWebServer.git
  http://git.xpablo.cz/pablo2048/Trace.git
  http://git.xpablo.cz/pablo2048/WiFiConfig.git
framework = arduino
upload_port = esp32.local
;upload_port=192.168.4.1
upload_speed = 230400
;board_build.partitions = xpablo.csv
;board_build.partitions = xpablo_noEEPROM.csv
build_flags =
  ;-fexceptions
  ;-D CORE_DEBUG_LEVEL=0
  -D DEBUG_ESP_PORT=Serial
  -D BOARD_HAS_PSRAM
  -mfix-esp32-psram-cache-issue
  -D CORS_DEBUG
  -D USE_OTA
  -D USE_TRACE
  -D FORMAT_SPIFFS_WHEN_FAILED=true
  -D USE_WIFICONFIG
  -D USE_NTP
  ;-D USE_PROFILING
  -D USE_EDITOR
  -D USE_HTTPUPDATE
extra_scripts = post:scripts/pio_copy.py
monitor_speed=115200
```

GeorgeFlorian, 2019-07-02:
> on platformio project, you can find the .bin file into
> (i'm on windows 10)
> `your-project-path\project-folder\.pioenvs\you-env-board-in-your-case: esp32-evb`
> the firmware.bin is the compiled firmware you have to upload through html form.

So, it makes a `.bin` file without all the scripts from @Pablo2048 ?

Pablo2048, 2019-07-02:
:-) My script just copies this file to non-hidden directory...

MkLHX, 2019-07-02:
yep

GeorgeFlorian, 2019-07-02:
> :-) My script just copies this file to non-hidden directory...

I am sorry, I am a moron with 0 knowledge of Python.

Build sometimes works, sometimes doesn't. It doesn't always produces a `firmware.bin`

Also, I have some questions related to the firmware.bin file : 
 * I have a lot of HTML files inside the ESP32's flash memory, that are being served by the server using SPIFFS. By uploading a `firmware.bin` will those file be deleted ?
 * And what if I want to upload some files that are inside the flash memory ? Are those also compiled inside firmware.bin ?

Pablo2048, 2019-07-02:
1. No - SPIFFS should not be affected with OTA (at least until you didn't change partitions mapping)
2. Firmware.bin is just the firmware - AFAIK you can upload whole spiffs image, but I didn't try this by myself, anyway my script preserves (and copies) spiffs.bin image if you run
```
pio run --target buildfs
```
IIRC there is some guidance on how to upload binary files into ESP32 somewhere in PIO documentation...

GeorgeFlorian, 2019-07-02:
> pio run --target buildfs

`spiffs.bin` sounds promising. Unfortunately, I can't seem to make your script work.
Also, I have no idea what to do with this: `pio run --target buildfs`.

I am amazed of how much I don't know, and how lacking is the documentation.

Pablo2048, 2019-07-02:
Ok - you have using VSCode so click on the terminal icon in Platformio toolbar ("PlatformIO: New Terminal") and then just type ```pio run --target buildfs``` - after finish there has to be spiffs.bin file inside ```pionevs\esXXX\``` directory. It expects that your SPIFFS files are in ```data``` subdirectory.

GeorgeFlorian, 2019-07-02:
> k - you are using VSCode so click on the terminal icon in Platformio toolbar ("PlatformIO: New Terminal") and then just type pio run --target buildfs

That does nothing. It doesn't even work.

Actually, I don't have `spiffs.bin`. Not even in `/.pioenvs/`. Huh... what about that ?

I am using Arduino IDE to upload files. Maybe it's because of that ?

Pablo2048, 2019-07-02:
Can you please post screenshot what happen when you click on the terminal icon?
You don't have to use Arduino IDE - even PIO can upload SPIFFS with command ```pio run --target uploadfs``` :-)

GeorgeFlorian, 2019-07-02:
> Can you please post screenshot what happen when you click on the terminal icon?
> You don't have to use Arduino IDE - even PIO can upload SPIFFS with command `pio run --target uploadfs` :-)

By using Arduino IDE to upload files I did not have a spiffs.bin.
Now I have made a data directory inside the project's directory and now it also builds a SPIFFS Image.

My god, this feels good. I am genuinely happy right now. xD

GeorgeFlorian, 2019-07-02:
Soo, returning with a quick update: uploading `spiffs.bin` returns a lot of `Wrong Magic Byte`.

```
void handleDoUpdate(AsyncWebServerRequest *request, const String& filename, size_t index, uint8_t *data, size_t len, bool final) {
  if (!index){
    log_i("Update");
    // if filename includes spiffs, update the spiffs partition
    int cmd = (filename.indexOf("spiffs") > 0) ? U_SPIFFS : U_FLASH;
    if (!Update.begin(UPDATE_SIZE_UNKNOWN, cmd)) {
      Update.printError(Serial);
    }
  }

  if (Update.write(data, len) != len) {
    Update.printError(Serial);
  }

  if (final) {
    AsyncWebServerResponse *response = request->beginResponse(302, "text/plain", "Please wait while the switch reboots");
    response->addHeader("Refresh", "20");  
    response->addHeader("Location", "/");
    request->send(response);
    if (!Update.end(true)){
      Update.printError(Serial);
    } else {
      Serial.println("Update complete");
      Serial.flush();
      ESP.restart();
    }
  }
}
```
I have followed @lbernstone 's [example](https://github.com/lbernstone/asyncUpdate/blob/master/AsyncUpdate.ino)

Pablo2048, 2019-07-03:
Well, hard to guess - I'll do some tests at the weekend (if I get some spare time :-) )...