me-no-dev/ESPAsyncWebServer #39: Get rid of urldecode

andig, 2016-06-15:
With https://github.com/esp8266/Arduino/commit/35fd2ccd52de7d6a8076ad2eea05bf87a804e6d1 it should be possible to use the static function of the sync web server?

me-no-dev, 2016-06-15:
I don't think it's a good idea to hook to the regular server at all :) why should we do it for one method? If the method was exposed like digitalRead directly from the framework, then yes, but no when it's on another web server (that might get replaced anyway)

andig, 2016-06-15:
It just got exposed so likely wont go away soon. But you're right. Would it make sense though to add en urlencode which people would very likely need?

me-no-dev, 2016-06-15:
If you ask me, urlEncode/Decode should get exposed like the rest of the low level functions of arduino and be used by all equally :) but yeah, we will need it for when I start on the AsyncHTTP/WS Client

andig, 2016-06-15:
:+1: much appreciated. Closing for now.

closed 2016-06-15 by andig