me-no-dev/ESPAsyncWebServer #46: Missing file "Hash.h"

kuthullu, 2016-06-18:
At line 40 on file AsyncWebSocket.cpp, there's an include of file "Hash.h" that isn't present on the library, and isn't on it's pre-requisites. Compiling the example "as-is" generates error.
- Using IDE PlataformIO with framework Arduino and plataform Expressif, on board NodeMCU V1.0

me-no-dev, 2016-06-18:
that is platform.io's fault. the library is in the ESP core libs.
One fix is to add ```#include <Hash.h>``` in AsyncWebSocket.h

kuthullu, 2016-06-18:
I searched the Lins directory and the generated libs for the project, the
file doesn't exist...

me-no-dev, 2016-06-18:
https://github.com/esp8266/Arduino/blob/master/libraries/Hash/src/Hash.h

kuthullu, 2016-06-18:
It isn't found by default, and even

kuthullu, 2016-06-18:
It isn't included by default, but explicitly including it on your main.cpp solves the problem... maybe this is worth a note on the docs ? Anyway, thanks for your time, you really helped me

me-no-dev, 2016-06-18:
it should really be fixed on platform.io side :) it follows Arduino guidelines and should by all means work.
@ivankravets can maybe shed some light?

ivankravets, 2016-06-18:
A deep library search is turned off by default. We had it enabled before but a lot of users had the problems with it. The easy way is to include required library in the `src` files. In this case, PlatformIO will understand that you really need this library, otherwise you can enable http://docs.platformio.org/en/latest/projectconf.html#lib-dfcyclic

ivankravets, 2016-06-18:
@kuthullu just add this line into the of one file from `src` folder
```cpp
#include <Hash.h>
```

me-no-dev, 2016-06-18:
@ivankravets has to be in one of the headers, correct?

me-no-dev, 2016-06-18:
can I make a header that will actually not be included in the build and define the required libs there?

ivankravets, 2016-06-18:
@me-no-dev no matter. In any place/ in any file (`*.c`, `*.cpp`, `.h`, etc.)

me-no-dev, 2016-06-18:
it is in the CPP file [here](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/AsyncWebSocket.cpp#L40)

ivankravets, 2016-06-18:
@me-no-dev your code is fine. This is a behaviour of PlatformIO 2.0. We are working on the PlatformIO 3.0 where the Library Manager and Library Builder will be significantly refactored.

The one reason why deep finder is turned off is that current Library Builder doesn't handle `#ifdef` and etc constructions.

### Conclusion

* @me-no-dev you don't need to change your code.
* @kuthullu please include `#include <Hash.h>` library manually.

P.S: It should work without any changes with PlatformIO 3.0.

closed 2016-09-23 by me-no-dev

fab33, 2016-12-12:
In 3.2.1 it doesn't work. You have to include <Hash.h>

ivankravets, 2016-12-12:
@fab33 You can specify extra dependencies for a project using [lib_deps](http://docs.platformio.org/en/stable/projectconf.html#projectconf-lib-deps) option
```ini
lib_deps = Hash
```

ivankravets, 2016-12-12:
@me-no-dev do you have any info when new release of ESP8266 for Arduino will be issued?

me-no-dev, 2016-12-13:
@ivankravets asking for 2.4.0? I would rather have it sooner than later, but it depends on Ivan's time and by the looks of it.... it will be maybe January. There are some goals that need to be covered by the end of the year.

ivankravets, 2016-12-13:
@me-no-dev  Thanks! We will wait for that! 

/cc @xoseperez

probonopd, 2019-05-02:
This is still an issue for me. https://github.com/probonopd/WirelessPrinting/pull/118

roberttidey, 2020-02-06:
I can't get this OTA to work on ESP32. Compiles OK and interface comes up but upload stalls and fails after about 10% progress and serial shows an abort is called on core 1.

Arduino environment 3MB/1MB flash split. Compiled binary in 1.1MB

So I thought I'd try using the example code. First issue I got was missing Hash.h (which is why I have raised this here). Tried commenting that include out and compiled ok, but behaved as before with abort and stall. I don't know whether missing Hash.h out could be associated with that but I would have expected a compile error of it was actually being used.

I have checked my ESP8266 and ESP32 environments. ESP8266 does indeed have the Hash library but ESP32 does not.