me-no-dev/ESPAsyncWebServer #48: wdt reset when websocket client disconnect

kosmonavtX, 2016-06-22:
When the client disconnects are wdt reset:

ws[/ws][3] disconnect: 0
ets Jan  8 2013,rst cause:4, boot mode:(3,1)
wdt reset

Error instable.

ahmadmn, 2016-06-23:
I see similiar problem of kosmonavtX in your example:
[](https://gist.github.com/me-no-dev/116e417ea6a3bbc98b08)

me-no-dev, 2016-06-23:
can you guys please post some code that I can have a look at. the lib is asynchronous which means that if you do not waste time in the callbacks, there is no way for it to cause wdt. All calls return immediately and process the data in background.

me-no-dev, 2016-06-23:
@ahmadmn client should be set to NULL [here](https://gist.github.com/me-no-dev/116e417ea6a3bbc98b08#file-asynctelnetserver-ino-L40)

ahmadmn, 2016-06-23:
Thanks a lot

ahmadmn, 2016-06-23:
Thanks a lot
Now it works perfectly!

kosmonavtX, 2016-06-23:
@me-no-dev, code is large enough.
 I'm using a different [library](https://github.com/Links2004/arduinoWebSockets). In the synchronous and asynchronous mode runs without resetting. Code function event from websocket almost the same:
1.  [with wdt reset](https://gist.github.com/kosmonavtX/2acb1ce7332a3e8f9e9f767e2ca54161)
2.  [without reset, library arduinoWebSockets](https://gist.github.com/kosmonavtX/79eb7e904ce18072c02dee35c4523729)

me-no-dev, 2016-06-23:
@kosmonavtX interesting.. the code looks fine. How should I run it to trigger the error?

me-no-dev, 2016-06-23:
when you say disconnect, do you mean that you run through the disconnect routine, or that the browser closes, or how?

kosmonavtX, 2016-06-23:
Wdt reset when the page is refreshed, the client is disconnected.
For web pages I use your async server. But wdt reset came only when connected websockets. Prior to this, all worked well.

me-no-dev, 2016-06-24:
what browser are you using? I used to see those problems sometimes with chrome but I believe I fixed them

kosmonavtX, 2016-06-24:
Most firefox. Less chrome.

me-no-dev, 2016-06-24:
cool, will give it a go and see what the results will be.

pabjr, 2016-06-25:
So I was having the same problem and tracked it down to this piece of [code](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/AsyncWebSocket.cpp#L395).

If I comment out this line, all works well.  

I am unsure what other ramifications this has but I imagine that it is meant to free memory after the socket closes.  I suspect that the 'cl' would pass out of scope anyway and the delete is not needed. But, there is a high probability that I am wrong.

Also,  I don't know if the line about it does anything at all.  The function it call just returns a bool.

Hope this helps, BTW ... This is a great little piece of code me-no-dev.

me-no-dev, 2016-06-25:
this is interesting... the line you talk about just frees the client from the memory. Are there any stack traces coming out? onDisconnect should be actually the very last piece of code that get's executed before the connection is totally closed, so why would it error? Btw I have tested this extensively with Chrome and Safari (mac user here) and did find differences in how they handle disconnects, but got to fix all of that.
@pabjr if you remove that line, there should be a mem leak. Can you confirm that?

me-no-dev, 2016-06-25:
ok I think I might have a pointer to what could be going wrong (maybe).
Try to remove the very next line ```server->handleDisconnect(this);``` and replace it with ```delete this;```
could be the fact that the web server is expecting to clear a request object, instead of WebSocket

kosmonavtX, 2016-06-25:
```
ws[/ws][1] disconnect: 0
hDisconnect2
_client->onDisconnect

 ets Jan  8 2013,rst cause:4, boot mode:(3,1)

wdt reset
```

_server->_handleDisconnect(this):

```
void AsyncWebSocket::_handleDisconnect(AsyncWebSocketClient * client){
  if(_clients == NULL){
    return;
  }
  if(_clients->id() == client->id()){
    _clients = client->next;
    delete client;

    Serial.println("hDisconnect2");

    return;
  }
  AsyncWebSocketClient * c = _clients;
  while(c->next != NULL && c->next->id() != client->id()) c = c->next;
  if(c->next == NULL){
    return;
  }
  c->next = client->next;
  delete client;
}
```
Maybe somewhere here?

`_client->onDisconnect([](void *r, AsyncClient* c){ ((AsyncWebSocketClient*)(r))->_onDisconnect(); Serial.println("_client->onDisconnect"); }, this);`

me-no-dev, 2016-06-25:
hmmm... that's not it... if you see ```_client->onDisconnect``` printed to serial, that means that the free went fine. your boot mode also looks funky? ```boot mode:(3,1)``` I wonder why 3,1?
do you see a delay between the last print and the wdt triggering?

pabjr, 2016-06-25:
O.K. a little update...

Yes, my last edit caused a mem leak but I traced the problem a little further and have isolated it to the following [code](https://github.com/me-no-dev/ESPAsyncTCP/blob/master/src/ESPAsyncTCP.cpp#L232)

If I comment out:

 `if(err != ERR_OK) {
      err = abort();
    }`

everything works as written and there is no mem leak as far as I can tell.

To this point I have not been able to understand why it does not look like the PCB is closed or why the abort statement causes a reset.

Any thoughts?

me-no-dev, 2016-06-25:
I scratched my head quite some time today also. I could not find a way to reproduce it constantly though... I had chrome and firefox both open refreshing, reconnecting and whatnot and every now and then I'll get wdt or funky exception on connect. Saw some strange packets coming from chrome about already closed connections. I saw chrome to ack one less byte (maybe some sort of keep alive), but nothing that can get me a clue at what point exactly the code breaks.
I'll read the source for close and what are the possible scenarios of return.
Can you put debug there and note whet is the error code that triggers it on your end?

me-no-dev, 2016-06-26:
ok... if you replace [this line](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/AsyncWebSocket.cpp#L288) with ```client->close(true);``` it should fix your problem, though it opens another one... for some reason the connection is kept in wait state for another minute till it's released, so if you have something going connect -> disconnect ->.... you will at some point run out of heap (given connect/disconnects happen one after another) which is not cool...

me-no-dev, 2016-06-28:
guys, this is fixed! Please update your TCP libs as well and give it a go!

me-no-dev, 2016-06-30:
Yes i did :) all should be good now :) there are some new features you might like. You can check the readme for more info

ahmadmn, 2016-06-30:
What is application of FSBrowser?

me-no-dev, 2016-06-30:
it lets you browse, edit, download and delete contents from SPIFFS through the browser.
Upload the example to your ESP, Upload SPIFFS using the tool for Arduino IDE and you will get it if you try to open http://esp-ip/edit

ahmadmn, 2016-07-01:
Thanks a lot,
1- Is there any tutorial or document for FSBrowser usage?
2- Is "Uploading SPIFFS using the tool for Arduino IDE" as follow: https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md#uploading-files-to-file-system
3- Is FSBrowser useful for my costumers to upgrade firmware(sketch)?
Regards

me-no-dev, 2016-07-01:
1. No, you do not need one :) open your browser and it's self explanatory.
2. Yes
3. No

this is what it looks like:

[image]

closed 2016-07-21 by me-no-dev