me-no-dev/ESPAsyncWebServer #49: Need an example of TCP Client + WiFiManager(@tzapu)

ahmadmn, 2016-06-24:
Hi
Thanks for your fastest ESP lib.
I'm new in ESP8266 and need a TCP Client example with Async capability same as your server example:
https://gist.github.com/me-no-dev/116e417ea6a3bbc98b08

if you have an example, can I mix WiFiManager(@tzapu) with it?

closed 2016-06-29 by ahmadmn

alanswx, 2016-07-02:
Here is a first version of this - it needs some work still:
https://github.com/alanswx/ESPAsyncWiFiManager
Also it should have a non-modal mode.

ahmadmn, 2016-07-04:
Thanks Alan,
I am testing it,
Regards.

gnkarn, 2017-04-26:
could you please comment if wifi manager works with this version on asyncmode? , thank you

Juntendo11, 2021-07-08:
I used it to upload the data from MPU6050 to asynch web server and worked like a charm cheers!