me-no-dev/ESPAsyncWebServer #50: How to use instructions missing

thehellmaker, 2016-07-03:
Hi,
I am seeing that ESPAsyncWebServer library is not there even in the latest version of ESP8266. Theres very little information about how to use this library.

Could you please help me out here? I will write improve the documentation once I get to know how to use this.

Cheers,
Akash A

andig, 2016-07-03:
It is here and has an extensive readme. So whats missing?

closed 2016-07-21 by me-no-dev

thehellmaker, 2016-07-22:
Oh My bad it had mentioned that it was a part of https://github.com/esp8266/Arduino.

Readme says:
To use this library you need to have the latest git versions of either ESP8266 or ESP31B Arduino Core.

I thought it meant that this library is already a part of https://github.com/esp8266/Arduino  so i didn't have to download this library. I think installation has to be made clear in the readme.