me-no-dev/ESPAsyncWebServer #52: Web Server crashed

thehellmaker, 2016-07-06:
Hi,
I started Http calls continuously from an android and I see that when simultaneous calls are made esp8266 crashes. Here is the crash log 

Exception (29):
epc1=0x402057fc epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffff8a0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffa40:  00000010 3fffbd88 3fffbd88 3ffe8548  
3ffffa50:  00000020 000000c8 00000000 40206b0c  
3ffffa60:  3fffbfac 00000010 3ffffabc 40206b5b  
3ffffa70:  000005b4 00000000 3ffffabc 3ffe8548  
3ffffa80:  3ffffab0 000000c8 3ffffae0 401004d8  
3ffffa90:  3fffbf24 00000076 3ffffae0 3ffe8548  
3ffffaa0:  3ffffbd8 000000c8 00000000 40204162  
3ffffab0:  00000000 00000000 00000000 3fffbfd4  
3ffffac0:  0000001f 00000010 3fffb964 40206c78  
3ffffad0:  3fffb964 000000c8 3fffb964 402041d5  
3ffffae0:  00000000 00000000 00000000 3fffbfac  
3ffffaf0:  0000001f 00000010 3ffffbfc 402023f6  
3ffffb00:  3fffb964 3ffffbd8 3fffbdf8 402026c7  
3ffffb10:  7473227b 73757461 73223a22 65636375  
3ffffb20:  2c227373 6d616e22 223a2265 6d6f7441  
3ffffb30:  33312d38 33313338 2c223732 69777322  
3ffffb40:  5f686374 74617473 3a227375 7773227b  
3ffffb50:  68637469 223a2230 222c2230 74697773  
3ffffb60:  22316863 2230223a 7773222c 68637469  
3ffffb70:  223a2232 222c2230 74697773 22336863  
3ffffb80:  2230223a 00007d7d 00000002 40101c72  
3ffffb90:  40104f34 3fff1414 3ffed930 3ffef0ec  
3ffffba0:  00000010 3ffffc88 00000000 4010053d  
3ffffbb0:  00000010 3ffffc94 00000000 3ffef0ec  
3ffffbc0:  00000010 3ffffc50 00000000 4010053d  
3ffffbd0:  00000010 3ffffc88 3fffbf24 0000007f  
3ffffbe0:  00000076 000000c7 00000076 3ffe8548  
3ffffbf0:  00000010 3fffbdec 00000200 3fffbefc  
3ffffc00:  0000001f 00000010 3ffffc50 0000001f  
3ffffc10:  000000c8 3ffffc70 3ffffc70 40206b0c  
3ffffc20:  3ffe8cd8 00000000 3ffffc70 00000002  
3ffffc30:  00000002 3fffb964 3fffb964 40207554  
3ffffc40:  3fffb978 3fffbcb4 3ffffc70 40205b49  
3ffffc50:  3ffe8cd8 3fffbcb4 3fffb978 40206aac  
3ffffc60:  3fffb978 3fff923c 3fffb978 40206d2d  
3ffffc70:  00000000 3ffea7e7 3fffb964 40204580  
3ffffc80:  00000002 3ffea7e7 3fffb978 40206d90  
3ffffc90:  3ffea7e7 00000000 00000000 00000002  
3ffffca0:  00000002 3ffea7e7 3fffb964 40204974  
3ffffcb0:  3fffb978 3fff923c 3fffb978 0000001f  
3ffffcc0:  00000021 3ffea7c8 3fffb964 402045b6  
3ffffcd0:  3fffb978 3ffea7c8 3fffb978 40206d90  
3ffffce0:  3ffea7c8 00000000 00000000 0000001f  
3ffffcf0:  00000021 3ffea7c8 3fffb964 402047a8  
3ffffd00:  3fffb978 3fff923c 3fffb978 00000036  
3ffffd10:  00000057 3ffea792 3fffb964 402045b6  
3ffffd20:  3fffb978 3ffea792 3fffb978 40206d90  
3ffffd30:  3ffea792 00000000 00000000 00000036  
3ffffd40:  00000057 3ffea792 3fffb964 402047a8  
3ffffd50:  3fffb978 00000001 3fffb978 00000018  
3ffffd60:  0000006f 3ffea77a 3fffb964 402045b6  
3ffffd70:  3fffb978 3ffea77a 3fffb978 40206d90  
3ffffd80:  3ffea77a 00000000 00000000 00000018  
3ffffd90:  0000006f 3ffea77a 3fffb964 402047a8  
3ffffda0:  3ffe8cd8 3fffb978 3fffb978 00000013  
3ffffdb0:  00000082 3ffea767 3fffb964 402045b6  
3ffffdc0:  3fffb978 3ffea767 3fffb978 40206d90  
3ffffdd0:  3ffea767 00000000 00000000 00000013  
3ffffde0:  00000082 3ffea767 3fffb964 402047a8  
3ffffdf0:  3ffe8cd8 3fffb964 3fffb978 00000015  
3ffffe00:  00000097 3ffea752 3fffb964 4020452d  
3ffffe10:  3fffb978 3ffea752 3fffb978 40206d90  
3ffffe20:  3ffea752 3ffed270 3fffb928 00000015  
3ffffe30:  00000097 3ffea752 3fffb964 402047a8  
3ffffe40:  3ffea06c 00000000 00000000 4010068c  
3ffffe50:  00000000 09ce2c9c 40102f52 00000100  
3ffffe60:  3fffb978 3ffea06c 3ffea06c 00000001  
3ffffe70:  00000001 00007fff 000004f8 3fffb900  
3ffffe80:  00000001 3fff31d4 3fffb8ac 402049a1  
3ffffe90:  3fffc278 40102cec 3fffb8ac 40202e10  
3ffffea0:  3fff9e9c 3ffeefbc 3fff9e9c 40225cf9  
3ffffeb0:  3fff09fc 3fff0a08 00000006 3fff0810  
3ffffec0:  3fff9e9c 3fff0818 3fff0814 40202e63  
3ffffed0:  3fff9e9c 3fff0818 3fff0814 4022b460  
3ffffee0:  00000097 0501a8c0 00000018 00000097  
3ffffef0:  00000018 00000000 3ffef1cc 40107304  
3fffff00:  40100000 00000000 00000064 3fff0a00  
3fffff10:  3ffea72a 3fff0a08 3fff31d4 40229969  
3fffff20:  3fff0724 3fff127c 3fff127c 3ffee390  
3fffff30:  00000000 3fff31d4 0000001c 3fff127c  
3fffff40:  3ffea71c 00000000 3fff31d4 40228d65  
3fffff50:  0201a8c0 000000f5 00000000 00000019  
3fffff60:  00000002 0000001a 4020977f 3ffed270  
3fffff70:  3ffea6f4 3fffdcc0 3ffe9908 3ffe9908  
3fffff80:  402096f2 3ffed270 00000000 3fff12ec  
3fffff90:  3fffdc80 00000000 3fff31d4 4021d7e7  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld

Here is my handler method
void handleDiscovery(AsyncWebServerRequest *request) {
  AsyncResponseStream *response = request->beginResponseStream("text/json");
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["status"] = "success";
  root["name"] = wifiManager.getStationSSID();
  JsonObject& switchStatus = jsonBuffer.createObject();
  switchStatus["switch0"] = "0";
  switchStatus["switch1"] = "0";
  switchStatus["switch2"] = "0";
  switchStatus["switch3"] = "0";
  root["switch_status"] = switchStatus;
  
  char buffer[200];
  root.printTo(buffer, sizeof(buffer));
  request->send(200, "application/json", buffer);
}

Can you please help me as to how to debug/fix the issue?

thehellmaker, 2016-07-06:
Changed the reponse sending code to 

`root.printTo(*response);
  request->send(response);`

I think it was out of memory issue

me-no-dev, 2016-07-06:
you were instantiating a response and then not sending it, but sending a basic 200 with the json

me-no-dev, 2016-07-06:
you can see examples on how to send json in the README :)

thehellmaker, 2016-07-12:
Thanks. That was my bad. Closing

closed 2016-07-12 by thehellmaker