me-no-dev/ESPAsyncWebServer #54: Web Server Stops responding after a while

thehellmaker, 2016-07-12:
Hi All,
ESP async webserver becomes unavailable after sometime intermittently where it says 
Connection to http://192.168.1.4:80 refused
org.apache.http.conn.HttpHostConnectException: Connection to http://192.168.1.4:80 refused

The same happens from the web browser and then it starts working randomly. Checking the logs of the ESP device itself there is no crash.

Am i missing something in the setup of the server which can keep it alive all the time.

Cheers,
Akash A

me-no-dev, 2016-07-16:
Can you share some code? I have not experienced such a thing before

ClaudioHutte, 2016-07-19:
Can you ping the ESP? I've had a similar problem but apparently was at ARP level and I resolved with this code that I found elsewhere:

`extern "C" {
  extern char *netif_list;
  uint8_t etharp_request(char *, char *);
}

void forceARP() {
  char *netif = netif_list;

  while (netif)
  {
    etharp_request((netif), (netif+4));
    netif = *((char **) netif);
  }
}`

Within the loop the forceARP(); is cyclically called:

`void loop()
{
    if(one_second_flag) {
      forceARP();
      one_second_flag = false;
    }
}
`

one_second_flag is set by Ticker (see Ticker.h) every second.
It's clearly a workaround.

thehellmaker, 2016-07-22:
Hi Claudio,
If its a work around would the root cause and long term solution be at EspAsyncTcp or at EspArduino level? Do you happen to know?

Cheers,
Akash A

thehellmaker, 2016-07-23:
Hi Claudio, 
Looks like Ticker.h does't have one_second_flag 

https://github.com/esp8266/Arduino/blob/master/libraries/Ticker/Ticker.h

Do you happen to know the new way of doing it?

Cheers,
Akash A

thehellmaker, 2016-07-23:
Anyways thats not a problem delay ill figure out. but when calling forceARP its crashing 

`#include <ArduinoJson.h>
#include "FS.h"
#include "Ticker.h"

extern "C" {
	extern char *netif_list;
	uint8_t etharp_request(char *, char *);
}

void forceARP() {
	char *netif = netif_list;
	while (netif) {
		etharp_request((netif), (netif + 4));
		netif = ((char *)netif);
	}
}

void setup(void) {
	Serial.begin(115200);
	Serial.println(ESP.getResetInfo());
}

void loop(void) {
	delay(1000);
	forceARP();
}
`

It crashed saying 

`
Soft WDT reset

ctx: cont 
sp: 3fff0180 end: 3fff0410 offset: 01b0

>>>stack>>>
3fff0330:  4021044e 3ffed130 3fff11ac 00000000  
3fff0340:  3ffef3d8 00000292 00000292 fffffffc  
3fff0350:  00000024 0000002a 3fff1624 401004d8  
3fff0360:  40217410 4021746e 3fff1454 40107304  
3fff0370:  03030201 40217614 3fff1454 4022c408  
3fff0380:  00000002 00000001 00000000 3ffed130  
3fff0390:  3fff21cc 4022ef7c 3fff21cc 3fff1624  
3fff03a0:  3fffdad0 3fff1657 3fff1624 4022eee9  
3fff03b0:  3fff1628 3fff1628 00000000 00000001  
3fff03c0:  3fffdad0 3ffef3e4 4020bd90 3ffef3e4  
3fff03d0:  3fffdad0 3fff1628 3fff1624 402067bc  
3fff03e0:  00000000 00000000 3ffef3dc 40206833  
3fff03f0:  feefeffe feefeffe 3ffef3dc 4020bddc  
3fff0400:  feefeffe feefeffe 3ffef3f0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

Port closed
Opening port
Port open

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset`

thehellmaker, 2016-07-23:
Hi Claudio, 
Referred to the link http://www.esp8266.com/viewtopic.php?p=12809
There was a problem with the post you had made

`netif = ((char *)netif);` is wrong 
The correct code is  `netif = *((char **)netif);`

Thanks for the reference claudio I will try to put it into test and see if it works.

Cheers,
Akash A

ClaudioHutte, 2016-07-23:
Hi Akash. Sorry for my confusing post. It's a bad thing to write in the dead of night :)
First. You are right, there was a missing asterisk, I don't know why it is disappeared as I just copied and pasted my code, which is correct of course.
Second. The one_second_flag is not part of the Ticker code, sorry again for my bad explanation. Here is a snipped of the whole code, hope this is helpful:

```cpp

bool one_second_flag = false;

// set the one second flag, this function is called every 1 second
void setTickflag() {
  one_second_flag = true;
}

extern "C" {
  extern char *netif_list;
  uint8_t etharp_request(char *, char *);
}

void forceARP() {
  char *netif = netif_list;

  while (netif)
  {
    etharp_request((netif), (netif+4));
    netif = *((char **) netif);
  }
}

void setup
{
   TimeTicker.attach(1, setTickflag);  // will call setTickflag every 1 second
}

void loop()
{
    if(one_second_flag) {
      one_second_flag = false;
      forceARP();
    }
}
```

The use of a global flag set by a function that in turn is called by Ticker is to avoid the code within the loop function be blocked while waiting for the delay() to expire.

Greetings,
Claudio

me-no-dev, 2016-07-25:
@thehellmaker this loop should be totally fine:
```cpp
void loop(void) {
  delay(1000);
  forceARP();
}
```
You can use [this tool](https://github.com/me-no-dev/EspExceptionDecoder) to decode the stack trace that you are getting, to get a better idea where the code crashes.
I personally have not dealt with arp on the esp yet.

thehellmaker, 2016-07-26:
Thanks a lot @me-no-dev for the tool link.

Also this wnt work as it will block the mcu and the responses will all be 1s delayed and not immediate
`void loop(void) {
  delay(1000);
  forceARP();
}`

Ticker approach works better because there is no blocking.

Tested this. There have been no arp issue till now after this fix so. Hence resolving this. Will reopen in case there are more issues.

me-no-dev, 2016-07-26:
@thehellmaker ```delay()``` does not block the CPU, it just blocks your code at that point for the given time and let's the rest of the system run in the background. ```delayMicroseconds()``` does block the CPU.

thehellmaker, 2016-07-26:
Ok. I didnt know that. Thanks @me-no-dev .Also I just checked I am facing the problem again.

From Chrome browser the device is responding but from android device its not able to connect to it.

thehellmaker, 2016-07-26:
Tried ping from my tablet, my phone and Laptop.
1. Laptop - Reachable
2. tablet(Nexus 7_ - not reachable
3. Phone(One plus one android) - reachable

me-no-dev, 2016-07-26:
can we move this discussion to https://github.com/esp8266/Arduino since it's not really related to the web server

thehellmaker, 2016-07-26:
Surely @me-no-dev. Makes sense

closed 2016-07-26 by thehellmaker

Sewmina7, 2019-04-11:
I had the same issue, now it's reduced after I started using edge explorer instead of chrome, don't know any reason, but it worked for me

WeightScale, 2019-12-30:
that's right
while (*netif){
		etharp_request((netif), (netif + 4));
		netif = *((char **) netif);
	}