me-no-dev/ESPAsyncWebServer #55: File upload: data past the boundary end

ClaudioHutte, 2016-07-18:
Hello, I noticed a problem with the upload of files. I used the code in the example at the readme page:

`void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){
  if(!index){
    Serial.printf("UploadStart: %s\n", filename.c_str());
  }
  for(size_t i=0; i<len; i++){
    Serial.write(data[i]);
  }
  if(final){
    Serial.printf("UploadEnd: %s, %u B\n", filename.c_str(), index+len);
  }
}`

On the client side a form that submits the file to upload looks like this:

`<form action="upload" method="POST" enctype="multipart/form-data">
<input type="file" required name="path">
<input type="submit" name="_submit" value="Upload">
</form>`

so the browser sends to the server the following data:

`Content-Type: multipart/form-data; boundary=---------------------------109992776831753
Content-Length: 1191

-----------------------------109992776831753
Content-Disposition: form-data; name="path"; filename="index.htm"
Content-Type: text/html

<html>
<head>
...(omitted data for brevity)....
</body>
</html>

-----------------------------109992776831753
Content-Disposition: form-data; name="_submit"

Upload
-----------------------------109992776831753--
`

The intended data to be received should be limited to the part after the two carriage return and before the following boundary marker, however the actual data returned by the function overflow.
This is what the function put into the file:

`<html>
<head>
...(omitted data for brevity)....
/body>
</html>

-----------------------------109992776831753
Content-Disposition: form-data; name="_submit"

Upload`

Submitted file size: 897 bytes, stored file: 1001 bytes.

me-no-dev, 2016-07-25:
interesting... what is your OS/Browser?

ClaudioHutte, 2016-07-26:
My system is Win7 64bit pro SP1 and Browser is Firefox 43 (well, it need to be updated but I'm lazy :) ).
If you want I can provide you the full body and headers of the request in an attached text file. I use the browser's debugger tools to retrieve these data.

closed 2019-10-05 by stale[bot]