me-no-dev/ESPAsyncWebServer #56: AsyncWebSocket printf bug

ClaudioHutte, 2016-07-24:
I found a bug in AsyncWebSocket printf function: wrong argument printed, perhaps invalid pointer.
The problem seems to happen with <object>.printf(client_id, arguments...) but not with client->printf(arguments...).
Example code to reproduce the problem (code of the setup function omitted):

```cpp
const char WSErr[] PROGMEM = "{\"type\":\"STATUS\",\"response\":\"Error at %d\"}";
int clientID;
bool readyToTest = false;

AsyncWebSocket ws("/.ws"); // access at ws://[esp ip]/.ws

void onEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len)
{
  //Handle WebSocket event
  if(type == WS_EVT_CONNECT) {
    Serial.printf("ws[%s][%u] connect\n", server->url(), client->id());
    client->printf("{\"type\":\"ACK\",\"response\":\"HELO\",\"client-id\":%u}", client->id());
    client->ping();
    clientID = client->id();
  } else if(type == WS_EVT_DISCONNECT) {
    Serial.printf("ws[%s][%u] disconnect: u\n", server->url(), client->id());
  } else if(type == WS_EVT_ERROR) {
    Serial.printf("ws[%s][%u] error(%u): %s\n", server->url(), client->id(), *((uint16_t*)arg), (char*)data);
  } else if(type == WS_EVT_PONG) {
    Serial.printf("ws[%s][%u] pong[%u]: %s\n", server->url(), client->id(), len, (len)?(char*)data:"");
    readyToTest = true;
  } else if(type == WS_EVT_DATA) {
  }
}

void testWS() {
   char tmp[100];
   unsigned char testvalue = 2;

   strcpy_P(tmp, WSErr);
   Serial.printf(tmp, testvalue);
   ws.printf(clientID, tmp, testvalue);
}

void loop {
   if(readyToTest) {
     readyToTest = false;
     testWS();
   }
}
```

Output at serial monitor:
```text/plain
{"type":"STATUS","response":"Error at 2"}
```

Output at browser console (see script below)
```text/plain
{"type":"STATUS","response":"Error at 1073698976"}
```

This is the javascript that handles websocket connections and report to console:
```javascript
        ws = new WebSocket("ws://10.0.22.101/.ws", WSPROTO);
	ws.onopen = function(event) {  
		connected = true;
	};
	ws.onclose = function(event) {
		connected = false;
	};
	ws.onmessage = function (event) {
           console.log(event.data);
        }
```

me-no-dev, 2016-07-25:
Yes I stumbled upon this issue in another library and meant to look at my code here, but got busy with other things and forgot :)
Thanks for noting, will fix.

ClaudioHutte, 2016-07-26:
Happy to be helpful, thanks to you.

masonevans, 2019-07-24:
Just noting that this is still an issue.  At least maybe an update to the docs noting this?  I spent a long time trying to figure out why `ws.printf(client_id...` was giving me very strange output when `ws.printfAll(...` worked fine.

closed 2019-10-06 by stale[bot]