me-no-dev/ESPAsyncWebServer #58: Problem with requestAuthentication() in Internet Explorer

EduardUA, 2016-07-27:
Request Authentication doesn't work in Internet Explorer.
After input login and password request repeated.

```
server.on("/index.html", [](AsyncWebServerRequest *request)
	{		
		if (!request->authenticate(_login, _pass))			
			return request->requestAuthentication();
		AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/index.html", "text/html");
		request->send(response);
	});	
```

luc-github, 2016-10-17:
I can confirm issue with windows 7 / IE 11
No problem so far with chrome/safari/firefox

me-no-dev, 2016-10-17:
@luc-github can you grab the actual request? I might have an idea...

luc-github, 2016-10-17:
Sure, what can I do to help ?

me-no-dev, 2016-10-17:
does internet explorer have some sort of inspector? I need to see the request headers
something like:
[image]

if not maybe you can print [here](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebRequest.cpp#L123) before ```_parseLine();``` and see what comes out

luc-github, 2016-10-17:
Sorry my browser is in french - this one is Windows 10/IE11

here before I type admin/admin:
[image]
 and after:
[image]
[image]

is it what you expected ?

me-no-dev, 2016-10-17:
yes! thank you

me-no-dev, 2016-10-17:
hm it seems to use WWW-Authenticate header instead

me-no-dev, 2016-10-17:
hm actually no. there should be another request from the browser asking you for user/pass. is it not asking you at all?

luc-github, 2016-10-17:
nope there is the failed and the new one:
[image]
`en attente` means waiting

me-no-dev, 2016-10-17:
http://stackoverflow.com/questions/2192811/digest-authentication-not-working-on-ie8-firefox-and-chrome-are-fine
see this link. maybe propagates through the versions

luc-github, 2016-10-17:
seems second has more :
[image]
[image]

not sure it helps

me-no-dev, 2016-10-17:
https://jira.mongodb.org/browse/SERVER-5890
also here saying that there is a new line in the header? Maybe using the print in the lib where I asked could tell us that

luc-github, 2016-10-17:
sure what command do you want me to add before _parseLine(); to print the request ?

me-no-dev, 2016-10-17:
Serial.println(_temp);

luc-github, 2016-10-17:
here it is, I typed admin/admin twice
```
connected with NETGEAR_2GEXT_OFFICE, channel 3
dhcp client start...
ip:192.168.1.18,mask:255.255.255.0,gw:192.168.1.1
GET /edit HTTP/1.1
Accept: text/html, application/xhtml+xml, image/jxr, */*
Accept-Language: fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.6,zh-Hant-TW;q=0.5,zh-Hant;q=0.4,zh-Hans-CN;q=0.3,zh-Hans;q=0.1
User-Agent: Mozilla/5.0 (Windows NT 10.0; WOW64; Trident/7.0; Touch; rv:11.0) like Gecko
Accept-Encoding: gzip, deflate
Host: 192.168.1.18
Connection: Keep-Alive

GET /edit HTTP/1.1
Accept: text/html, application/xhtml+xml, image/jxr, */*
Accept-Language: fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.6,zh-Hant-TW;q=0.5,zh-Hant;q=0.4,zh-Hans-CN;q=0.3,zh-Hans;q=0.1
User-Agent: Mozilla/5.0 (Windows NT 10.0; WOW64; Trident/7.0; Touch; rv:11.0) like Gecko
Accept-Encoding: gzip, deflate
Host: 192.168.1.18
Connection: Keep-Alive
Authorization: Digest username="admin",realm="asyncesp",nonce="96448e594139537eb3a91b328faf5ac3",uri="/edit",cnonce="a461c687ff395d4f010be32286e20657",nc=00000001,response="cc2917706af94cf3a1f999cc0c1a9b75",qop="auth",opaque="badf84932bd8862f56479ba0acd1a8cb"

GET /edit HTTP/1.1
Accept: text/html, application/xhtml+xml, image/jxr, */*
Accept-Language: fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.6,zh-Hant-TW;q=0.5,zh-Hant;q=0.4,zh-Hans-CN;q=0.3,zh-Hans;q=0.1
User-Agent: Mozilla/5.0 (Windows NT 10.0; WOW64; Trident/7.0; Touch; rv:11.0) like Gecko
Accept-Encoding: gzip, deflate
Host: 192.168.1.18
Connection: Keep-Alive
Authorization: Digest username="admin",realm="asyncesp",nonce="f73a8231a3977372dbadc3684f234d5f",uri="/edit",cnonce="399ee6b5da255da7c7c47bb65aad44f5",nc=00000001,response="d7dd556917e33e8399661fd19869799d",qop="auth",opaque="8dcbfd7d081a680b3869014707977d45"

```

me-no-dev, 2016-10-18:
aaah
[here](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebAuthentication.cpp#L135) and [here](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebAuthentication.cpp#L154) is checked for comma and space after that and IE does not send space. Should be checked just for comma and [avLine](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebAuthentication.cpp#L152) should be trimmed.

luc-github, 2016-10-18:
Sorry for delay - I was little bit busy.

I did the modifications (remove space and add trim)
but no success under IE and still ok for firefox :
```
GET /edit HTTP/1.1
Accept: text/html, application/xhtml+xml, */*
Accept-Language: fr-FR
User-Agent: Mozilla/5.0 (Windows NT 6.1; WOW64; Trident/7.0; NP08; MAAU; NP08; rv:11.0) like Gecko
Accept-Encoding: gzip, deflate
Host: 192.168.1.18
Connection: Keep-Alive
Authorization: Digest username="admin",realm="asyncesp",nonce="cdf8fdae15d635fdd29b533353b8f60d",uri="/edit",cnonce="434d8a0f3effdcc0893f953757e829db",nc=00000001,response="e408aa2f3d35a20bb85311227e465580",qop="auth",opaque="fa15b74913db5b22c512ca11ea2860d8"

GET /edit HTTP/1.1
Accept: text/html, application/xhtml+xml, */*
Accept-Language: fr-FR
User-Agent: Mozilla/5.0 (Windows NT 6.1; WOW64; Trident/7.0; NP08; MAAU; NP08; rv:11.0) like Gecko
Accept-Encoding: gzip, deflate
Host: 192.168.1.18
Connection: Keep-Alive
Authorization: Digest username="admin",realm="asyncesp",nonce="873cb98eebd8e5d660af45790e06b95b",uri="/edit",cnonce="6bac3f6e5ab2840e38561063c076572e",nc=00000001,response="2b2a017c1b1115238e5de0fb49001fff",qop="auth",opaque="2424b6555db609aa683f83e56e0d28f7"

```

me-no-dev, 2016-10-18:
can you print avName and value [here](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebAuthentication.cpp#L161) maybe space was not trimmed?

luc-github, 2016-10-18:
Ok I print values between `>` and `<` to see if has space, and add trim to check but seems not the issue:
```
GET /edit HTTP/1.1
Accept: text/html, application/xhtml+xml, */*
Accept-Language: fr-FR
User-Agent: Mozilla/5.0 (Windows NT 6.1; WOW64; Trident/7.0; NP08; MAAU; NP08; rv:11.0) like Gecko
Accept-Encoding: gzip, deflate
Host: 192.168.1.18
Connection: Keep-Alive
Authorization: Digest username="admin",realm="asyncesp",nonce="8a7b69977ed9eb05e0cb9bc1709156ad",uri="/edit",cnonce="d9e106770d7f2892533b47eff6d5287a",nc=00000001,response="3404e4d0437c78b4a1f4b4bc5c543485",qop="auth",opaque="b7e0872e6d0171b0c718dff7cdb3df33"

varName:>username<
avLine:>"admin"<
varName:>ealm<
avLine:>"asyncesp"<
varName:>once<
avLine:>"8a7b69977ed9eb05e0cb9bc1709156ad"<
varName:>ri<
avLine:>"/edit"<
varName:>nonce<
avLine:>"d9e106770d7f2892533b47eff6d5287a"<
varName:>c<
avLine:>00000001<
varName:>esponse<
avLine:>"3404e4d0437c78b4a1f4b4bc5c543485"<
varName:>op<
avLine:>"auth"<
varName:>paque<
avLine:>"b7e0872e6d0171b0c718dff7cdb3df33"<
```

me-no-dev, 2016-10-18:
not good! cuts one letter.
what is your code change in there from before?

me-no-dev, 2016-10-18:
like, how are you trimming avLine?

luc-github, 2016-10-18:
well normal way I think same as you :
```
String varName = avLine.substring(0, eqSign);
    Serial.print("varName:>");
    Serial.print(varName);
    Serial.println("<");
    avLine = avLine.substring(eqSign + 1);
    Serial.print("avLine:>");
    Serial.print(avLine);
    Serial.println("<");
    avLine.trim();
```

luc-github, 2016-10-18:
and before just remove space and add trim as you said:
```
 int nextBreak = myHeader.indexOf(",");
  if(nextBreak < 0){
    //os_printf("AUTH FAIL: no variables\n");
    return false;
  }

  String myUsername = String();
  String myRealm = String();
  String myNonce = String();
  String myUri = String();
  String myResponse = String();
  String myQop = String();
  String myNc = String();
  String myCnonce = String();

  myHeader += ", ";
  do {
    String avLine = myHeader.substring(0, nextBreak);
    avLine.trim();
    myHeader = myHeader.substring(nextBreak+2);
    nextBreak = myHeader.indexOf(",");
```

me-no-dev, 2016-10-18:
myHeader = myHeader.substring(nextBreak+2); << change to +1

luc-github, 2016-10-18:
you found it :smile: 
Now it is ok - tested on IE and firefox

luc-github, 2016-10-18:
I redo clean one: 
first trim is enough - no need the second one
so fix is :
removing space L135/L154, trim avLine after line 152 and L153 change to:    
`myHeader = myHeader.substring(nextBreak+1);`

me-no-dev, 2016-10-18:
done :) https://github.com/me-no-dev/ESPAsyncWebServer/commit/49d5a24815aa1e56ab4c8e05f4f31e788780af39

closed 2016-10-18 by me-no-dev

luc-github, 2016-10-18:
Thank you ^_^

me-no-dev, 2016-10-18:
thanks @luc-github for your help!

luc-github, 2016-10-18:
my pleasure ^_^