me-no-dev/ESPAsyncWebServer #60: strftime strftime strftime

kiralikbeyin, 2016-07-28:
Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp:67:64: error: 'strftime' was not declared in this scope
   strftime (result,30,"%a, %d %b %Y %H:%M:%S %Z", last_modified);
                                                                ^

Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp:72:60: error: 'strftime' was not declared in this scope
   return setLastModified((struct tm *)gmtime(&last_modified));

Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp:77:25: error: 'strftime' was not declared in this scope
   if(time(&last_modified) == 0) //time is not yet set

How to solve?

me-no-dev, 2016-07-28:
old ESP8266 Arduino?

kiralikbeyin, 2016-07-28:
Tried with 1.6.8 and 1.6.9
esp 2.3 stable
Nodemcu v3 LoLin

Some how it compiled once but now it doesn't without any change in sketch...

Trying to convert https://github.com/gmag11/FSBrowser/ to ESPAsyncWebServer

kiralikbeyin, 2016-07-28:
https://community.platformio.org/t/errors-after-updating-espasyncwebserver/476

changing name to _time.h didn't solve ..

kiralikbeyin, 2016-07-28:
```
  //#ifdef ESP8266
    AsyncStaticWebHandler& setLastModified(time_t last_modified);
    AsyncStaticWebHandler& setLastModified(); //sets to current time. Make sure sntp is runing and time is updated
//  #endif
```

comment on ifdef and everything is ok.

me-no-dev, 2016-07-28:
interesting.... that makes no sense...

kiralikbeyin, 2016-07-29:
I uploaded same sketch maybe 10 times now it is giving exception again...

Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(tm*)':
/Users/EvAkilli/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp:67:64: error: 'strftime' was not declared in this scope
   strftime (result,30,"%a, %d %b %Y %H:%M:%S %Z", last_modified);
                                                                ^
Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(time_t)':
/Users/EvAkilli/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp:72:60: error: 'gmtime' was not declared in this scope
   return setLastModified((struct tm *)gmtime(&last_modified));
                                                            ^
Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified()':
/Users/EvAkilli/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp:77:25: error: 'time' was not declared in this scope
   if(time(&last_modified) == 0) //time is not yet set

Can it be a core problem with time lib?

me-no-dev, 2016-07-29:
time.h is located [here](https://github.com/esp8266/Arduino/blob/master/tools/sdk/libc/xtensa-lx106-elf/include/time.h). See if you have that file

kiralikbeyin, 2016-07-29:
my lib have a Time.h and only one row : #include "TimeLib.h"

TimeLib.h

```
/*
  time.h - low level time and date functions
*/

/*
  July 3 2011 - fixed elapsedSecsThisWeek macro (thanks Vincent Valdy for this)
              - fixed  daysToTime_t macro (thanks maniac bug
*/     

#ifndef _Time_h
#ifdef __cplusplus
#define _Time_h

#include <inttypes.h>
#ifndef __AVR__
#include <sys/types.h> // for __time_t_defined, but avr libc lacks sys/types.h
#endif

#if !defined(__time_t_defined) // avoid conflict with newlib or other posix libc
typedef unsigned long time_t;
#endif

// This ugly hack allows us to define C++ overloaded functions, when included
// from within an extern "C", as newlib's sys/stat.h does.  Actually it is
// intended to include "time.h" from the C library (on ARM, but AVR does not
// have that file at all).  On Mac and Windows, the compiler will find this
// "Time.h" instead of the C library "time.h", so we may cause other weird
// and unpredictable effects by conflicting with the C library header "time.h",
// but at least this hack lets us define C++ functions as intended.  Hopefully
// nothing too terrible will result from overriding the C library header?!
extern "C++" {
typedef enum {timeNotSet, timeNeedsSync, timeSet
}  timeStatus_t ;

typedef enum {
    dowInvalid, dowSunday, dowMonday, dowTuesday, dowWednesday, dowThursday, dowFriday, dowSaturday
} timeDayOfWeek_t;

typedef enum {
    tmSecond, tmMinute, tmHour, tmWday, tmDay,tmMonth, tmYear, tmNbrFields
} tmByteFields;	   

typedef struct  { 
  uint8_t Second; 
  uint8_t Minute; 
  uint8_t Hour; 
  uint8_t Wday;   // day of week, sunday is day 1
  uint8_t Day;
  uint8_t Month; 
  uint8_t Year;   // offset from 1970; 
} 	tmElements_t, TimeElements, *tmElementsPtr_t;

//convenience macros to convert to and from tm years 
#define  tmYearToCalendar(Y) ((Y) + 1970)  // full four digit year 
#define  CalendarYrToTm(Y)   ((Y) - 1970)
#define  tmYearToY2k(Y)      ((Y) - 30)    // offset is from 2000
#define  y2kYearToTm(Y)      ((Y) + 30)   

typedef time_t(*getExternalTime)();
//typedef void  (*setExternalTime)(const time_t); // not used in this version

/*==============================================================================*/
/* Useful Constants */
#define SECS_PER_MIN  (60UL)
#define SECS_PER_HOUR (3600UL)
[... 53 lines cut ...]
void    adjustTime(long adjustment);

/* date strings */ 
#define dt_MAX_STRING_LEN 9 // length of longest date string (excluding terminating null)
char* monthStr(uint8_t month);
char* dayStr(uint8_t day);
char* monthShortStr(uint8_t month);
char* dayShortStr(uint8_t day);
	
/* time sync functions	*/
timeStatus_t timeStatus(); // indicates if time has been set and recently synchronized
void    setSyncProvider( getExternalTime getTimeFunction); // identify the external time provider
void    setSyncInterval(time_t interval); // set the number of seconds between re-sync

/* low level functions to convert to and from system time                     */
void breakTime(time_t time, tmElements_t &tm);  // break time_t into elements
time_t makeTime(tmElements_t &tm);  // convert time elements into time_t

} // extern "C++"
#endif // __cplusplus
#endif /* _Time_h */

```)

How to include / install time.h ?

kiralikbeyin, 2016-07-29:
I reinstall esp8266 2.3 and no problem now.

I am not sure if it makes trouble again.

Thank you Markus.

kiralikbeyin, 2016-07-29:
Again... Sketch and all error here -- > https://github.com/kiralikbeyin/convert-to-async

kiralikbeyin, 2016-07-29:
/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include
Make a copy of time.h and name it to "_time.h"

replace in WebHandlerImpl.h #include <time.h> to #include <_time.h>

I will give info when i get problem again...

sauttefk, 2016-07-31:
I have the same problems.
The workaround of @kiralikbeyin allows me to compile my program.
My setup:
Include of <TimeLib.h>
Arduino 1.6.10
ESP8266 V2.3
ESPAsyncWebServer-master as of 2016-07-31

closed 2016-08-01 by kiralikbeyin

zoutepopcorn, 2016-10-11:
@kiralikbeyin  
Thx for your help 👍

zenmanenergy, 2017-04-04:
I am getting this error too. It's clearly caused by the TimeLib.h library. This is a kludge fix to modify the core files. New releases will overwrite the fix.

dmitry-rivlin-vrp, 2017-05-16:
Probably we have an conflict with **NtpClientLib.h**

micha1983, 2017-05-27:
HI, 
I also have this error. 
i haven't understood how it was solved.

zoutepopcorn, 2017-05-27:
replace in WebHandlerImpl.h #include <time.h> to #include <_time.h>

mathurv, 2017-06-02:
I got out with an easy way.
go to libraries -> Time (where TimeLib.h exists) and rename Time.h to _Time.h and everything is fine after that

sarfata, 2017-07-25:
If you are using platformio you can add in your `platformio.ini` file:
````
lib_ignore=Time
````

MHz000, 2017-08-11:
While compiling for an ESP in 1.8.3 IDE  I too haye this error :-(
is mathurv's recommendation "rename Time.h to _Time.h"  save with no side effects to already running programs containg  #include <TimeLib.h>, #include <Timezone.h> etc after recompiling?
Thank-you

Ductapemaster, 2017-09-10:
I'd like to second mathurv's comment above - renaming "Time.h" to "_Time.h" in the libraries folder fixed the issue for me.  However I have not tested if this has any repercussions on other sketches or projects.

jirmjos, 2017-09-18:
For me this problem with the great Timezone library was a real headache for a long time.
The problem in my case came about when updating the ESP8266 Core with the version of Git and not with the 2.3.0 due to the need to use ESPAsyncUDP (https://github.com/me-no-dev/ESPAsyncUDP).
I'm using Windows platform in this moment with Arduino IDE compiler, and maybe the issue are too realated with windows case insensitive filenames, but anyway...

From that change of ESP8266 Timezone refused to compile and there was no way to solve it until it did not replace in Timezone.h the reference to Time.h by TimeLib.h
That is, I think if we using in the same sketch TimeLib.h (https://github.com/PaulStoffregen/Time) all references to related libraries (Timezone, etc...) that are used too should be referenced to TimeLib.h and not more references Time.h that seems have some weird cross-references  with library time.h in ESP core libs.

Regards

me-no-dev, 2017-09-18:
Yes we have the same issues on Mac as well. Case insensitive file systems can not distinguish Time.h from time.h and Arduino puts Libraries after the core/sdk so it picks up Time.h instead. Honestly Time.h should have never been named Time.h for that exact reason (time.h is the standard *nix header and present in ESP SDKs) but since it's too late to change that, we are in a sticky situation now and from now on...

jirmjos, 2017-09-18:
Totally agree whit you @me-no-dev.

Of course all library developers need take so much care with this weird "naming" issues, because can easily avoid to useless their contributions in a very short time.
Cross-platfform developments cares (Linux, Mac, Windows) I think is mandatory in this times.

Best regards.

yasheena, 2017-10-17:
I read all the infos above. Have I understand them in the right way and actually there is no solution for using strftime with ESP8266?

Voha888, 2020-07-16:
> I read all the infos above. Have I understand them in the right way and actually there is no solution for using strftime with ESP8266?

Rename Time.h in Timelib