me-no-dev/ESPAsyncWebServer #62: Web Server stops responding if client dies in the middle of sending a request

ovaltineo, 2016-07-31:
My ESP async web server application (based on ESP_AsyncFSBrowser) handles thousands of requests per hour from other ESP clients day after day.  However, if a client gets powered down in the middle of sending a request, the web server stops accepting requests - WifiClient.connect will fail on the clients.  The web server never recovers and needs to be reset.

The web server still responds to ping, and it can connect to another web server -- I tried using using that as a self-check but it keeps working.  It even accepts OTA updates.  It just won't accept client connections.

I tried connecting to itself as a self-check, but strangely, I find that this always fails even if the web server is working and accepts requests from other clients.  

Is there a way to recover from this state or detect that it is not accepting connections so I can initiate a self-reset?

me-no-dev, 2016-08-12:
this is really interesting... even if one request clears and for some reason is left hanging, the web server should still be able to handle another 4 requests in parallel. How can I test this? At what point of sending the request does the client die?

ovaltineo, 2016-08-12:
I'm not sure exactly what point it dies - the connect, sending, and receiving happens too fast.  To reproduce the problem, you'll need two esp8266.  The first one is the async web server, the other is a http client that makes a request often, e.g. every 300 milliseconds.  Pull the power on the client.  If  you get the timing right, the server will hang.  It takes me less than 5 tries to do it.

lalalandrus, 2016-08-16:
perhaps it is waiting for an ack, the stock esptcp stack has a function defined as 
#define TF_NODELAY     ((u8_t)0x40U)   /* Disable Nagle algorithm */
#define          tcp_nagle_disable(pcb)   ((pcb)->flags |= TF_NODELAY)
#define          tcp_nagle_enable(pcb)    ((pcb)->flags &= ~TF_NODELAY)

        void setNoDelay(bool nodelay){
          if(!_pcb) return;
          if(nodelay) tcp_nagle_disable(_pcb);
          else tcp_nagle_enable(_pcb);
        }
i havent dug into the code enough to locate where "flags" gets processed

me-no-dev, 2016-08-17:
the lib is not waiting for anything. It processes data when and as it comes. It has a periodic check on all client connections to see if too much time has passed since the last packet send and ack is not yet received, but that is also done asynchronously. Only possible scenario is to not get a disconnect even from the remote client connection loss and keep the request and client in memory, but that still needs 5 of those to overfill the server and stop answering.

closed 2016-09-23 by me-no-dev