me-no-dev/ESPAsyncWebServer #63: First ~2.5KByte is served for static files

reluxa, 2016-08-01:
Hi me-no-dev, 

I have uploaded the ESP_AsyncFSBrowser to my [nodemcu](http://www.aliexpress.com/item/V3-Wireless-module-NodeMcu-4M-bytes-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266-for/32460412205.html) but it looks like it struggles sending large files. 

[image]

[image]

I have tried it with many different client (IE, firefox, chrome) but it's the same. The same SPIFFS content seems to work the default arduino webserver, but it's terribly slow. I have tried both gzipped and non gzipped version of the file but it's the same.

I am using the master/head for both the ESPAsyncWebserver and ESPAsyncTCP.

me-no-dev, 2016-08-02:
I am serving quite large files here, so things should be fine. Your log shows that it can not find many files. Are you sure you have all files there and that your SPIFFS is ok? I have it tested with 2MB files up and down ;)

reluxa, 2016-08-02:
Thanks for your fast response. I am happy to hear that ESPAsyncWebServer is capable of handling even very large files. Then the problem must be in my setup. In order to rule out SPIFFS problems, I will add a second regular synchronous webserver to port 81 to see if it the download is ok with that. I will come back with the results.

reluxa, 2016-08-02:
I've investigated further. And it looks the issue is related to my wifi network at home. (I have two AP-s one of them is running OpenWrt and that is linked via wifi to another AP, when I've removed the openwrt AP then it started to work fine)

closed 2016-08-02 by reluxa

keep3r, 2016-08-08:
I have the same problem on a WIFI network based on multiple Ubiquiti UniFi AP-LR. It works with no problems on a Asus wifi router. Maybe someone sees this and knows a possble answer to this problem.

tbnobody, 2016-11-10:
I have the same problem. It does not work with Cisco 3602 AP's but works with iPhone as client.