me-no-dev/ESPAsyncWebServer #64: Case-insensitive header field names

bergos, 2016-08-07:
The [RFC 7230 3.2](https://tools.ietf.org/html/rfc7230#section-3.2) and also the old [RFC 2616 4.2](https://tools.ietf.org/html/rfc2616#section-4.2) define the field name as case-insensitive. [node-fetch](https://www.npmjs.com/package/node-fetch) sends the content-length header field in lower case. That means the web server doesn't go into [that if branch](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebRequest.cpp#L282). I tested it with a quick 'n dirty or condition. I would like to make a PR. I think a generic solution, calling `.toLowerCase()` after [this line](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebRequest.cpp#L268) would be the best solution, but that would also cause changes in many other places in the code and maybe even applications that use the web server library. What do you prefer?

me-no-dev, 2016-08-12:
this is a good point. I did put some thought on it, no clear resolution in my mind :) You have methods for getting, searching and adding headers and I am not sure how all that will play out and what is the best way to deal with it.

ClaudioHutte, 2016-08-12:
Even applications that use the web server library should deal with field names as case-insensitive, so providing the names in lower case only fashion could be useful. Not sure about it, though.

penfold42, 2016-08-13:
+1 for toLowerCase in the library.

It neatly forces some standards compliance on _my_ code

me-no-dev, 2016-08-27:
maybe equalsIgnoreCase should be used throughout the lib so case would not matter. I do not really want to force either situation.

me-no-dev, 2016-09-02:
Should be resolved in the above commit

closed 2016-09-23 by me-no-dev