me-no-dev/ESPAsyncWebServer #67: Cheat sheet for converting from standard esp web server library.

mmiscool, 2016-08-14:
I am trying to convert a rather large project from the standard esp web server library to this one.

I was kind of hoping there might be a list of equivalent functions one to the other to make porting easier. 

Things like server->arg("name") in the esp web server library and what there equivalent is in this one.

me-no-dev, 2016-08-15:
check the readme ;) it has everything documented. Main difference is that those functions are attached to the request object in the callbacks and not on the server. So instead of server.arg("blah") you have request->arg("blah") and so on

ClaudioHutte, 2016-09-21:
Hello mmiscool, just out of curiosity, why do you move from the standard web server library?

closed 2016-09-23 by me-no-dev