me-no-dev/ESPAsyncWebServer #73: Resetting authentication

AcuarioCat, 2016-09-16:
Is there a way to revoke authentication?
After authenticating with a login I need to 'sign out' and revoke the authentication but I can't see any way of doing it.

Thanks.

me-no-dev, 2016-09-16:
```request->requestAuthentication();``` on authenticated user will clear the authentication from the browser

AcuarioCat, 2016-09-17:
Works thanks :+1:

closed 2016-09-17 by AcuarioCat

GeorgeFlorian, 2019-05-09:
How does it work ?
How can you turn this OFF ?
I have a LOGIN and a SKIP button inside a HTML page.

This is the LOGIN:
```
  server.on("/register", HTTP_POST, [](AsyncWebServerRequest * request){
    int params = request->params();
    String values_user[2];
    for(int i=0;i<params;i++){
      AsyncWebParameter* p = request->getParam(i);
      if(p->isPost()){
          logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");            
          values_user[i] = p->value();            
        } else {
            logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
          }
    } // for(int i=0;i<params;i++)
    
    if(values_user[0] != NULL && values_user[0].length() != 0 &&
      values_user[1] != NULL && values_user[1].length() != 0) {
          File userWrite = SPIFFS.open("/user.txt", FILE_WRITE);
          if(!userWrite) logOutput((String)"ERROR_INSIDE_POST ! Couldn't open file to write USER credentials !");
          userWrite.println(values_user[0]);  // Username
          userWrite.println(values_user[1]);  // Password
          userWrite.close();
          logOutput("Username and password saved !");
          
          request->redirect("/config");
          // digitalWrite(LED, LOW);
    } else request->redirect("/register");  
  });
```

And this is the SKIP:
```
  server.on("/button", HTTP_POST, [](AsyncWebServerRequest * request){
    int params = request->params();
    String values_button = "";
    for(int i=0;i<params;i++){
      AsyncWebParameter* p = request->getParam(i);
      if(p->isPost()){
          logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");            
          values_button = p->value();            
        } else {
            logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
          }
    } // for(int i=0;i<params;i++)
    values_button.trim();
    if(values_button == "Skip") {
      request->requestAuthentication(); // this doesn't change a thing
    }
    request->redirect("/config");    
  }); 
```

This is the "/config" :
```
  server.on("/config", HTTP_GET, [](AsyncWebServerRequest *request){  
    if(!request->authenticate(http_username, http_password))
      return request->requestAuthentication();            
    request->send(SPIFFS, "/STA_configPage.html", "text/html");
  });
```

If I press SKIP it redirects me to config but still asks for a username and password.