me-no-dev/ESPAsyncWebServer #74: Change webserver port

AcuarioCat, 2016-09-16:
Is it possible to dynamically change the server port?
I want to allow configuration of the server and part of that is allowing the user to determine the port that is used.
I see that in the constructor for the server (which is a global variable) the port is hardcoded:
AsyncWebServer server(80);

How can I then change the port number in my Setup routine?

I tried adding the following in my startup (port = 8081 by the way):
	server = AsyncWebServer(port);
but as soon as I try to use the server (by sending an http request) my ESP crashes:
Exception (9):
epc1=0x402129ce epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000002 depc=0x00000000

Setting the port to 8081 in the initial constructor doesn't cause me any problem.

Thanks

me-no-dev, 2016-09-23:
the server is not made to be able to change the port currently and what you try above is also not supported. Will look into it though

anhvu100, 2017-06-12:
I used the following code and it seems to work on the first time.
    std::unique_ptr<AsyncWebServer> server;
    server.reset(new AsyncWebServer(port));
When I call server.reset(new AsyncWebServer(port)) again in the loop() to change the listen port, the ESP crashes. Do you have any idea?

anhvu100, 2017-06-13:
FYI... I just modified the ESPAsyncTCP.cpp and add this code and now it works:
void AsyncServer::end(){
  if(_pcb){
    tcp_arg(_pcb, NULL);
    tcp_sent(_pcb, NULL);
    tcp_recv(_pcb, NULL);
    tcp_err(_pcb, NULL);
    tcp_poll(_pcb, NULL, 0);
    tcp_accept(_pcb, NULL);
    if ( tcp_close(_pcb)!=ERR_OK ) {
      //cleanup all connections?
      tcp_abort(_pcb);
    }
    //tcp_arg(_pcb, NULL);
    //tcp_accept(_pcb, NULL);
    _pcb = NULL;
  }

me-no-dev, 2017-06-21:
@anhvu100 thanks! merged into AsyncTCP

closed 2017-06-21 by me-no-dev

thbl, 2019-09-15:
@me-no-dev @anhvu100 is this function removed ?
when i do
`server.reset(new AsyncWebServer(port));`
i get 
```
note:   candidate expects 0 arguments, 1 provided
exit status 1
no matching function for call to 'AsyncWebServer::reset(AsyncWebServer*)'
```

andrecortica, 2021-09-15:
> @ me-no-dev @ anhvu100 esta função foi removida?
> quando
> `server.reset(new AsyncWebServer(port));`
> eu consigo
> 
> ```
> note:   candidate expects 0 arguments, 1 provided
> exit status 1
> no matching function for call to 'AsyncWebServer::reset(AsyncWebServer*)'
> ```

Did you manage to solve this problem?

andykaaa, 2023-01-20:
Yes!, .... ESP32

...
std::unique_ptr <AsyncWebServer> server;
...
void setup() {
...
server.reset(new AsyncWebServer(new_port));
...

Account-Git, 2023-02-12:
I have the same problem, but I cannot get the above suggestions to work.
What am I doing wrong?
When I add "std::unique_ptr server;" I get an "invalid use of template name" error.
And "server.reset(new AsyncWebServer(new_port));" returns the same error that andrecortica has above.
Could someone please explain this more clearly?