me-no-dev/ESPAsyncWebServer #75: example needed for TLS

yhua537, 2016-09-16:
Do you have an example TLS with cer.cert and k.key are uploaded memory (with/without password) ?

yhua537, 2016-09-18:
Trying to setup some basic code to test, wondering why this doesn't work. 

`  AsyncWebServer server(443);                  // The Webserver`
`  server.onNotFound ( [](AsyncWebServerRequest *request) {`
`      Serial.println("Page Not Found: "+  request->url());`
`      for(int i=0; i< request->headers(); i++){`
`        Serial.println("server.header:"+ request->headerName(i)+ ":"+ request->header(i));`
`      }`
`    request->send ( 200, "text/html", "it works, but url not found" );`
`  }  );`
`  Serial.println( "HTTP server is starting" );`
`  server.beginSecure("server.cer", "server.key", NULL);`
`  Serial.println( "HTTP server is starting" );`

me-no-dev, 2016-09-18:
You need to provide callback to load the certificate :)
```cpp

  server.onSslFileRequest([](void * arg, const char *filename, uint8_t **buf) -> int {
    Serial.printf("SSL File: %s\n", filename);
    File file = SPIFFS.open(filename, "r");
    if(file){
      size_t size = file.size();
      uint8_t * nbuf = (uint8_t*)malloc(size);
      if(nbuf){
        size = file.read(nbuf, size);
        file.close();
        *buf = nbuf;
        return size;
      }
      file.close();
    }
    *buf = 0;
    return 0;
  }, NULL);
  server.beginSecure("/server.cer", "/server.key", NULL);
```

yhua537, 2016-09-19:
Thank you. That works.

closed 2016-09-19 by yhua537

ccoenen, 2018-02-17:
@me-no-dev this should also be in the readme, in my opinion. I first had the impression TLS wasn't supported. `beginSecure` isn't mentioned at all, neither are TLS or SSL.