me-no-dev/ESPAsyncWebServer #76: Problem uploading multiple files

TheOriginalMrWolf, 2016-09-26:
Hi,

First, thanks for all your work on an awesome set of libraries and other contribution to ESP arduino dev!!!

I've experienced a small issue trying to upload multiple files.

Single file upload based on your examples works great (for instance, upload "one.txt").  Chrome client on windows, simple html form as follows:

 `
<html>
<body>
<form action="/upload" method="post"
enctype="multipart/form-data">
<label for="file">Filename:</label>
<input type="file" name="file" id="file" ><br>
<input type="submit" name="submit" value="Submit">
</form>
</body>
</html>
`

Supported by this code:
`
void handleFileUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){
  static bool _authenticated;
  static String _username = "", _password="";

  dbg_printf("File: %s, size:%u bytes, index: %u, final: %s\n", filename.c_str(), len, index, final?"true":"false");
  if(!index){
    _authenticated = false;
    if(!_username.length() || request->authenticate(_username.c_str(),_password.c_str())){
      _authenticated = true;
      filename= "/" + filename;
      request->_tempFile = SPIFFS.open(filename, "w");
    }
  }
  if(_authenticated && request->_tempFile){
    if(len){
      request->_tempFile.write(data,len);
    }
    if(final){
      request->_tempFile.close();
    }
  }
}
`

However, if I try to modify the HTML to support selecting multiple files (so I can upload "one.txt" followed by "two.txt" in one go, eg: 
`
<html>
<body>

<form action="/upload" method="post"
enctype="multipart/form-data">
<label for="file">Filename:</label>
<input type="file" name="file" id="file" **multiple**><br>
<input type="submit" name="submit" value="Submit">
</form>
</body>
</html>
`

Then I get a fatal exception:
`
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4010011d, epc2=0x00000000, epc3=0x40000f68, excvaddr=0x40036c00, depc=0x00000000

Exception (3):
epc1=0x4010011d epc2=0x00000000 epc3=0x40000f68 excvaddr=0x40036c00 depc=0x00000000

ctx: sys
sp: 3ffffbf0 end: 3fffffb0 offset: 01a0
`
The debug output from the above sprintf is: File: two.txt, size:3 bytes, index: 0, final: true
... which is identical to what I see if I upload just one file.

Any suggestions as to what I'm doing wrong (apart from trying to upload two files :P )?

Thanks again!!

luc-github, 2016-10-12:
@me-no-dev I have same issue uploading multiple files
with normal esp8266 webserver I use '<input type="file"  id="file-select" name="myfiles[]" multiple />' and it handle every file, one by one
but using same code on async it crash after first one  :

```
Exception (3):
epc1=0x4010011d epc2=0x00000000 epc3=0x00000000 excvaddr=0x40033990 depc=0x00000000

ctx: sys 
sp: 3ffffbf0 end: 3fffffb0 offset: 01a0
```
[image]

is it a limitation of asyncwebserver or a bug ?

luc-github, 2016-11-29:
Hi I get this solved by reset _itemBuffer to null after free here: https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebRequest.cpp#L455
```
        free(_itemBuffer);
        _itemBuffer = NULL;
```

I was able to upload 8 files at once without crash - I hope it help

commit 5159d8b8 (2016-11-29): Fix #76 Problem uploading multiple files (#104)
closed 2016-11-29 by me-no-dev