me-no-dev/ESPAsyncWebServer #80: [Question]Yield/Delay cannot be used how to use SD library ?

luc-github, 2016-10-08:
Hi first thanks for this great library - excellent work.
I try to port my project using normal web server to use your async web server.

In readme there is : `You can not use yield or delay or any function that uses them inside the callbacks`

So how to workaround this ? like for native ESP8266 SD library or 3rd party SD Fat library, they use `optimistic_yield` https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/SD/src/utility/Sd2Card.cpp#L142 or `delay(0)` https://github.com/greiman/SdFat/commit/bc32f5afdd35f8344cc6953c99287b3dc4255667#diff-be566363cc70b580a34aa2032a13e4e1R56

I also use `delay` when waiting for serial data answer after sending command.

What is your suggestion to no break anything ?

Thanks in advance

me-no-dev, 2016-10-08:
here is something I have worked on before :) give it a go and use as example for SD
https://gist.github.com/me-no-dev/a1975076563ab7146f69f89df5ca27d8

luc-github, 2016-10-08:
Thanks a lot your are excellent, will use it

closed 2016-10-08 by luc-github

Misiu, 2018-04-27:
@me-no-dev thanks for link :)
I'm working on similar thing. I need to read from couple of DS18B20 and store results to SD every X minutes.  I want to use Your awesome library to build web UI that would allow users to download files with data.
I am bit concern about async access to SD - user can download file in same time when write should happen.
Should I even bother about that? I'm new to ESP8266, I know it is not multi-threaded, but Your library is somehow async :)
I have C# background and there it is easy, You do lock and it works.

I'm building prototype right now, but I'd like to know if this would a potential issue.