me-no-dev/ESPAsyncWebServer #82: Run websockets on different port than webserver

vjacob, 2016-10-12:
We designed our application to run websockets on a different port (81) than the normal webserver traffic (80). However, I can't seem to find out how to set the port for the ESPAsyncWebServer websockets. This doesn't seem possible or am I missing something?

Regards,
V

me-no-dev, 2016-10-12:
Every WebSocket starts as a regular Web GET client, so WebSockets here do not need separate server and everything runs on port 80 or whatever you start your server on. To have WebSockets on another port, you need to start another AsyncWebServer on that port and just have WebSockets there but that does not make much sense.

vjacob, 2016-10-12:
Thank you for the swift reply.

closed 2016-10-12 by vjacob