me-no-dev/ESPAsyncWebServer #83: Limit of Requests

mafe, 2016-10-18:
At first, thanks for that lib. It does a wonderful job so far.

I am struggling with 404 errors sometimes, especially with Chrome.

The WebServer serves via SPIFFS 10 files, when index.html has been requested.
Chrome is very aggressive with its limit (I think it is around 8 in parallel).

Is it possible to limit the requests by the WebServer?

me-no-dev, 2016-10-18:
lwip limits connections to 5 so you can not have more than 5 concurrent requests.
I advise you to combine what files you can (js / css) and gzip them. If some files are static and will not change, there is a way to put them gzipped in progmem and serve from there (faster). Current FSBrowser example is running from progmem and the result is created with ```gzip edit.htm``` and then passed through [this](https://gist.github.com/me-no-dev/f137a950ce6dedb641d427d8db6355d2) to output the progmem array for your code.

mafe, 2016-10-19:
I thought of something like declining any further requests, if there are already five requests ongoing and force the webbrowser to retry.
But anyway, I try to reduce the files to serve.

Thanks you.

closed 2016-10-19 by mafe