me-no-dev/ESPAsyncWebServer #85: WiFi.scanNetworks() always return 0 or -1 during handling of request

yhua537, 2016-10-23:
Hi 

Following code always return 0 or -1. 

  server.on ( "/scan", HTTP_ANY, [](AsyncWebServerRequest * request) {
    Serial.println( "sending scanWiFi" );
    request->send ( 200, "text/plain", scan(request));
  });
 and 

String scan(){
    return "no of connection: "+ (String)WiFi.scanNetworks();
}

Is it because it is unable to scan WiFi during responding to request? the same code works scan() is calling from loop directly.

rakeshpai, 2016-11-05:
I'm facing this issue as well. If it's any help, it seems to return -1 only for the second call, if two `WiFi.scanNetworks` calls are made in quick succession. If there's enough "cool-down" time, the result is always 0.

me-no-dev, 2016-11-05:
```cpp
  server.on("/scan", HTTP_GET, [](AsyncWebServerRequest *request){
    String json = "[";
    int n = WiFi.scanComplete();
    if(n == -2){
      WiFi.scanNetworks(true);
    } else if(n){
      for (int i = 0; i < n; ++i){
        if(i) json += ",";
        json += "{";
        json += "\"rssi\":"+String(WiFi.RSSI(i));
        json += ",\"ssid\":\""+WiFi.SSID(i)+"\"";
        json += ",\"bssid\":\""+WiFi.BSSIDstr(i)+"\"";
        json += ",\"channel\":"+String(WiFi.channel(i));
        json += ",\"secure\":"+String(WiFi.encryptionType(i));
        json += ",\"hidden\":"+String(WiFi.isHidden(i)?"true":"false");
        json += "}";
      }
      WiFi.scanDelete();
      if(WiFi.scanComplete() == -2){
        WiFi.scanNetworks(true);
      }
    }
    json += "]";
    request->send(200, "text/json", json);
    json = String();
  });
```

rakeshpai, 2016-11-05:
Thanks, @me-no-dev. I tested this, and it works.

However, the first time /scan is called, it's always an empty array `[]`. It's the subsequent requests that give any useful response. Even so, it appears that the response is always the result of a previous scan if I follow the logic above correctly (I could be wrong - I'm a n00b), and so the results are always a little outdated.

Is there any way around these problems?

PS: I see that there's a [`WiFi.scanNetworksAsync`](https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/scan-class.md#scannetworksasync) documented, but I can't seem to find this call in the actual library. I'm guessing that should be a much more elegant fix when it lands, but I can't seem to get my code to compile with the call right now.

me-no-dev, 2016-11-05:
logic is pretty simple and if you think about it that is how it's supposed to be. If you want to have full result the first time, call ```WiFi.scanNetworks(true);``` in ```setup()```.

- Check status of wifi scan
- If scan results are empty, start new scan and return empty result
- if scan is available, get all results, empty them and start new scan
- if scan is running return empty result

me-no-dev, 2016-11-05:
true async scan can work only through EventSource or WebSocket.

rakeshpai, 2016-11-05:
Thanks for the reply, @me-no-dev 

I still disagree that this is the best possible result. It definitely works as an approximation, if outdated scan results are acceptable (which it might be in most cases).

If I understand the logic correctly, `/scan` results are only ever returned if there's a previous scan result to return. While a call to `/scan` would start a new async scan, it's never these new results that are returned. You'd have make a second call to `/scan` to get the results of the last scan.

This is why the first hit is an empty array. It's because `/scan` schedules a new scan, and returns the result of the previous scan, which has never been performed in this case, and is hence an empty array. However, the expectation is that `/scan` would do a scan immediately, and respond with the results of this scan, even if that means that the user's request is held waiting for this period. However, just because the user's request is kept waiting doesn't imply a `delay` or any other form of blocking call, and could still be perfectly async. While EventSource or WebSockets will be interesting ways to stream the results, they are essentially just fancy transports, and don't themselves imply that the underlying call is async.

For example, one possible way to get the expected behaviour right now in the sketch (untested) is to call `WiFi.scanNetworks(true);` in the handler for `/scan`, and in the `loop` check if the scan is complete with `WiFi.scanComplete()`, and respond with the results of the scan from within the loop. However, this is ugly, since it means that the request object will somehow have be to made available outside the request handler for `/scan` and inside the `loop` (locality and scope would be completely messed up), and will require some ugly book-keeping to deal with corner-cases (eg. multiple in-flight scan requests). However, it will be a proper (according to my understanding at least) async scan that performs the scan when the request comes in, keeps the user waiting without blocking the ESP (even letting it serve other incoming requests during the scan), and returns with the freshest results of the scan when it's available.

I hope I'm correct with the assessment above. (If I'm wrong, I'll be happy to be corrected.) I'm not sure there's an easy solution, except hoping that `WiFi.scanNetworksAsync` becomes available soon. I guess your code above is the best we could do until then.

me-no-dev, 2016-11-05:
I think you misunderstand Async ;) You can not wait in async! So obviously you can not wait for the scan to finish ;) If you are going to hold to the request while executing the scan... well what's the point of async in this case?

me-no-dev, 2016-11-05:
and when you say old data... how old you think the data will be if you call it every 5-10 seconds? does it matter that it's 5 seconds old?

rakeshpai, 2016-11-05:
You may be right - maybe I don't understand things right. However, please bear with me so I understand this correctly.

By using async, we are trying to avoid the processor from waiting, not necessarily the user. If async is just about responding immediately without delays, we would just get very quick results which aren't necessarily complete or correct.

I've done a lot of async programming in node.js, so I'll use an example from there. Let's say on the desktop environment, a user wants to seek a file from disk. In an async world, the processor would ask the disk to seek, and then go off to do other things. That's what makes it async - that the processor can do other things without waiting for the disk. When the disk has a result, it calls back. The user though is definitely waiting for a result while all this happens. It's the processor's waiting that we have avoided with async, not the user's. If the response is returned without waiting for the disk seek to complete, well, what's the point of doing the disk seek at all?

Holding on to a request doesn't break async. On the contrary, it allows you to respond at a later time when you have results. If we were to implement `/scan` with `WiFi.scanNetworksAsync`, the scan class would call back when it has results. The ESP wouldn't be wasting cycles waiting for the scan to finish, and would have been off to handle other business like serving other requests. If you use an inline lambda to define the callback for `.scanNetworksAsync`, you would still have a reference to the request object, which has been held on to because of lexical scope, making for very elegant code. So, it's C's scoping rules that holds on to the request object instead of us doing it explicitly as I suggested earlier. The 'holding on to the request' bit is necessary, and doesn't make it synchronous at all. In fact, this behaviour can't guarantee that responses will in the order that you made requests in, which is what makes it "asynchronous" to start with.

With the approach you're suggesting, I'll have to poll `/scan` at least twice in somewhat quick succession, discard the first result and use the second to be sure that I have the right result. How long should I wait between the two requests? I have no way of knowing - it's the server that knows when it has a result. I've written many async servers in node.js, and this behaviour would definitely be considered a bug, and a usability and documentation issue. If this behaviour is imposed because the server is async (which I know from node that it isn't), that would result in a pretty poor experience for consumers of an async server's API.

Sorry for the long response. All that said, maybe we just mean different things with our use of the terms 'async' and 'waiting'?

me-no-dev, 2016-11-05:
it's probably more of you that you are accustomed to node.js and JavaScript in general. We are not on a computer and resources here are really really limited. Same goes for number of open connections and so on. If we are purely talking about "cpu doing other things while we are waiting for things to be done" then the regular web server does that same thing in general. your loop is not running but all your interrupts, timers and so on are.
If you want to hold to the request, please be my guest :) beware though that if something happens to the connection it will be freed without you knowing and you will end up in memory exception. If you are interested in making it happen, check the source for the WebSockets and EventSource where both detach the Request from the server and manage it themselves from then on. You can also create your own response object that will not close the connection while it's waiting for something, then send the data and be done.

rakeshpai, 2016-11-05:
Thanks for the response @me-no-dev. It may be my node.js familiarity getting in the way. I'll look at the stuff you mentioned. It looks like it should set me on the right track.

If I can bother you with one last thing: Assuming that `.scanNetworksAsync` works, would the following code be valid? In particular, would `request` be available in the innermost function without memory exceptions?
```cpp
server.on("/scan", HTTP_GET, [](AsyncWebServerRequest *request) {
  WiFi.scanNetworksAsync([](int numNetworks) {
    // construct json by iterating over networks
    request->send(...);
  });
});
```

me-no-dev, 2016-11-05:
you should capture the request
```cpp
server.on("/scan", HTTP_GET, [](AsyncWebServerRequest *request) {
  WiFi.scanNetworksAsync([request](int numNetworks) {
    // construct json by iterating over networks
    request->send(...);
  });
});
```

me-no-dev, 2016-11-05:
and in this case if the client/request closes before you have scanned the wifi, the request will be cleared by the server and you will get exception

rakeshpai, 2016-11-05:
That makes sense. Thanks for taking the time to explain this.

yhua537, 2016-11-08:
Thanks guys, I have been doing a workaround by setting a flag and response in next request. scanNetworksAsync would be a better solution once that's available.

DrP2016, 2017-02-13:
Hello people,

i just found and read this 'issue'. Im facing the same problem, and i am able to get a get callback from WiFi.scanNetworksAsync(...), that builds my JSON String. But i am not able to get the request inside my callback, so i can send the JSON String when the scan is finished.

Im currently declaring this function before setup():
```
void WLANscanAsyncCB( int8_t networks ){
  WLANscanJSONstring = WLANscanJSON( networks );
  WiFi.scanDelete();
  WLANscanHandler.send(200, "text/plain", WLANscanJSON( networks ) );
}
```
And i use this inside the setup():
```
auto WLANscanHandler = server_async.on("/config/wifiscan",HTTP_GET,[](AsyncWebServerRequest *request){
  WiFi.scanNetworksAsync(WLANscanAsyncCB,false);  
});

server_async.addHandler(&WLANscanHandler);
```

i also tried to set a flag in `WLANscanAsyncCB` and use the `WLANscanHandler` in loop().

Both gives me the error, that `WLANscanHandler ` is not declared. I do understand that the Handler isnt available in the Callback, because the function is outside the setup(), but why cant i use it in the loop()?

me-no-dev, 2017-02-19:
```server_async.on``` already adds the handler. You need to find a way your callback to know the request, else I do not see how you will reply. ```WLANscanHandler``` is just a reference to a function. You can not send from it :)

DrP2016, 2017-02-20:
Hello me-no-dev,
i already looked through the code in the meantime and came to the same conclusion, that this doesn´t make sense. I wasn´t able to get the request inside my `WLANscanAsyncCB`. I currently have a working solution by just polling every 500ms from my Page until the Scan is done.
 Could you help me getting the request inside my Callback?

me-no-dev, 2017-02-24:
use EventSource or WebSocket ;) detaching a request from the server and responding later is not a good idea, though doable.

omersiar, 2017-06-22:
In my [project](https://github.com/omersiar/esp-rfid/) i send a web socket command to init a scan then after a couple of seconds ESP replies with a JSON object on WebSocket. then I manipulate HTML with DOM Javascript.

```arduino
// Send Scanned SSIDs to websocket clients as JSON object
void printScanResult(int networksFound) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["command"] = "ssidlist";
  JsonArray& data = root.createNestedArray("ssid");
  for (int i = 0; i < networksFound; ++i) {
    // Print SSID for each network found
    data.add(WiFi.SSID(i));
  }
  size_t len = root.measureLength();
  AsyncWebSocketMessageBuffer * buffer = ws.makeBuffer(len); //  creates a buffer (len + 1) for you.
  if (buffer) {
    root.printTo((char *)buffer->get(), len + 1);
    ws.textAll(buffer);
  }
  WiFi.scanDelete();
}
```

```arduino
if (strcmp(command, "scan")  == 0) {
        WiFi.scanNetworksAsync(printScanResult);
}
```

J6B, 2017-11-20:
Hi, @me-no-dev 

Using this kind of code (see bellow), you wrote that if client closes connection before answer, we should get exception (that's what i'm getting yes ;-) )
BUT does there is a way to verify connection is still opened (using request object) just before doing send and so avoid exception?

thanks

```
server.on("/scan", HTTP_GET, [](AsyncWebServerRequest *request) {
  WiFi.scanNetworksAsync([request](int numNetworks) {
    // construct json by iterating over networks
    request->send(...);
  });
});
```

me-no-dev, 2017-11-20:
like ```request->client()->connected()```?

J6B, 2017-11-20:
mm, not sure about that.
in the capture of lambda, we are taking request pointer by value, OK
But if client disconnect, I assume that request variable target is destoyed.
So, even when trying to call ->client() will do exception.
Do you know a way to check that target of request pointer is still alive?

simap, 2019-01-26:
> use EventSource or WebSocket ;) detaching a request from the server and responding later is not a good idea, though doable.

> `request->client()->connected()`

wasn't working for me (crashes).

This is what I came up with, and seems to work well. It handles simultaneous requests efficiently (if for whatever reason that happened), and it handles if one of them drops during the scan. I'm also using ArduinoJson to avoid any issues in case someone uses double quotes or backslashes in the ssid.

With this, I can make a single request that always returns the freshest scan data when complete.

This is in a larger RequestHandler class I'm working on, extracted here for readability (I may have missed a reference to two). 

```c++
#include <list>
std::list<AsyncWebServerRequest *> scanRequests;

void handleWifiScan(AsyncWebServerRequest *request) {

    //keep track of this request for later response, and forget it if it disconnects
    request->onDisconnect([request, scanRequests]() {
        scanRequests.remove(request);
    });
    scanRequests.push_back(request);

    //start a scan if one is not already running
    if (WiFi.scanComplete() != WIFI_SCAN_RUNNING) {
        WiFi.scanNetworksAsync([&](int numNetworks) {
            //now that the scan is complete, send the results to all scanRequests
            DynamicJsonBuffer buffer;
            JsonArray &array = buffer.createArray();
            for (int i = 0; i < numNetworks; ++i) {
                JsonObject &item = array.createNestedObject();
                item["rssi"] = WiFi.RSSI(i);
                item["ssid"] = WiFi.SSID(i);
//                item["bssid"] = WiFi.BSSIDstr(i);
//                item["channel"] = WiFi.channel(i);
//                item["secure"] = WiFi.encryptionType(i);
//                item["hidden"] = WiFi.isHidden(i);
            }
            WiFi.scanDelete();
            String res;
            array.printTo(res);
            buffer.clear();
            for (auto &r : scanRequests) {
                r->send(200, "application/json", res);
            }
            scanRequests.clear();
        });
    }
}
```

closed 2019-10-05 by stale[bot]

guestisp, 2022-04-30:
guys, i'm trying to use async websocket, where a connected client send a command for a VERY long operation, then when the operation result is ready, i have to send back the result to the client.

any idea? trying to use a simple wifiscan trigger the watchdog. putting the wifi Scan in a new thread , fix the Watch dog but Guru Meditation 1 triggers trying to write to the websocket (i'm passing a struct with a pointerr to AsyncWebSocketClient inside to the task lambda)

any idea?

BrianALowe, 2022-04-30:
I avoid this problem by using a Ticker one-time callback to make socket messages independent of other server processing
1. Client sends a message to start the long operation
2. Server gets the message and does a one-time Ticker callback:
triggerAction.once_ms(500, longOperation);
and socket message processing is done
3. When complete, longOperation sends a socket message to client that operation is done, or a message with data from the operation

guestisp, 2022-04-30:
Yes, i do something similiar because the websocket is just one of the 3 transport i'm using (the same operation code is shared between websockets, esp-now and mqtt).
So:

1. client sends a message to start the long operation
2. server start the operation in a FreeRTOS task
3. when complete, the freeRTOS task reply to the client

but step 3 dies because even passing the websocket instance to the task, when trying to use that class, a guru meditation 1 is triggered.

Are you referring to the Ticket class ? if I understood properly, with `triggerAction.once_ms()` you are delaying 500ms the `longOperation`, right ? But from `longOperation` how do you reply through the websocket ?

Could you please post some code ? I'm struggling from a week......

Belleson, 2022-04-30:
I don't have code of a complete operation. This code will not compile, but the idea is here in **bold**
#include <ArduinoJson.h>
#include <Preferences.h>
#include <Ticker.h>

#define DEBUG_OUTPUT true
#define STATUS_MONITOR true
#define WIFISAVED "wifiSaved"
#define WIFISID "wifisid"
#define WIFIKEY "wifikey"
#define BUFSIZE 64

bool wifiSaved;
Preferences prefs;
AsyncWebSocket socket("/ws");
**Ticker triggerAction;**

void notifyClients(const char *key, const char *data)
{
#if MONITOR_STATUS
	Serial.printf("\nENTERING notifyClients()"); 
#endif
#if DEBUG_OUTPUT
	Serial.printf("\nkey: %s data: %s", key, data);
#endif
	const uint8_t size = JSON_OBJECT_SIZE(8);
	StaticJsonDocument<size> json;
	json[key] = data;

#if DEBUG_OUTPUT
	Serial.printf("\n");
	serializeJsonPretty(json, Serial);
#endif
	char buffer[2048];
	size_t len = serializeJson(json, buffer);
	socket.textAll(buffer, len); // Send message to all socket clients
}

void longOperation()
{
#if MONITOR_STATUS
	Serial.printf("\nENTERING longOperation()");
#endif
	delay(60000); // Bogus long operation

	// Notify clients that operation is complete or send data
	**notifyClients("result", "longOperationDone");**
}

void handleSocketMessage(void *arg, uint8_t *data, size_t len)
{
// Maybe it's better to avoid Serial.print in message processing due to time?
#if MONITOR_STATUS
	Serial.printf("\nENTERING handleSocketMessage()");
#endif

	AwsFrameInfo *info = (AwsFrameInfo *)arg;
	if (info->final && info->index == 0 && info->len == len && info->opcode == WS_TEXT)
	{
		const uint8_t size = JSON_OBJECT_SIZE(8);
		StaticJsonDocument<size> json;
		DeserializationError err = deserializeJson(json, data);
		if (err)
		{
			Serial.printf("\ndeserializeJson() failed with code %s", err.c_str());
			return;
		}

		const char *action = json["action"];
		if (strcmp(action, "resetwifi") == 0)
		{
			wifiSaved = false;
			prefs.remove(WIFISAVED);
			prefs.remove(WIFISID);
			prefs.remove(WIFIKEY);

			**// AsyncTCP has a fast watchdog timer and will fail and reboot ESP32 before
			// some operations are complete. Set a ticker (timer) to trigger a callback
			// some milliseconds later. This allows a socket message call to complete
			// and return before the watchdog barks and bites. 500msec delay is probably too much
			triggerAction.once_ms(500, longOperation);**
		}
	}
}

guestisp, 2022-04-30:
thank you.
can i use a lambda with some captures as ticker callback?

Belleson, 2022-04-30:
> thank you. can i use a lambda with some captures as ticker callback?

I'm not sure -- using a Ticker allows handleSocketMessage to complete quickly. If a lambda becomes inline code that executes inside handleSocketMessage then longOperation makes it slow again. Anyway IMO lambdas offer no execution benefit and make code harder to read and debug.

guestisp, 2022-04-30:
But how do you notify just the client that asked for the results ? In your code you are notifying all clients, I need to notify just one

guestisp, 2022-04-30:
and my issue still exists: how can i pass arguments to longOperation based on the request?

my longOperation need to know the client and some arguments . without these, asyncwebsocket is3unusable due to the very shot watchdog and lack of support for callbacks

Belleson, 2022-04-30:
> and my issue still exists: how can i pass arguments to longOperation based on the request?

Maybe in message handler you can store the arguments in global variables?

> my longOperation need to know the client and some arguments . without these, asyncwebsocket is3unusable due to the very shot watchdog and lack of support for callbacks

a client object is passed into onSocketEvent, which can also be stored for later use

guestisp, 2022-04-30:
what happens if i store the client, when a second one is connected? the class i'm using is not spawned every time, so if i store the id, It Will be the same for all

Belleson, 2022-04-30:
> what happens if i store the client, when a second one is connected? the class i'm using is not spawned every time, so if i store the id, It Will be the same for all

If there are multiple concurrent messages, you'll need a queue to save and process them "per client"

guestisp, 2022-04-30:
like this? https://github.com/me-no-dev/ESPAsyncWebServer/issues/85#issuecomment-457866545

Belleson, 2022-04-30:
Yes, similar to this with a list of struct containing
	AsyncWebSocketClient *socketclient,
	void *arg,
	uint8_t *data,
	size_t len

guestisp, 2022-04-30:
still not clear how to differentiate clients and request without using any parameter in the function

even setting a list of request, i have to use the ticker and the same issue happens: the method called doesn't know the client

Belleson, 2022-04-30:
I think you are confused about http request versus client socket message

guestisp, 2022-04-30:
could you please post an example?

Belleson, 2022-04-30:
Sorry no, you can research this. Basically

- webserver.on(path, Get/Post/etc, callbackFunction) calls _callbackFunction_  with HTTP request object argument when web server receives http request from browser

- onSocketEvent(AsyncWebSocket *socketserver,
				   AsyncWebSocketClient *socketclient,
				   AwsEventType type,
				   void *arg,
				   uint8_t *data,
				   size_t len)
 is called when websocket server receives a message from websocket client. My suggestion was to make a Ticker callback from this event for socket messages.

I spent 6 months to understand how this works, it's not simple. Good luck!

guestisp, 2022-04-30:
but if i have to call and external function, without parameters, or the Watch dog kick in, it's impossible to use a One to One connection

guestisp, 2022-05-01:
> My suggestion was to make a Ticker callback from this event to process socket messages.

that's what i'm dosing, but without any reference to the event inside the ticker callback, is impossible to know to which client i have to reply, i can only reply to all

Pablo2048, 2022-05-01:
Why don't you start new question/issue and hijack this one? The solution is simple IMHO - websocket `.onEvent` method has `AsyncWebSocketClient *` client argument so you can easily get `clientid`, Ticker callback can have any argument (is templated). So I don't see any problem here...

guestisp, 2022-05-01:
because i had the same issue wrote here.

and no, passing the client id won't work, as soon as i try to reply to the client, the mcu panic

Pablo2048, 2022-05-01:
It definitely work - I'm using it. So you did something wrong...

guestisp, 2022-05-01:
could you please post a sample?

Pablo2048, 2022-05-01:
Sorry, but no - it's just waste of my time. If you need help, then open new issue, create MCVE, put it there and then we can help.