me-no-dev/ESPAsyncWebServer #87: Memory Leak with Websockets

rmeister, 2016-11-06:
Love this library, speed is really great with the async server. I'd love to use async websockets too, but I think there is a memory leak. [Here is my code](https://gist.github.com/rmeister/81510b5c522a4e8a8b8e04be7e3f26fa). 

When receiving (much JSON) data, heap goes down until there is a wdt. What I just observed is that disconnecting and then reconnecting will free the heap.

Cheers,
Richard

me-no-dev, 2016-11-06:
Your sketch looks perfectly fine. I have tested for leaks before and found none. You say to test this I should connect using WebSocket to the Async server and pass decent amount of data to it and look at the heap?

rmeister, 2016-11-07:
The sketch hosts a page that includes a HTML range input element. When I move the slider, for every step a string with the current value is sent via websocket to the server. So when I drag the input element up and down a few times and then check the heap, it is significantly decreased.

bkp7, 2017-01-16:
I also love this library but also encountered what is probably the same issue.

When passing data the heap goes down but is freed when disconnecting. After some investigation I think the data is being buffered in the Websocket code when the client can't keep up with the rate of transmission. I'm attaching a simple sketch and simple html web page. I'm using Chrome on Windows 10.
[WebsocketsLeakTest.zip](https://github.com/me-no-dev/ESPAsyncWebServer/files/708348/WebsocketsLeakTest.zip)

If I am correct it would seem that a 'slow' client could cause the server to crash by forcing it to queue/buffer unsent messages.  I don't know how websockets are meant to work but it seems to me there needs to be some way to manage the queue. In my current use case each message effectively makes all previous messages obsolete so being able to clear the queue before adding a new message would be great. However other use cases my need the queue to be sent. In this case maybe the size of the queue needs to be limited somehow to prevent crashing.

closed 2019-10-05 by stale[bot]

Oleg-Perevyshin, 2023-09-30:
Yes, I have the same problem.
I call the network scan and transmit the found access points.
And with each request the heap becomes smaller.
As soon as I comment out the output line to the `websocket ws.textAll(jsonString);` - everything is getting better.

Tell me, will there be a fix for this problem?