me-no-dev/ESPAsyncWebServer #89: Async Client connecting to Async Web Server, server stop responding after 5 requset

yhua537, 2016-11-08:
Hi 

Master version from 7 Sep seem to have issue when async client have 5 open connection with async server. The server responds to ping but stops responding to any http request including if the request from browser/telnet. 

Is there an issue where both server and client keeps the connection therefore the 6th connection request is dropped? Not sure if that's related to this discussion [https://github.com/esp8266/Arduino/issues/2569](url). 

To fix the problem, it seems like there be a callback/function for both the AsyncClient and AsyncWebServer be able to actively close connections once either side thinks the communication is completed. I did find AsyncClient::close(true), but need more time to test it as it previously causing esp to reset. Is there better function than this? I wasn't able to find similar function with server. 

Thank you in advance.

me-no-dev, 2016-11-08:
5 open connections is limitation imposed by LwIP and not by my code. The server sends header "Connection: close" which tells the browser on the other end to actively close the connection once it receives the whole response.

yhua537, 2016-11-19:
My issue was between ESPs, seems to work if I add the header for all responses.

closed 2019-10-05 by stale[bot]