me-no-dev/ESPAsyncWebServer #92: Editor not working in AP-Mode

sleepingshorty, 2016-11-15:
Hi.

I want to use AP-Mode with your great webserver. But it tries to receive external content. Could you please add the missing content into the zip that is implemented in the code?

I tried to add it, but I failed.

Thank you for your help.

Kind regards,
Sleepingshorty

me-no-dev, 2016-11-16:
are you talking about the FSBrowser plugin? That one loads quite a few files from the internet and it's not easy to have them all on the flash ;) they are big

sleepingshorty, 2016-11-16:
Yep the Plugin.

I have 1.8MB in SPIFFS left. I think it should fit. But I don't know how to generate that byte-html-gz-file...

me-no-dev, 2016-11-16:
open it through STA interface, right click -> view page source :) that will give you the source code of the page. Download it and get all of the required files for local install. Once you get it running like that, you can first gzip the file and then use [this](https://gist.github.com/me-no-dev/f137a950ce6dedb641d427d8db6355d2) to turn it into the byte array that you see for progmem

closed 2019-10-05 by stale[bot]