me-no-dev/ESPAsyncWebServer #94: Bad request (400) on WebSocket

HanVertegaal, 2016-11-18:
I'm trying to open a WebSocket on the ESP from a .NET program:

`_ws = new ClientWebSocket();
 _ws.ConnectAsync(new Uri($"ws://{_address}/ws"), CancellationToken.None).Wait(_timeout);`

Wireshark captures the following request:

> Hypertext Transfer Protocol
    GET /ws HTTP/1.1\r\n
        [Expert Info (Chat/Sequence): GET /ws HTTP/1.1\r\n]
        Request Method: GET
        Request URI: /ws
        Request Version: HTTP/1.1
    Connection: Upgrade,Keep-Alive\r\n
    Upgrade: websocket\r\n
    Sec-WebSocket-Key: F4aMDIKjQx7YQnvW9n8JrA==\r\n
    Sec-WebSocket-Version: 13\r\n
    Host: 192.168.178.31\r\n
    \r\n
    [Full request URI: http://192.168.178.31/ws]
    [HTTP request 1/1]
    [Response in frame: 5]

The response is a bad request:

> Hypertext Transfer Protocol
    HTTP/1.1 400 Bad Request\r\n
        [Expert Info (Chat/Sequence): HTTP/1.1 400 Bad Request\r\n]
        Request Version: HTTP/1.1
        Status Code: 400
        Response Phrase: Bad Request
    Content-Length: 0\r\n
        [Content length: 0]
    Connection: close\r\n
    Accept-Ranges: none\r\n
    \r\n
    [HTTP response 1/1]
    [Time since request: 0.004665000 seconds]
    [Request in frame: 4]

I can see that AsyncWebSocket::canHandle returns true, but in AsyncWebSocket::handleRequest there are no headers in the Request (request->headers() == 0). Why aren't the headers copied from the message?

me-no-dev, 2016-11-19:
I have no idea! is everything else working as expected? or are you just using the WebSocket?
You can see that canHandle request a list of headers to be collected and those include the headers that handleRequest needs.
Can you put some debug [here](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebRequest.cpp#L297) and see what headers pass through and confirm that the server sees the headers as "interesting"?

HanVertegaal, 2016-11-19:
Thanks for your fast response. The AsyncWebServer also serves web pages and a REST API, and these are working fine. I can see the interesting headers being added in canHandle, and _interestingHeaders->length() returns 6 at that point. But then in _parseRequestHeader,  _interestingHeaders->length() returns 0 !? It almost looks as if the request gets re-instantiated somewhere.
There is around 21K free RAM available

The server setup looks as follows:

`
void HttpServer::Init()

{
    _webServer = new AsyncWebServer(80);
    _webSocket = new AsyncWebSocket("/ws"); // access at ws://<esp ip>/ws
    _webSocket->onEvent(\[&\](AsyncWebSocket* server, AsyncWebSocketClient* client, AwsEventType type, void* arg, uint8_t* data, size_t len) {
        OnWebSocketEvent(server, client, type, arg, data, len);
    });
    _webServer->addHandler(_webSocket);
    _webServer->on("/api/echo", \[&\](AsyncWebServerRequest* request){ HandleEcho(request); });
    _webServer->on("/api/netinfo", HTTP_ANY, \[&\](AsyncWebServerRequest* request){ HandleNetinfo(request); });
    _webServer->on("/api/netscan", \[&\](AsyncWebServerRequest* request){ HandleNetscan(request); });
    _webServer->on("/api/distance", HTTP_ANY, \[&\](AsyncWebServerRequest* request){ HandleDistance(request); });
    _webServer->on("/api/adc", HTTP_GET, \[&\](AsyncWebServerRequest* request){ HandleADC(request); });
    _webServer->on("/api/reset", HTTP_POST, \[&\](AsyncWebServerRequest* request){ HandleReset(request); });
    _webServer->on("/api", \[&\](AsyncWebServerRequest* request){ HandleAPI(request); });
    _webServer->serveStatic("/", SPIFFS, "/", "max-age=86400").setDefaultFile("index.html");
    // reply to all requests with same HTML
    _webServer->onNotFound(\[&\](AsyncWebServerRequest* request){ NotFound(request); });
    _webServer->begin();
}
`

HanVertegaal, 2016-11-19:
It seems that AsyncWebServerRequest::_parseReqHeader assumes that 'Host' is the first header, because only when processing Host are the 6 interestingHeaders added (in _attachHandler). Headers appearing before 'Host' are never detected as interesting. Would you agree that this is a correct interpretation?

HanVertegaal, 2016-11-20:
RFC 7230 states:

> Since the Host field-value is critical information for handling a
   request, a user agent SHOULD generate Host as the first header field
   following the request-line.

So clearly this is a problem in the .NET ClientWebSocket implementation.

closed 2016-11-20 by HanVertegaal

me-no-dev, 2016-11-20:
but I thought I had fixed that already. We had issue with some MS nonsense before and I remember working this out so Host does not have to be the first header hmmmm...

universam1, 2017-04-19:
Any idea how to solve that? Having the same issue with IE and Edge with the current master.
Thanks!

jeremypoulter, 2017-09-22:
Just come across this issue, is there any more info/guidance on how this can be resolved? Note the quoted RFC is a SHOULD which means it is just a recommendation that the client send the Host: header first. It is not mandatory so I think this is something that should be resolved within ESPAsyncWebServer so can this issue be re-opened? or should a new issue be raised?

HanVertegaal, 2017-09-23:
You can bypass the issue in .NET by using [WebSocket4Net](https://websocket4net.codeplex.com/)

me-no-dev, 2017-09-25:
Isn't this already resolved? I thought I merger a PR that fixes this?

jeremypoulter, 2017-09-30:
@me-no-dev Yes it is, was my mistake, Platform IO not doing what I expected.