me-no-dev/ESPAsyncWebServer #96: Example doesn't compile because Time.h

ssaguiar, 2016-11-19:
Your example doesn't compile:

I am using PlatformIO to compile:

Processing nodemcuv2 (platform: espressif8266, lib_deps: ESPAsyncTCP, board: n
odemcuv2, framework: arduino)
.
.
.
.
lib\ESPAsyncWebServer\src\WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticW
ebHandler::setLastModified(tm*)':
lib\ESPAsyncWebServer\src\WebHandlers.cpp:67:64: error: 'strftime' was not declared in this scope
strftime (result,30,"%a, %d %b %Y %H:%M:%S %Z", last_modified);

^
lib\ESPAsyncWebServer\src\WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticW
ebHandler::setLastModified(time_t)':
lib\ESPAsyncWebServer\src\WebHandlers.cpp:72:60: error: 'gmtime' was not declared in this scope
return setLastModified((struct tm *)gmtime(&last_modified));
^
lib\ESPAsyncWebServer\src\WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticW
ebHandler::setLastModified()':
lib\ESPAsyncWebServer\src\WebHandlers.cpp:77:25: error: 'time' was not declared in this scope
if(time(&last_modified) == 0) //time is not yet set
^
*** [.pioenvs\nodemcuv2\lib\ESPAsyncWebServer\WebHandlers.o] Error 1
 [ERROR] Took 97.28 seconds
 
Any ideas?

ruchuf, 2017-01-17:
I had similar issues with some incompatible time libraries.
If you use the preferred Time library (https://github.com/PaulStoffregen/Time) you can use
_#include Timelib.h_
instead of Time.h
The time library uses an header file with an uppercase T, while there are some other time.h with an lower-case t from other projects.
OSses like Windows and Mac don't make an difference there :-(

jamesmyatt, 2018-12-05:
I have this problem also, on windows only, not linux. I solved it by deleting the "Time.h" from the Time library (https://github.com/PaulStoffregen/Time), and using TimeLib.h only in my code.

igrowing, 2018-12-08:
This compilation problem happens in Windows only. Linux and Mac are case sensitive for files, so these OSes differ between `time.h` and `Time.h`. Windows doesn't.
The problem comes when `Time_ID44` lib in installed (that's often used lib) and `ESPAsyncWebServer_ID306` lib includes `time.h` in `WebHandlerImpl.h`.
So wrong time lib is included due to OS case insensitivity.
There is advice to include `TimeLib.h` instead of `time.h`. Tried this, doesn't work.
Workaround that works:
1. Rename the lib:
- Go to: <drive>:\Users\<username>\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include
- Rename `time.h` to `_time.h`.
2. Find all occurrences of `WebHandlerImpl.h` in your Windows.  Replace `#include <time.h>` with `#include <_time.h>`
3. Build your project. (Recommended restart platformio before the build).

closed 2019-10-05 by stale[bot]