me-no-dev/ESPAsyncWebServer #97: ESP8266 crashes on loading html application

mafe, 2016-11-21:
Sometimes (actually quite often), the ESP8266 crashes, when I load my html-Application.

`Line 127 of ".platformio\packages\framework-arduinoespressif8266\cores\esp8266\WString.cpp" starts at address 0x40202719 <String::String(String const&)+5> and ends at 0x4020271d <String::String(String const&)+9>.`

`
DEBUG:40204ms:ESP.getFreeHeap()<5000=4112 Bytes<\r><\n>
DEBUG:40205ms:ESP.getFreeHeap()<5000=4112 Bytes<\r><\n>
DEBUG:40206m:ESP.getFreeHeap()<5000=4112 Bytes<\r><\n>
DEBUG:40212ms:ESP.getFreeHeap()<5000=4112 Bytes<\r><\n>
DEBUG:42725ms:ESP.getFreeHeap()<5000=4760 Bytes<\r><\n>
DEBUG:42726ms:ESP.getFreeHeap()<5000=4192 Bytes<\r><\n>
DEBUG:42727ms:ESP.getFreeHeap()<5000=4192 Bytes<\r><\n>
`
The application consists of more than five files and some are big, but this should not lead to a crash, I guess. 

Is it possible, that the ESP-mainloop needs more memory, at this moment?

me-no-dev, 2016-11-21:
you probably have quite large application and given that you need to use separate memory for each request in parallel, you might end up in that situation yes. Try to combine + gzip files for the interface. That could reduce greatly the time it needs to send data out and might just be the ticket for success

mafe, 2016-11-23:
thanks for your fast response.

it is indeed the fact, that there are parallel requests and consume more memory.

Now, it is working with delivering a singlefile, that includes HTML,JS, Images.

I am using the following tools, in order to achieve this:
[Inliner](https://github.com/remy/inliner)
[Online binary to char converter](tools.garry.tv/bin2c/)

Thank you for making this great library. It is really fast, keeping in mind, that it runs on a microcontroller.

closed 2016-11-23 by mafe

me-no-dev, 2016-11-23:
great! :)