me-no-dev/ESPAsyncWebServer #99: An intermediate step before connecting ESP to a Wi-Fi in a station mode.

nikigter, 2016-11-23:
Hi, 
I want to make the following scenario.  Please tell me,  if I am doing it right. 

> The soft-AP mode is often used and an intermediate step before connecting ESP to a Wi-Fi in a station mode. This is when SSID and password to such network is not known upfront. ESP first boots in soft-AP mode, so we can connect to it using a laptop or a mobile phone. Then we are able to provide credentials to the target network. Once done ESP is switched to the station mode and can connect to the target Wi-Fi.

```

AsyncWebServer server(80); 

const char * hostName = "myHost.local";

IPAddress Ip(10, 1, 1, 10);
IPAddress NMask(255, 255, 255, 0);

void setup()
{
	WiFi.mode(WIFI_STA); 
	delay(500);
	Serial.begin(115200);
	
    SPIFFS.begin(); 
	Serial.println("Connecting..."); 
	
    WiFi.begin(); // instruct module to switch to the station mode and connect to the last used access point basing on configuration saved in flash memory
 	delay(500);

	uint8_t connResult = WiFi.waitForConnectResult();

	if (connResult == WL_CONNECTED)
	{
		Serial.print("STA IP: ");
		Serial.println(WiFi.localIP());
	}
	else
	{ 
		Serial.printf("STA: Failed!\n");

		if (connResult == WL_CONNECT_FAILED) // if password is incorrect
		{
			WiFi.disconnect(false); // Sets currently configured SSID and password to  null  values and disconnects the station from an access point. 
		}
 
		WiFi.softAPConfig(Ip, Ip, NMask); 
        
		WiFi.mode(WIFI_AP);
		WiFi.softAP(hostName);  
		Serial.print("Soft-AP IP address = ");
		Serial.println(WiFi.softAPIP()); 

		server.on("/save", HTTP_GET, [](AsyncWebServerRequest *request) {

			 // TODO: get params from the GET request

			WiFi.begin("xxxxx", "xxxxxx"); // for testing ...
 
			if (WiFi.status() == WL_CONNECTED) // Here is always FALSE!!!
			{
				request->send(200, "text/plain", "OK");
			}
			else
			{
				request->send(200, "text/plain", "Wrong");
			}		
            
		});
  
		server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.htm").setFilter(ON_AP_FILTER);
		server.begin();
	}
}

void loop()
{  
}
```

me-no-dev, 2016-11-24:
you do not need to wait for wifi to connect to start the server. Also in the code above, the server will not start if the WiFi just connects from the start.

nikigter, 2016-11-24:
Yes, I want the server to start only when the IP and password for WIFI_STA are not correct. 
I want in method **"_server.on("/save", HTTP_GET,....."_** to test if the  entered IP and Password from the user are correct and to switch to WIFI_STA. If not, then the user to try one more time.  The problem is that  **_if (WiFi.status() == WL_CONNECTED)_**  always returns false.  What happens  with **_request->send(..._**   when i switch to  WIFI_STA? I want the user to be notified for the result.

me-no-dev, 2016-11-29:
when you switch to WiFi_STA do you turn off AP? If so, you need to notify the user before changing mode. Like set a flag and respond with OK, then in loop check for the flag and switch mode.

closed 2019-10-05 by stale[bot]