me-no-dev/ESPAsyncWebServer #102: SpiffsEditor

eketjall, 2016-11-29:
I'm probably just stupid and missing something, but is the editor actually supposed to be able to edit files?
I get the file listing and kan view file content. 
I can also upload files. 
But how can I edit existing files? My first thought is that a save-button is missing.
I can also create new files, but the content of the file is always just a singel dot 

[image]
 .

me-no-dev, 2016-11-29:
yeah... i keep forgetting to add save button, but Ctrl+S on Windows/Linux and Cmd+S on Mac do the job of saving

eketjall, 2016-11-29:
So for creating a new file the steps is the following?
1 - create file
2 - edit file
3 - save file with Ctrl+S

me-no-dev, 2016-11-29:
yes, or open->edit->ctrl+s

hallard, 2017-01-16:
Here is the modified index.htm file extracted, with monokai theme and save button :-)
I'm always using file directly from SPIFFS system because when testing it's really easy doing changes (with the editor itself) and test, and almost if you use SPIFFSEditor, mainly you have SPIFFS system file to put file into. That's just a matter of preference of course, but If you want to update you need to gz file then create the C code to put into `SPIFFSEditor.cpp`

To use directly edit.htm file I just changed info `SPIFFSEditor.cpp` [lines](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.cpp#L386-L388)
by
```cpp
  //AsyncWebServerResponse *response = request->beginResponse_P(200, "text/html", edit_htm_gz, edit_htm_gz_len);
  //response->addHeader("Content-Encoding", "gzip");
  //request->send(response);
  //Send edit.htm with default content type
  request->send(SPIFFS, "/edit.htm");
```
and of course removed [edit_htm_gz](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.cpp#L5-L317)

**Editor looks like**
[image]

**Modified file index.htm**
```htm
<!-- This file has been written by Me-No-Dev
see https://github.com/me-no-dev/ and is part
of the sample code of ESPAsyncWebServer
see https://github.com/me-no-dev/ESPAsyncWebServer -->
<!DOCTYPE html>
<html lang="en">
<head>
<title>ESP8266 SPIFFS File Editor</title>
<meta name="author" content="Me-No-Dev" />
<meta name="Description" content="ESP8266 SPIFFS File Web Editor"/>
<link rel="shortcut icon" href="favicon.ico"/>
<style type="text/css" media="screen">
.cm {
  z-index: 300;
  position: absolute;
  left: 5px;
  border: 1px solid #444;
  background-color: #F5F5F5;
  display: none;
  box-shadow: 0 0 10px rgba( 0, 0, 0, .4 );
  font-size: 12px;
  font-family: sans-serif;
  font-weight:bold;
}
.cm ul {
  list-style: none;
  top: 0;
  left: 0;
  margin: 0;
  padding: 0;
}
.cm li {
  position: relative;
  min-width: 60px;
  cursor: pointer;
}
.cm span {
  color: #444;
  display: inline-block;
  padding: 6px;
}
.cm li:hover { background: #444; }
.cm li:hover span { color: #EEE; }
.tvu ul, .tvu li {
  padding: 0;
  margin: 0;
  list-style: none;
}
.tvu input {
  position: absolute;
  opacity: 0;
[... 519 lines cut ...]
    if(lang !== "plain") editor.getSession().setMode("ace/mode/"+lang);
    httpGet(file);
  };
  return editor;
}
function onBodyLoad(){
  var vars = {};
  var parts = window.location.href.replace(/[?&]+([^=&]+)=([^&]*)/gi, function(m,key,value) { vars[key] = value; });
  var editor = createEditor("editor", vars.file, vars.lang, vars.theme);
  var tree = createTree("tree", editor);
  createFileUploader("uploader", tree, editor);
};
</script>
<script src="https://cdnjs.cloudflare.com/ajax/libs/ace/1.2.3/ace.js" type="text/javascript" charset="utf-8"></script>
</head>
<body onload="onBodyLoad();">
  <div id="loader" class="loader"></div>
  <div id="uploader"></div>
  <div id="tree"></div>
  <div id="editor"></div>
  <div id="preview" style="display:none;"></div>
  <iframe id=download-frame style='display:none;'></iframe>
</body>
</html>
```

omersiar, 2017-05-15:
@hallard 

[image]

closed 2019-10-05 by stale[bot]