me-no-dev/ESPAsyncWebServer #103: Not possible to stop the server ?

denkovi1, 2016-11-29:
Hello,
Is it possible to stop the server, like server.end or server.stop for some period and after that to make server.on(80) again? Or at least to make the server stop listening and after that resume it ?

I am trying to run SMTP and NTP client along with the Async Web Server. The problem is if the server receive any HTTP request while the ESP8266 is sending the email message (up on certain event), it is rebooted. I suppose it's because of not enough memory to do the both jobs at the same time.
The same is with the NTP request.
Without server started both SMTP and NTP clients works fine.

I can post code if necessary.
Thank you in advance!

me-no-dev, 2016-11-29:
why don't we fix your issues instead? :)
Are your SMTP and NTP clients blocking or are they based on one of the Async libs?
It could be that you are not yielding enough in them and network gets clogged.
Do you want to share some code that we can then discuss?

denkovi1, 2016-11-30:
Both SMTP and NTP clients are not based on Async libs. Only the server is based on Async libs. You can watch my problem by uploading file (maybe about 300kB+) and while uploading press Send Email button. Then ESP restarts. I am using NodeMCU ESP8266 board with 4M(1M SPIFFS) flash. Any help will be appreciated. Thanks in advance. Here is my code:

Sorry for the inconvinience, but github does not allow me to upload zip file. (Just remove the .txt).
[Gsender.cpp.txt](https://github.com/me-no-dev/ESPAsyncWebServer/files/621309/Gsender.cpp.txt)
[Gsender.h.txt](https://github.com/me-no-dev/ESPAsyncWebServer/files/621310/Gsender.h.txt)
[index.html.txt](https://github.com/me-no-dev/ESPAsyncWebServer/files/621311/index.html.txt)
[srv_ntp_smtp.ino.txt](https://github.com/me-no-dev/ESPAsyncWebServer/files/621317/srv_ntp_smtp.ino.txt)

```
/*
   Uses ESPAsyncWebServer library downloaded from https://github.com/me-no-dev/ESPAsyncWebServer
   Uses Gmail Sender librabry downloaded from http://www.instructables.com/id/ESP8266-GMail-Sender/
   Uses NTP project downloaded from https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/NTPClient/NTPClient.ino
*/

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <EEPROM.h>
#include "Gsender.h"
extern "C" {
#include "user_interface.h"
}

#define RELAY_PIN               D5

unsigned int ticks = 0;
static os_timer_t myTimer;
AsyncWebServer server(80);
File uploadFile;
WiFiUDP Udp;

bool send_email = false;

IPAddress timeServerIP; // time.nist.gov NTP server address
const char* ntpServerName = "time.nist.gov";
const int NTP_PACKET_SIZE = 48; // NTP time stamp is in the first 48 bytes of the message
byte ntpBuffer[NTP_PACKET_SIZE]; //buffer to hold incoming and outgoing packets

const char* ssid = "DenYo";
const char* password = "64216421";

void timerCallback(void *pArg) {
  ticks++;
}

void handleFileUpload(AsyncWebServerRequest * request, String filename,
                      size_t index, uint8_t *data, size_t len, bool final) {
  if (!index) {
    Serial.printf("UploadStart: %s\n", filename.c_str());
    if (!filename.startsWith("/"))
      filename = "/" + filename;
    if (SPIFFS.exists(filename)) {
      SPIFFS.remove(filename);
    }
    uploadFile = SPIFFS.open(filename, "w");
    filename = String();
  }
  for (size_t i = 0; i < len; i++) {
    uploadFile.write(data[i]);
  }
  if (final) {
    uploadFile.close();
    Serial.printf("UploadEnd: %s, %u B\n", filename.c_str(), index + len);
  }
}

// send an NTP request to the time server at the given address
[... 112 lines cut ...]
  server.begin();

  os_timer_setfn(&myTimer, timerCallback, NULL);
  os_timer_arm(&myTimer, 1, true);
}

void loop() {
  if (ticks >= 500) {
    ticks = 0;
    digitalWrite(RELAY_PIN, !digitalRead(RELAY_PIN));
  }
  if (send_email) {
    Gsender *gsender;
    gsender = Gsender::Instance();
    String subject = "ESP8266 test!";
    if (gsender->Subject(subject)->Send("email@domain.com", "This is a test message from ESP8266!")) {
      send_email = false;
      Serial.println("Message send.");
    } else {
      send_email = false;
      Serial.print("Error sending message: ");
      Serial.println(gsender->getError());
    }
  }
}

matasondrak, 2016-12-11:
Thanks, these notifications just ate over 1GB data in a week. FU

Matheus-Garbelini, 2017-02-08:
@me-no-dev I also needed to disable async server interrupts. I believi it is slowing down my SD card transfers

me-no-dev, 2017-02-08:
https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/ESPAsyncWebServer.h#L389

then you will need to setup all server.on things again

mkeyno, 2017-02-09:
@me-no-dev  why not develop lib for such   communicate with webs  in async lib , use any sync udp or tcp port make the async sketch unstable

closed 2019-10-05 by stale[bot]