me-no-dev/ESPAsyncWebServer #106: std::bind class function for HandleRequest

hreintke, 2016-12-04:
I must be making a mistake with the std::bind usage but just out of places to search and ideas.

Made a minimal application to show : 
```
#include <Arduino.h>
#include <ESPAsyncWebserver.h>
#include <Hash.h>

void handleRequest(AsyncWebServerRequest *request)
{
}

class WebClass
{
public :
	AsyncWebServer ws;
	void classRequest (AsyncWebServerRequest *request)
	{
	}
	void begin()
	{
		// This ws.on() compiles OK
		ws.on("/test", HTTP_ANY, handleRequest);
		
		// This ws.on() gives compile time error 
		ws.on("/test", HTTP_ANY, std::bind(&WebClass::classRequest,this));
	}
};
```

I get the compile error : 
```
../../WebDemo.cpp: In member function 'void WebClass::begin()':
../../WebDemo.cpp:21:67: error: no matching function for call to 'AsyncWebServer::on(const char [6], WebRequestMethod, std::_Bind_helper<false, void (WebClass::*)(AsyncWebServerRequest*), WebClass* const>::type)'
   ws.on("/test", HTTP_ANY, std::bind(&WebClass::classRequest,this));
                                                                   ^
```
Can you give me a hint on usage ?

me-no-dev, 2016-12-04:
Why bind at all? ```ws.on("/test", HTTP_ANY, classRequest);``` should work

hreintke, 2016-12-04:
That is one of the trials I've done : 
```
../../WebDemo.cpp: In member function 'void WebClass::begin()':
../../WebDemo.cpp:22:40: error: no matching function for call to 'AsyncWebServer::on(const char [6], WebRequestMethod, <unresolved overloaded function type>)'
   ws.on("/test", HTTP_ANY, classRequest);
```

hreintke, 2016-12-04:
Have it compiling with : 

```
ws.on("/test", HTTP_ANY, std::bind(&WebClass::classRequest,this, std::placeholders::_1));
```

Expect it to work but will let you know

hreintke, 2016-12-04:
That is indeed the solution, works as expected.

Do you want me to PR the above minimal application in the readme.md ?

closed 2016-12-04 by hreintke

me-no-dev, 2016-12-04:
For sure! Make sure you add it to the table of contents also :)